esphome/issues #4066: dockerhub is stuck to 2022.12.3

koying, 2023-01-23:
### The problem

I get a message in HA telling me to update esphome to 2022.12.4 or later, but dockerhub only has 2022.12.3.
The doc still says `docker pull esphome/esphome`, so I assume dockerhub is still master?

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP32

olympia, 2023-01-23:
I am also experiencing the same issue, but with unRAID. Latest version what I can install is 2022.12.3.

koying, 2023-01-24:
It has been updated, now. Thanks.

closed 2023-01-24 by koying