esphome/issues #4067: Esp32 cam no image in HA Ui after OTA flash - worked previous to espHome 2022.12. Logs indicate everything working normal and getting images

ballakers, 2023-01-23:
### The problem

Everything used to work fine. VINDRIKTNING pm sensor, Dallas temp sensor, and esp32s cam board https://www.instructables.com/ESPHome-Esp32Cam-IKEA-VINDRIKTNING/ just this but with a Dallas sensor added and a better cam lens.  Then after the update to one of the 2022.12.x versions I noticed the temp sensor was unavailable and was also planning on flashing over serial to take advantage of the newest BLE proxy updates etc.  I have a few more of these and successfully have the camera working on the other after removing the Dallas sensor and flashing over serial. I then add back the Dallas sensor and logs are good with reporting temp etc but then after that OTA update the camera is no longer viewable in HA - still says idle in the HA entity and logs still report it's getting frames. All other sensors report fine, flashing over serial seems to fix but when I tried last that reintroduced the temp sensor errors again.

### Which version of ESPHome has the issue?

2022.12.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.x.x

### What platform are you using?

ESP32

### Board

esp32s cam generic popular one

### Component causing the issue

esp32_camera

### Example YAML snippet

```yaml
esphome:
  name: ikeasensor-med-cam
    
esp32:
  board: esp32cam
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

#API,OTA,Wifi info removed for post

captive_portal:

esp32_camera:
  name: ikeasensor-med-cam
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  max_framerate: 15 fps
  idle_framerate: 1.0 fps
  resolution: 800x600
  horizontal_mirror: true
  vertical_flip: true

output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4

binary_sensor:
  - platform: status
    name: "ikeasensor-med-cam status"
    
light:
  - platform: binary
    output: gpio_4
    name: "ikeasensor-med-cam light"

switch:
  - platform: restart
    name: "ikeasensor-med-cam restart"
  
# Example configuration entry
uart:
  rx_pin: GPIO3
  baud_rate: 9600

sensor:
  - platform: pm1006
    pm_2_5:
      name: "ikea med PM 2.5µm"
  - platform: wifi_signal
    name: "ikea-med-cam WiFi"
    update_interval: 60s    
  - platform: dallas
    address: 0x2b3cdef6486b9428
    name: "PM Ikea med temp"
    

dallas:
  pin:
    number: GPIO16
    mode:
        input: true
        pullup: true
  update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
no they report no errors except the usual GPIO0&4 strapping warning
```

### Additional information

They worked solid for many months on older versions of espHome. I await 2023.1.XX!!! Thanks a bunch to the devs!

SeByDocKy, 2023-01-25:
I have exactly the same behaviour when trying to us a pulse_meter with the ESP32cam.
After @jesserockz  disscussion sometimes ago, it seems the ESP32cam lib is no more compatible with new arduino lib used by ESPhome. As fat I understood, we need to wait the new version the ESP32cam lib to be linked with ESPhome. You can always install a former version of ESPhome to have your node working again. Sorry... I am stucked too with this silly bug.
This is the last working ESPhome version with ESP32cam sharing something with: pip install git+https://github.com/esphome/esphome.git@d8746266625c5a4c9127577297876e445cffe859

szerencl, 2023-02-03:
For me the device stopped responding through the API at all, but luckily OTA was still possible.
After narrowing it down to the esp32_cam module I tried downgrading my docker image but it didn't work, However executing the above https://github.com/esphome/issues/issues/4067#issuecomment-1403791497 command in the container worked, so thanks for this, you saved me from a lot of hair pulling.

szerencl, 2023-12-27:
It's still a problem with ESPhome 2023.12.5.