esphome/issues #4068: remote_receiver - set_filter_us() not working

ShadyMCShadesen, 2023-01-23:
### The problem

I recently updated one of my esphome nodes which hadn't had one for a long time.
Now remote_receiver_remotereceivercomponent throws an " unsigned conversion from 'int' to 'uint_8t' " error for the `set_filter_us()` function.
This changes the value for the filter attribute and my rf_receiver is not working anymore.

### Which version of ESPHome has the issue?

2022.12.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
remote_receiver:
  pin: 
    number: $rf_receiver_pin
    inverted: true
    mode: INPUT_PULLUP
  tolerance: 50%
  filter: 0.65ms
  idle: 10ms
  buffer_size: 10kb
```

### Anything in the logs that might be useful for us?

```txt
src/main.cpp: In function void setup();
src/main.cpp:423:58: warning: unsigned conversion from 'int' to 'uint8_t (aka unsigned char'} changes value from 650 to 138 [-Woverflow]
  remote_receiver remote receivercomponent-
>set_filter_us(650);

^~~
```

hphde, 2023-02-06:
Saw this also this morning.
```
src/main.cpp: In function 'void setup()':
src/main.cpp:197:27: warning: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '350' to '94' [-Woverflow]
  197 |   rf433_in->set_filter_us(350);
      |                           ^~~
```

ShadyMCShadesen, 2023-02-06:
> Saw this also this morning.
> ``` c++
> src/main.cpp: In function 'void setup()':
> src/main.cpp:197:27: warning: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '350' to '94' [-Woverflow]
>   197 |   rf433_in->set_filter_us(350);
>       |                           ^~~
> ```
> 
> Config snippet:
> ``` yaml
> remote_receiver:
>   - id: rf433_in
>     pin: 
>       number: D7
>     tolerance: 60%
>     filter: 350us
>     idle: 4ms
>     buffer_size: 2kb
> ```

Yeah i copied the original remote_receiver component folder to my custom_components folder of esphome and edited it for myself.

``` yaml
# Import custom_components
external_components:
  - source:
      type: local
      path: custom_components
    components: [ remote_receiver ]
```

In "remote_receiver.h":
Edit line 48 from "uint8_t" to "uint16_t" or "uint32_t"

ShadyMCShadesen, 2023-02-06:
> In "remote_receiver.h":
> Edit line 48 from "uint8_t" to "uint16_t" or "uint32_t"

If you use the ESP32 😅, i would recommend to repeat it for line 23 and 64

closed 2024-05-02 by jesserockz