esphome/issues #4069: sprinkler.queue_valve fails without run_duration

benbkoch, 2023-01-23:
### The problem

The docs state "If run_duration is not specified or is zero, the sprinkler controller will use the valve’s configured run duration" however if a run_duration is not specified, it generates the following error `Don't know what '0' means as it has no time *unit*! Did you mean '0s'?.`

### Which version of ESPHome has the issue?

stable

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
- seconds: 0
      minutes: 0
      hours: 5
      days_of_week: SUN
      then:
        - sprinkler.queue_valve:
            id: backyard_sprinklers
            valve_number: 1
```

gednet, 2023-04-17:
A have similar problem - when valve run time is set for 0 sec then starts a full cycle but stops at this valve. Nothing happens, counting is frozen. This behavior is problematic because I use "Smart irrigation" to calculate my water budget. When that integration count 0 for one valve then controller stop working. Probably it's the same problem like in previous post.

Alex1602, 2023-05-10:
I stumbled uppon the same issue, It might be related to the way the timer is set in the SprinklerQueueItem:
https://github.com/esphome/esphome/blob/8822b6c808881390b1bf77423ac78430bc0b6346/esphome/components/sprinkler/sprinkler.cpp#LL852C20-L852C20

`SprinklerQueueItem item{valve_number.value(), run_duration.value()};`
vs
`this->fsm_request_(valve_number.value(), run_duration.value_or(0));` 
a few lines above.
https://github.com/esphome/esphome/blob/8822b6c808881390b1bf77423ac78430bc0b6346/esphome/components/sprinkler/sprinkler.cpp#LL846C3-L846C70

I havent checked it thoroughly yet though.

Alex1602, 2023-05-10:
Error is comming from here:
https://github.com/esphome/esphome/blob/8822b6c808881390b1bf77423ac78430bc0b6346/esphome/config_validation.py#L685 

`
def time_period_str_unit(value):
    """Validate and transform time period with time unit and integer value."""
    check_not_templatable(value)

    if isinstance(value, int):
        raise Invalid(
            f"Don't know what '{value}' means as it has no time *unit*! Did you mean '{value}s'?"
        )
    if isinstance(value, TimePeriod):
        value = str(value)
    if not isinstance(value, str):
        raise Invalid("Expected string for time period with unit.")
`

Alex1602, 2023-05-10:
Here the value check is specified
https://github.com/esphome/esphome/blob/8822b6c808881390b1bf77423ac78430bc0b6346/esphome/components/sprinkler/__init__.py#L275-L283

Here the check is actually assembled (multiple checks)
https://github.com/esphome/esphome/blob/8822b6c808881390b1bf77423ac78430bc0b6346/esphome/config_validation.py#L751C30-L756

Fortunately it helps if you just add the following to your config:
`run_duration: 0s`
The code itself will replace the 0s with the number you have SET on the valve.

So either the docs need to be changed or the default number (which is now set as 0) should be set to 0s or the check needs to be changed entirely

kbx81, 2023-05-11:
Also was mentioned [here](https://github.com/esphome/issues/issues/4502)

closed 2023-05-11 by kbx81