esphome/issues #4070: Unable to upload to Pico W

Maarten-buelens, 2023-01-24:
### The problem

i have tried a lot of different ways but none of them work on windows and on a raspberry pi

- via https://web.esphome.io/ able to install and configure wifi but no logs or device on network
- via the local dashboard -> install -> manual install -> copy to pico it resets but unable to view logs
- via esphome run failed to upload

After a lot of trying it was discovered in the dashboard but as a esp01_1m and i was still unable to upload
Logs where sometimes working over wifi but not over serial.

I also tried different usb cables but no change

### Which version of ESPHome has the issue?

2023.1.0-dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
esphome:
  name: led-blink-pico

rp2040:
  board: rpipicow  # rpipico for Pico (non-W)

  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

output:
  - platform: gpio
    pin:
      number: 32  # 25 for Pico (non-W)
      mode: output
    id: LED

wifi:
  ssid: REDACTED
  password: REDACTED
  fast_connect: true

logger:
  level: DEBUG

interval:
  - interval: 500ms
    then:
      - output.turn_on: LED
      - delay: 250ms
      - output.turn_off: LED
```

### Anything in the logs that might be useful for us?

```txt
esphome run pico.yaml  --device /dev/sda1

Processing led-blink-pico (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
RAM:   [===       ]  26.3% (used 68944 bytes from 262144 bytes)
Flash: [====      ]  36.3% (used 379000 bytes from 1044480 bytes)
Building .pioenvs/led-blink-pico/firmware.bin.signed
========================= [SUCCESS] Took 5.47 seconds =========================
Processing led-blink-pico (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20604.0 (2.6.4) 
 - tool-mklittlefs-rp2040-earlephilhower @ 5.100300.221223 (10.3.0) 
 - tool-openocd-rp2040-earlephilhower @ 5.100300.221223 (10.3.0) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
RAM:   [===       ]  26.3% (used 68944 bytes from 262144 bytes)
Flash: [====      ]  36.3% (used 379000 bytes from 1044480 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Using manually specified: /dev/sda1
Forcing reset using 1200bps open/close on port /dev/sda1
Uploading .pioenvs/led-blink-pico/firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
========================= [FAILED] Took 16.02 seconds =========================
```

Maarten-buelens, 2023-01-24:
Today i was able to upload via ota.

I first uploaded a config on my brothers pc. Serial does work on his.
Still not able to upload via serial myself.

TheCre8r, 2023-01-29:
I am also having this issue.
I received this error in the logs:
`Terminal disconnected: TypeError: Cannot read properties of null (reading 'pipeThrough')`

rixvet, 2023-10-22:
OTA update issue on RP2040 still persists, running 203.10.1 stable 

Updater log entries:
```
INFO ESPHome 2023.10.1
INFO Reading configuration /config/esphome/growatt-reader.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing growatt-reader (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.30400.0 (3.4.0) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- noise-c @ 0.1.4
|-- MD5Builder @ 1.0.0
|-- lwIP-Ethernet @ 1
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
RAM:   [===       ]  28.1% (used 73692 bytes from 262144 bytes)
Flash: [=====     ]  51.1% (used 533624 bytes from 1044480 bytes)
Building .pioenvs/growatt-reader/firmware.bin.signed
========================= [SUCCESS] Took 7.58 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of growatt-reader.local
INFO  -> 192.168.178.142
INFO Uploading /data/build/growatt-reader/.pioenvs/growatt-reader/firmware.bin (545656 bytes)
Uploading: [==                                                          ] 4% 
ERROR Error sending data: [Errno 104] Connection reset by peer
```

Log entries on local serial console:
```
[D][ota:151]: Starting OTA Update from 192.168.178.151...
[D][ota:320]: OTA in progress: 0.2%
[W][ota:311]: Error writing binary data to flash!, error_code: 131
```

ssieb, 2023-10-22:
```
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
```
According to that, there's no space to put the OTA data.  I don't know why there's a filesystem section...

leftsock, 2024-05-01:
Several DIFFERENT errors have been reported in this bug. I have the same error as the original first post.

`
pip3 install esphome==2024.4.0

cat >recirc.yaml <<'__EOF__'
esphome:
  name: "recirculation-pump"
  comment: "Hot water recirculation pump"
  area: "Garage"

rp2040:
  board: rpipicow
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

wifi:
  ssid: "REDACTED"
  password: "REDACTED"
  # fallback hotspot in case wifi connection fails
  ap:
    ssid: "recirculation-pump"
    password: "REDACTED"

api: # Home Assistant API
  password: "REDACTED"
ota:
  password: "REDACTED"

logger:
switch:
  - platform: gpio
    id: relay
    restore_mode: ALWAYS_OFF
    pin:
      number: GPIO0
      #inverted: true
  - platform: template
    name: "Hot water recirculation pump; on, 100s, off"
    turn_on_action:
      - switch.turn_on: relay
      - delay: 10s # 100s
      - switch.turn_off: relay
__EOF__
env LANG=C.UTF-8 esphome -s device /dev/ttyACM0 run recirc.yaml
INFO ESPHome 2024.4.0
INFO Reading configuration recirc.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing recirculation-pump (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.30702.0 (3.7.2) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
Dependency Graph
|-- WiFi @ 1.0.0
|-- LEAmDNS @ 1.2
|-- Updater @ 1.0
|-- MD5Builder @ 1.0.0
|-- lwIP-Ethernet @ 1
|-- lwIP_CYW43 @ 1
|-- SPI @ 1.0
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
RAM:   [===       ]  28.1% (used 73588 bytes from 262144 bytes)
Flash: [====      ]  44.0% (used 459928 bytes from 1044480 bytes)
Building .pioenvs/recirculation-pump/firmware.bin.signed
========================= [SUCCESS] Took 3.66 seconds =========================
INFO Successfully compiled program.
Found multiple options for uploading, please choose one:
  [1] /dev/ttyACM0 (Board in FS mode - Board CDC)
  [2] Over The Air (recirculation-pump.local)
(number): 1
Processing recirculation-pump (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...

Auto-detected: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Uploading .pioenvs/recirculation-pump/firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
========================= [FAILED] Took 13.28 seconds =========================
`

Running on Ubuntu 22.04.4 LTS

My main complaint is that there's no log for me to inspect; no hint as to what failed.

leftsock, 2024-05-02:
WORKAROUND: hold down button, plug in, mount as drive, copy .esphome/build/recirculation-pump/.pioenvs/recirculation-pump/firmware.uf2 to PicoW, unmount, and that works for the very first install. Once that works, I can use OTA. Note that update via USB still fails, but for a different reason.

`...
Loading into Flash: [==============================]  100%

*** [nobuild] Source `.pioenvs/recirculation-pump/firmware.bin.signed' not found, needed by target `nobuild'.
========================== [FAILED] Took 9.93 seconds ==========================
`

rixvet, 2024-09-17:
FYI: Managed to get OTA to work again. Simply mounting as a drive and copy manually did not do the trick for me. I had to issue an explicit "reset the flash memory" with instructions provided by Raspberry:

_Pico’s BOOTSEL mode lives in read-only memory inside the RP2040 chip, and can’t be overwritten accidentally. No matter what, if you hold down the BOOTSEL button when you plug in your Pico, it will appear as a drive onto which you can drag a new UF2 file. There is no way to brick the board through software. **However, there are some circumstances where you might want to make sure your Flash memory is empty. You can do this by dragging and dropping a special UF2 binary onto your Pico when it is in mass storage mode.**_
source: https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#resetting-flash-memory

Ones completed I could mount as drive and next the OTA updates also started working again.

billmaterial, 2024-09-18:
@rixvet  @Maarten-buelens 
I am having similar problems, I have a pico display and pico display 2,0 both with a pico w attached.
I cannot get them to upload new code by OTA or USB as it will always stall half way through. I have to press install and then plug it into USB at exactly the right moment for it to upload correctly.
The weird thing is if I remove the "display" section from ESPhome it seems to works correctly.
Both displays run on the SPI bus.
GP19 is MOSI
GP18 is SCLK
GP17 is LCD_CS
GP16 is LCD_DC
The LCD_RESET is connected to the pico w RUN pin and as such is not used in the code as this prevents it from booting as would be expected.

rixvet, 2024-09-23:
@billmaterial two suggestions:

- First did you try to reset the flash memory as described above to see if this will fix it? 
- If this still fails, you might have issues with your power supply to the board/display, could you try detaching the display completely and try again?

billmaterial, 2024-09-24:
@rixvet Thank you for your suggestions.
I have reset/cleared the flash memory and the power supply is good. These displays use very little power.

I have a pico display and a pico display 2, both of the pico w's have the same issue. If I comment out the Display part then it updates correctly with OTA and over the USB port. I think the update part of the code is getting hung up with something still running in the background to do with the display. I have also increased the update interval of the display to 10 seconds but it still refuses to update every time.
It also stops receiving data from HA API at times and I have implemented a 1 minute timeout if no data is received.

I have used exactly the same code on an ESP-32 and have had no problems.
My only option at the moment is to make a PCB to convert the pinout of the pico displays to an ESP-32 based board as the Pico W seems to have issues.

YAML Code: (It's work in progress so please excuse the mess)
 
```yaml
esphome:
  name: pico-display-2
  friendly_name: Pico Display 2
  on_boot:
    priority: -10
    then:
      - script.execute: dim_backlight_once
      - script.execute: reboot_script

# Define a flag to track fallback execution
globals:
  - id: fallback_executed
    type: bool
    restore_value: no
    initial_value: 'false'

script:
  - id: dim_backlight_once
    mode: restart
    then:
      - delay: 1min
      - light.dim_relative:
          id: backlight_oled
          relative_brightness: -75%

  - id: reboot_script
    mode: restart
    then:
      - delay: 1min
      - if:
          condition:
            text_sensor.state:
              id: energy_led
              state: "on"
          then:
            - script.execute: reboot_script
          else:
            - logger.log: "Rebooting due to sensor state not changing"
            - button.press: restart_button

  - id: update_blue_led
    mode: restart
    then:
      if:
        condition:
          and:
            - api.connected:
            - wifi.connected:
            - lambda: 'return id(energy_led).state == "on" || id(energy_led).state == "off";'  # State is availiable  
            - lambda: 'return id(ha_time).now().hour >= 7 && id(ha_time).now().hour < 22;'  # Time is between 7am and 10pm
        then:
          - light.turn_on: led_blue
          - delay: 1s
          - light.turn_off: led_blue

  - id: felix_led
    mode: restart
    then:
[... 378 lines cut ...]
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%
  - id: my_white
    red: 100%
    green: 100%
    blue: 100%
  - id: my_black
    red: 0%
    green: 0%
    blue: 0%
  - id: my_cyan
    red: 0%
    green: 100%
    blue: 100%
```

rixvet, 2024-09-26:
@billmaterial I will try to reproduce your issue, how-ever I will need to source some hardware. Could you clarify which 'pico display' you are using?  Is this by any chance "Pico Display Pack by [Pimoroni]" and "Pico Display Pack 2.0 by [Pimoroni]"?

billmaterial, 2024-09-28:
@rixvet Thank you, yes you are correct it is both of those displays by Pimoroni. Another weird issue I have seen is the pwm fading of the led sometimes is smooth as expected and other times the fading is not working and the led changes between 2 discrete steps of brightness. I have a Pico w that has no display and has worked perfectly for over a year.

rixvet, 2024-10-01:
@billmaterial I have ordered to hardware and will try (hopefully soonish) try to reproduce your issue.

billmaterial, 2024-10-05:
@rixvet That's great! Thank you. I really hope together we can find a way to fix this annoying little issue as these are useful displays to have around to display specific information. Without the ability to update them using OTA they are less useful. I don't mind designing a custom PCB with an ESP32 instead of the Pico W but it would be a more expensive option and the whole point is to keep the costs down, also it would be larger as the ESP32 modules are wider than the pin pitch of the Pico W, might as well use a CYD (cheap yellow display) with a touch screen.

billmaterial, 2024-10-25:
I think I have found the cause of the Pico W not updating over OTA, its the sensors!
I commented out all the code then added in sections of it and updated wirelessly using OTA.
When I started to add the sensors back in it started to fail again.
Adding this causes it to fail when updating OTA.

```sensor:
  - platform: homeassistant
    id: power
    entity_id: sensor.esphome_web_Redacted_pzem_004t_v3_power
    internal: true
```
The text sensors seem to be OK.
Am I importing the sensor data from another device through `- platform: homeassistant` incorrectly?

ssieb, 2024-10-25:
You can try adding a safe mode button to put the device into safe mode before trying to OTA update it.

billmaterial, 2024-10-29:
@ssieb Thanks for your suggestion, I have tried this and it did update using OTA in safe mode, however the main issue with this approach is the Pico W will not connect to WiFi again and times out waiting for the OTA to start which never happens as it will not connect to WiFi again. This seems to happen to all my Pico W's after an update all with different programs running on them, I have one with just the onboard LED indicating a state from home assistant and this will update over OTA but then never connects back to WiFi, it has to be power cycled. Sometimes they can take well over 2 minutes to reconnect to WiFi again if at all. There seems to be a problem with the Pico W connecting to WiFi especially after an update or reset / safe mode reset.
So to sum up, currenty I cannot find a way to update the Pico W over OTA at all. I have no issues with my many ESP32's so I assume its an issue with the RP2040 code. I will carry on working on this (I like a challenge) but I am getting to the point of dumping my Pico W's for good!

billmaterial, 2024-10-29:
Just tried changing the ```interface: spi``` to ```interface: software```, after this change I could update it using OTA, this cannot be used as a workaround as the display update is much too slow but it may indicate where the issue may lie, maybe with the SPI bus?