esphome/issues #4073: Removing Components leads to compile failure and requres to clean the build files

bernikr, 2023-01-25:
### The problem

When I remove a component (in the following example the Status LED) from a yaml file that previously contained the component and was already built then I get the the following error on compilation: (blanked out the device name manually)

```
*** [/data/###/.pioenvs/###/src/esphome/components/status_led/status_led.o] Source `src/esphome/components/status_led/status_led.cpp' not found, needed by target `/data/###/.pioenvs/###/src/esphome/components/status_led/status_led.o'.
========================= [FAILED] Took 26.24 seconds =========================
```

When I then click on "Clean Build Files" for that device and run the compilation again, it works flawlessly.

### Which version of ESPHome has the issue?

v2022.12.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP32-IDF

### Board

shelly plus 1

### Component causing the issue

tested with status_led, button and switch

### Example YAML snippet

```yaml
status_led:
  pin:
    number: GPIO0
    inverted: true
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/###.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ### (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
*** [/data/###/.pioenvs/###/src/esphome/components/status_led/status_led.o] Source `src/esphome/components/status_led/status_led.cpp' not found, needed by target `/data/###/.pioenvs/###/src/esphome/components/status_led/status_led.o'.
========================== [FAILED] Took 8.95 seconds ==========================
```

blue-hound, 2023-09-06:
To chime in, I am having the same problem on ESPHome 2023.8.3. If I remove an output button from the YAML config and click install the compile fails because the button component is not found. If I clean the build files and click install it works without issue.

anoblet, 2024-07-19:
Same problem just now removing light component from `ld2410c`.

dvbit, 2024-08-07:
same here:
remove text component
got error

on esphome lvgl branch

clydebarrow, 2024-08-23:
Fixed by https://github.com/esphome/esphome/pull/7344

dzikus, 2024-09-21:
Same problem. Template sensor removed:
ESPHome 2024.9.0

[.pioenvs/airpurifier3h/src/esphome/components/template/sensor/template_sensor.o] Source `src/esphome/components/template/sensor/template_sensor.cpp' not found, needed by target `.pioenvs/airpurifier3h/src/esphome/components/template/sensor/template_sensor.o'.
========================== [FAILED] Took 1.65 seconds ==========================