esphome/issues #4075: UPDATE now can't connect to device

tonydm, 2023-01-25:
### The problem

Hello,

In the ESPHome panel I clicked the blue UPDATE to update one of my controllers (esp-wrover-kit).  I followed the update process via the ESPHome device log in another tab.  The update compiled and successfully updated.  However, the device never came back online, now the device is unreachable.  The device IP is staticly set in the config and reserved in the DHCP config.  Also, there were no config changes made, only the UPDATE.  WTH happened that an update breaks my device?

Thank you

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.1

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

esp-wrover-kit

lasry1, 2023-01-26:
It happens also to me every time i update the Bluetooth proxy over ESPHOME (Wireless)

Then i need to plug it into the computer and install again bluetooth proxy with a cable

randybb, 2023-01-26:
We need a serial log, without that nobody can help you.
Btw, I have just updated 27 devices + changed api pwd to api enc key without any issues.

Delta1977, 2023-01-26:
here the same after Updating the last 2 Versions. Its a ESP32 Devkit with a IR (   - platform: pulse_meter) attached.
A Powercyle helps.

tonydm, 2023-01-26:
@randybb With due respect, telling us that you updated 25 devices with no problem helps nobody.  I'm bringing up an issue I have.  In my case, my HA was/is functioning fine, for the last two years.  I've done all updates to Core, Addons, etc.  In this case, I simply followed the UPDATE prompting.  The update completed 100%, device went offline as they do with an update of any kind, but the device never came back online.  As far as a serial log, this device is two hours away from me.

randybb, 2023-01-26:
Basically it says that I cannot reproduce your issue based on provided information.
Without a log nobody will help you.

tonydm, 2023-01-26:
@randybb I get what you are saying.  However, since the update showed success, a little more help from the ESPHome side would be appropriate.  For anyone that runs into this issue and then having to get a "serial" log, your saying we have to climb to, crawl to, dig out the controller from where it's physically located so we can physically connect a cable.  That's the last step I or anyone wants to take.  So if ESPHome is going to push an UPDATE, in my book, there needs to be more transparency.  What was the update?  What in the update could cause the device to not come back online when, prior to the process, the device, ESPHome, and HA were functioning fine?

lasry1, 2023-01-26:
I think i may have find the cause of the issue, at least on my particular case.

When i install the BT proxy through the website I type the SSID and password of the wireless network. However when i adopt the device in home assistant (ESPHOME) it changes it to have the ssid and password written in secrets.yaml

I had not added this info into that file so every time ESPHOME did an update and rebooted the device it never connect back to the wifi.

I have added the secrets and now it seems to work

lasry1, 2023-01-27:
> I think i may have find the cause of the issue, at least on my particular case.
> 
> When i install the BT proxy through the website I type the SSID and password of the wireless network. However when i adopt the device in home assistant (ESPHOME) it changes it to have the ssid and password written in secrets.yaml
> 
> I have added the ssid and password instead of directing it to secrets and it seems to work (so far, we'll see in like 10 minutes when another update comes available)

It took longer than 10 minutes to receive yet another update but after 16 hours i have received again another ESPHOME update.

As suspected that has solved my issue, adding the wireless SSID and password through ESPHOME rather than using the secrets allows me to update the BT Proxy without loosing the connection

AussiSG, 2023-02-11:
I had the similair problem. 
Seem that some recent update has destroyed the use of !secrets

Now when I update/reflash a device it's unable to connect to the wifi because it cant find a corresponding wifi.
Only works when putting in the real ssid and password ..

AussiSG, 2023-02-11:
@OttoWinter do you have any idea why the !secrets part is not working anymore?

AussiSG, 2023-02-17:
@tonydm and @OttoWinter ,

I managed to solved my issue. Not sure if it will solve yours tonydm.
But what I did is that I cleaned out my secrets.yaml withing esphome.

I had something like ( cant remember correctly changed deleted the last 2 rows already)
```
<<: !include ../secrets.yaml
wifi_pass : !secrets wifi_pass
wifi_ssid : !secrets wifi_ssid
```

And I changed it to only 

`<<: !include ../secrets.yaml`

Now it works again as before.