esphome/issues #4076: Web server events crashing

barbiani, 2023-01-25:
### The problem

ESPAsyncWebServer-esphome server events causes crash on esp32 after a few minutes.

Building with platformio, espressif32 platform v6.0.0.

Runs for a few minutes and crashes.

Looks like something deep in the sdk that only happens with web server events and a running browser client.

Sample taken from https://randomnerdtutorials.com/esp32-web-server-sent-events-sse/

### Which version of ESPHome has the issue?

ESPAsyncWebServer-esphome v3.0.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

node32s

### Anything in the logs that might be useful for us?

```txt
[   412][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   510][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
[  1070][I][lib_ap.cpp:179] lib_ap_init(): AP IP address: 10.0.0.1

[  1071][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 11 - AP_STOP
[  1072][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
[275336][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[275529][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4018de80  PS      : 0x00060e30  A0      : 0x8018cebc  A1      : 0x3ffce0b0  
A2      : 0x00000009  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x3ffd8dd8  A7      : 0x3ffd920c  A8      : 0x00000029  A9      : 0x000001b0  
A10     : 0x00000002  A11     : 0x0000003f  A12     : 0x00000000  A13     : 0x3ffbff80
A14     : 0x12a1e359  A15     : 0x3ffc0568  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002d  LBEG    : 0x4008990c  LEND    : 0x40089922  LCOUNT  : 0xffffffff

Backtrace: 0x4018de7d:0x3ffce0b0 0x4018ceb9:0x3ffce170 0x4010b941:0x3ffce1b0 0x4010c17b:0x3ffce1e0 0x4018e24a:0x3ffce220

  #0  0x4018de7d:0x3ffce0b0 in ppResortTxAMPDU at ??:?
  #1  0x4018ceb9:0x3ffce170 in lmacEndFrameExchangeSequence at ??:?
  #2  0x4010b941:0x3ffce1b0 in lmacProcessTxSuccess at ??:?
  #3  0x4010c17b:0x3ffce1e0 in lmacProcessTxComplete at ??:?
  #4  0x4018e24a:0x3ffce220 in ppTask at ??:?
```

bf8392, 2023-02-01:
Same problem... running/starting a Webserver leads to crash...

ssieb, 2023-02-01:
@barbiani you didn't say which esphome version you have or what your yaml is.

bf8392, 2023-02-02:
Maybe mine will help?

```
substitutions:
  device_name: somename
  device_description: somedescription
  ap_fallback_password: somepassword
     
esphome:
  name: ${device_name}
  name_add_mac_suffix: false
  comment: "${device_description}"

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
 
# Disable serial Logging
logger:
  baud_rate: 0

ota:

web_server:
  port: 80
    
sensor:
 
  - platform: uptime
    name: "${device_name} Uptime"
    update_interval: 30s
    
  - platform: wifi_signal
    name: "${device_name} Wi-Fi Signal"
    update_interval: 60s
    
  - platform: adc
    pin: 34
    id: hauptwasserdruck
    name: "Pressure"
    device_class: pressure
    state_class: "measurement"
    update_interval: 30s
    unit_of_measurement: "bar"
    accuracy_decimals: 1 
    filters:
     - calibrate_linear:
       - 0.48 -> 0
       - 5 -> 20.68
    
  - platform: dht
    pin:
      number: 32
      mode:
        input: true
    temperature:
      name: "DHT1 Temperature"
    model: DHT22
    update_interval: 30s

text_sensor:

  - platform: wifi_info
    ip_address:
      name: "IP Address"
    ssid:
      name: "Wi-Fi SSID"
    bssid:
      name: "Wi-Fi BSSID"

  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true

i2c:
  sda: 13
  scl: 14
  scan: true
  id: bus_a

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(roboto), "Tester");

font:
  # gfonts://family[@weight]
  - file: "gfonts://Roboto"
    id: roboto
    size: 20
 
switch:
  - platform: gpio
    name: "Relay1"
    pin: 5
    inverted: true

  - platform: gpio
    name: "Relay2"
    pin: 12
    inverted: true

# esp32_ble_tracker:

esp32_ble_beacon:
  type: iBeacon
  uuid: 'c29ce823-e67a-4e71-bff2-abaa32e77a98'
```
Reboot happens as soon I access or try to reload the webserver...then it's stable for sometime, and crashes again when reloaded/connected too often or too long. ESP32 Wroom 32D, ESPHOME 2022.12.8

bf8392, 2023-02-02:
Related? https://github.com/esphome/issues/issues/4083

ssieb, 2023-02-02:
Can you get serial logs from around the crash?

bf8392, 2023-02-02:
Yes...as soon as I try to refresh, wifi gets disconnected (credentials are fine, double checked, and first load always okay)...just on multiple load this happens:

```
[W][wifi_esp32:494]: Event: Disconnected ssid='somewifi' bssid=[redacted] reason='Auth Expired'
[W][wifi:549]: Error while connecting to network.
Terminal disconnected: BufferOverrunError: Buffer overrun

bf8392, 2023-02-02:
The "buffer overrun" was due to stay connected after a flash...I always get different reasons when reloading the webserver why wifi is not working:
```
[W][wifi_esp32:494]: Event: Disconnected ssid='somewifi' bssid=[redacted] reason='Beacon Timeout'
[W][wifi:125]: WiFi Connection lost... Reconnecting...
[W][wifi:585]: Restarting WiFi adapter...
```
This seems to lead to the reboot

ssieb, 2023-02-02:
You can't get serial logs of the crash?

SecSelS, 2023-02-02:
Sadly that's all the "unusual" output...it doesn't give a reason why it crashed...

bf8392, 2023-02-02:
Sadly that's all the unusual output...it doesn't give a reason why it crashed...

ssieb, 2023-02-02:
Are you watching it over the serial?  It will definitely give you something.  And there will be the booting logs right after.

bf8392, 2023-02-02:
Yes I press "logs" and then "plug into this computer". If you mean the purple ones, yes they are there...

ssieb, 2023-02-02:
Can you join the discord server?  That will be much easier.

bf8392, 2023-02-02:
Yes of course =) give me a second...

bf8392, 2023-02-02:
sorry can you point me to it?

ssieb, 2023-02-02:
https://discord.gg/KhAMKrd

bf8392, 2023-02-02:
great great thanks =)

tomaszduda23, 2023-02-15:
I can reproduce this issue quite easy. Web server on esp32 is just broken. ESPAsyncWebServer-esphome is not thread safe but it is accessed from main thread and AsyncTCP-esphome thread. I did small patch https://github.com/esphome/ESPAsyncWebServer/pull/9 but it does not solve the problem completely.

```
WARNING Decoded 0x400f0620: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at .piolibdeps/adam-co/ESPAsyncWebServer-esphome/src/AsyncEventSource.cpp:320 (discriminator 2)
WARNING Decoded 0x400f061d: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at .piolibdeps/adam-co/ESPAsyncWebServer-esphome/src/AsyncEventSource.cpp:319 (discriminator 1)
WARNING Decoded 0x400e1942: esphome::web_server::WebServer::on_sensor_update(esphome::sensor::Sensor*, float) at src/esphome/components/web_server/web_server.cpp:374

```

tomaszduda23, 2023-02-15:
@barbiani @bf8392 would you be able to test if this improve the problem? With this implementation I'm not able to reproduce it anymore.
```
external_components:
  - source: github://pr#4444
    components: [async_tcp, web_server_base]
    refresh: always
```

bf8392, 2023-02-19:
Oh cool great great thanks! I'll try and test tomorrow :-)

bf8392, 2023-02-22:
Sorry it took some time...I also was not able to reproduce anymore thanks to the patch @tomaszduda23...also it seemed to improve webserver response time a lot!

barbiani, 2023-02-22:
I am building from the repo directly. Can you point me to the correct
commit?

tomaszduda23, 2023-02-22:
> ```
> external_components:
>   - source: github://pr#4444
>     components: [async_tcp, web_server_base]
>     refresh: always
> ```
Adding this to your configuration is enough. It fetch the patch automatically.

barbiani, 2023-02-22:
I am including the repository directly into my arduino project.

ssieb, 2023-02-24:
If you're running the esphome command to compile the yaml, then that bit will work.

tomaszduda23, 2023-02-24:
@ssieb any chance to get it merged? It may be a little bit controversial since no maintener.