esphome/issues #4077: PID climate heat action when climate mode set to cool and vice versa

Tr4um4z, 2023-01-25:
### The problem

I have the same problem as issue #2406

When Climate mode is set to "heat", and the set temp is below the current temp, the climate action will set to cooling. 

When Climate mode is set to "cool", and the set temp is above the current temp, the climate action will set to heating 

in my case i have the same output for heat or cool but i have tested it with two different outputs but it's the same

i have returned 0-10v signal (generated by a ledc output) to an adc GPIO input (attuenated to a 0-1v) for debugging directly in home assistant without using a tester and result that my signal goes from 10v - 0 - 10v  as you can see in the animated gif

the thermostat in heat mode should be inactive not in cooling and the output should be to 0

[image]

### Which version of ESPHome has the issue?

2022.12.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP32

### Board

esp32

### Component causing the issue

pid

### Example YAML snippet

```yaml
climate:
  - platform: pid
    visual:
      min_temperature: 10
      max_temperature: 30
      temperature_step: 0.5  
    id: pid_climate
    name: "Termostato Ufficio"
    sensor: temperature_sensor
    default_target_temperature: 21°C
    #Deadband per evitare continue attivazioni della valvola
    deadband_parameters:
      threshold_high: 0.5°C
      threshold_low: -1.0°C
    #To start with we recommend just setting the ki_multiplier to 0.05 (5%). Then set kp_multiplier to 0.05 (5%) if the controller is falling out of the deadband too often.
      kp_multiplier: 0.0   # proportional gain turned off inside deadband
      ki_multiplier: 0.05  # integral accumulates at only 5% of normal ki
      kd_multiplier: 0.0   # derviative is turned off inside deadband
      deadband_output_averaging_samples: 15   # average the output over 15 samples within the deadband
    heat_output: pwm_out
    cool_output: pwm_out
    control_parameters:
      kp: 0.55560
      ki: 0.00373
      kd: 20.69210
      max_integral: 0.0
      output_averaging_samples: 5      # << NEW!
      derivative_averaging_samples: 5  # << NEW!

output:
  - platform: ledc
    pin: GPIO19
    frequency: 1000 Hz
    id: pwm_out
    inverted: False
    min_power: 0.00
    max_power: 1.00
```

Tr4um4z, 2023-01-25:
[image]

Tr4um4z, 2023-01-26:
tested on a d1 mini , same problem

badilator, 2023-01-28:
My brother asked me to give a look, but I do not have any experience of this thing. My guess is the problem is around here:

```
void PIDClimate::write_output_(float value) {
  this->output_value_ = value;

  // first ensure outputs are off (both outputs not active at the same time)
  if (this->supports_cool_() && value >= 0)
    this->cool_output_->set_level(0.0f);
  if (this->supports_heat_() && value <= 0)
    this->heat_output_->set_level(0.0f);

  // value < 0 means cool, > 0 means heat
  if (this->supports_cool_() && value < 0)
    this->cool_output_->set_level(std::min(1.0f, -value));
  if (this->supports_heat_() && value > 0)
    this->heat_output_->set_level(std::min(1.0f, value));
```

I think that this->supports_cool_() and this->supports_heat_() only consider if there is the relative output configured, but do not consider if this->mode is HEAT, COOL or HEAT/COOL

so if temperature is under the target , you probably get here with a positive value.
if you are in COOL mode, and you have configured both outputs so you can hot switch mode between COOL and HEAT, you will skip the `cool_output_->set_level` but you will set `heat_output_->set_level`

I think that something like this may fix the problem:
`if (this->supports_heat_() && value > 0 && this->mode == HEATMODEVALUE)`
(HEATMODEVALUE should be the `mode` enum value for "heat")

it wold be better not even get here, and check for `this->mode` before, but I do not really know if that can impact other things.

Tr4um4z, 2023-01-28:
@OttoWinter can you check about this?
Thanks in advance

roma1141r1, 2024-09-22:
The problem is still relevant

TheBarnHome, 2025-06-07:
2025.5 here, I still have the same issue.

sgrasley, 2025-07-01:
I just ran into this behavior when I added the climate visual tag to my thermostat.  

climate: 
  - platform: ...
    visual: 

Looking forward to this being resolved.