esphome/issues #4079: FastLED Light (Clockless) can't compile on ESP32-S3 - fastpin.h:207:24: error: static assertion failed: Invalid pin specified

mishomvp, 2023-01-26:
### The problem

Hello,

board: Lolin  ESP32-S3 
I'm trying to get the LED (WS2812) on the board to light up. But it doesn't want to compile with that exact pin(gpio 38).
Thanks in advance!

### Which version of ESPHome has the issue?

2022.12.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP32

### Board

Lolin ESP32-S3

### Component causing the issue

fastled

### Example YAML snippet

```yaml
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO38
    num_leds: 1
    rgb_order: RGB
    name: "FastLED WS2812 Light"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp32-s3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-s3 (board: lolin_s3; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- FastLED @ 3.3.2
Compiling /data/esp32-s3/.pioenvs/esp32-s3/src/main.cpp.o
Archiving /data/esp32-s3/.pioenvs/esp32-s3/lib403/libAsyncTCP-esphome.a
[... 105 lines cut ...]
/data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/driver/include/driver/rmt.h:930:11: note: declared here
 esp_err_t rmt_set_pin(rmt_channel_t channel, rmt_mode_t mode, gpio_num_t gpio_num)
           ^~~~~~~~~~~
In file included from /data/esp32-s3/.piolibdeps/esp32-s3/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/esp32-s3/.piolibdeps/esp32-s3/FastLED/platforms.h:36,
                 from /data/esp32-s3/.piolibdeps/esp32-s3/FastLED/FastLED.h:52,
                 from src/esphome/components/fastled_base/fastled_light.h:16,
                 from src/esphome.h:19,
                 from src/main.cpp:3:
/data/esp32-s3/.piolibdeps/esp32-s3/FastLED/platforms/esp/32/clockless_rmt_esp32.h:399:20: warning: 'esp_err_t rmt_set_pin(rmt_channel_t, rmt_mode_t, gpio_num_t)' is deprecated: use rmt_set_gpio instead [-Wdeprecated-declarations]
         rmt_set_pin(mRMT_channel, RMT_MODE_TX, mPin);
         ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /data/esp32-s3/.piolibdeps/esp32-s3/FastLED/platforms/esp/32/clockless_rmt_esp32.h:97,
                 from /data/esp32-s3/.piolibdeps/esp32-s3/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/esp32-s3/.piolibdeps/esp32-s3/FastLED/platforms.h:36,
                 from /data/esp32-s3/.piolibdeps/esp32-s3/FastLED/FastLED.h:52,
                 from src/esphome/components/fastled_base/fastled_light.h:16,
                 from src/esphome.h:19,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32s3/include/driver/include/driver/rmt.h:930:11: note: declared here
 esp_err_t rmt_set_pin(rmt_channel_t channel, rmt_mode_t mode, gpio_num_t gpio_num)
           ^~~~~~~~~~~
*** [/data/esp32-s3/.pioenvs/esp32-s3/src/main.cpp.o] Error 1
========================== [FAILED] Took 3.80 seconds ==========================
```

nielsnl68, 2023-01-26:
Did you try my update at: https://github.com/esphome/esphome/pull/4257 

like: 
```yaml
external_components:
  - source: github://#4257
    components: [fastled]
    refresh: always

light:
  - platform: fastled
    rgb_order: GRB
    chipset: WS2812
    pin: GPIO4
    num_leds: 30
    restore_mode: RESTORE_AND_OFF
    name: "${device} NeoPixel Light"
    #effect: colored_strobe
    id: my_light
```

mishomvp, 2023-01-28:
Hello,
With your update, everything compiles without a problem. But the LED(WS2812) doesn't want to light up.
I installed Tasmota for testing and everything works without problems.
Any ideas?

nielsnl68, 2023-01-28:
That is strange. i use the same config as above and it did work fine for me.

mishomvp, 2023-01-28:
With ESP32-S3? I will also try an external WS2812.

PaulSchulz, 2023-02-01:
I am watching this issue and seeing the same issue with LED not lighting up. (They flash on power up).

. I have a Heltec LoRa Wifi V3 which I am trying to use to drive a WS2812 based 120 LED string (via pin 7).
(I am using config for board 'esp32-s3-devkitc-1' as the this hardware doesn't appear to have been added yet.) 

Thank you for the github pull request, this removed the errors completely.. I had to make one minor edit (add 'pr') to the URL.

external_components:
  - source: github://pr#4257
    components: [fastled]
    refresh: always

nielsnl68, 2023-02-02:
Top thanks for your testing. I have also updated the example i give above.

PaulSchulz, 2023-02-02:
I can confirm that it works for me on an esp32, bur not an esp32s3 platform.
- Esp32 -> Heltec Lora Wifi V2 - Works, on pin GPIO13
- Esp32s3 -> Heltec Lora Wifi V3 - Compiles and installs, but no control of LEDs on GPIO7

I think that this is a larger issue with Esp32s3 (or the WiFi_LoRa_V3 board). Inputs can be read but output don't appear to get set (not just for FastLED).

PaulSchulz, 2023-02-03:
It appears as though, on the esp32s3, gpio_reset_pin() needs to be called first for pins that are configured to be used for other peripherals. https://esp32.com/viewtopic.php?t=28028

(I have not confirmed this, other than it agrees with observations.)

Maybe I need to create another issue for this.

evlo, 2023-03-10:
Did you ever found solution for esp32s3

PaulSchulz, 2023-03-10:
**Not for the exp32s3**, other than use a different board (the Heltec V2 boards use the ESP32), but I would still like to know how to fix this.

I have recently gotten other other libraries to work on the esp32s3, namely for the SX1262 radio chip (Lora and LoRaWAN) over ISP, so I should have another look at this. If you are interested, my ESPhome code is here: https://github.com/PaulSchulz/esphome-esp32s3

I will pickup some more FastLED LED's (the last lot have gone on to be used) and give this a test.

PaulSchulz, 2023-03-13:
When I am trying to compile (for the setup below) I am getting the following error: '.piolibdeps/gpio-blink/FastLED/fastpin.h:207:24: error: static assertion failed: Invalid pin specified

Is there a different board definition I could use?

```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
```

I have been able to install the FastLED library directly and use a custom component to run a NeoPixel led using the library on a ESP32-S3 board.

Details are in the commit here: https://github.com/PaulSchulz/esphome-esp32s3/commit/c36747512140aa49abff9d441c01c02307248a53

PaulSchulz, 2023-04-20:
I now have a 'minimal' custom_component that works with NEOPIXELS

See: https://github.com/PaulSchulz/fastled_dev

Minimal functionality, and everything is hardcoded as I don't know enough about the ESPHome Python setup.

XtraLarge, 2023-08-22:
Hi, I've the T-Dongle-S3 and with this patch it works smooth. How will it be commited to the orginal brunch?

nielsnl68, 2023-08-22:
Can you try the PR i made for the FastLed and see of that will fix your issue: https://github.com/esphome/esphome/pull/4257

jesserockz, 2023-08-22:
Also try: https://esphome.io/components/light/esp32_rmt_led_strip

XtraLarge, 2023-08-23:
> Can you try the PR i made for the FastLed and see of that will fix your issue: [esphome/esphome#4257](https://github.com/esphome/esphome/pull/4257)

That is the relevant part of my my configuration and works without any problems
`#  - source: github://pr#4257
external_components:
  - source: github://pr#4257
    components: [fastled]
    refresh: always

light:
  - platform: fastled
  #- platform: fastled_spi
    chipset: APA102
    data_pin: GPIO40
    clock_pin: GPIO39
    num_leds: 1
    rgb_order: BGR
    name: "FastLED SPI Light"
  - platform: monochromatic
    output: output_38
    name: "LCD Backlight"
    id: lcd_backlight

spi:
  clk_pin: 5
  mosi_pin: 3`

XtraLarge, 2023-08-23:
> Also try: https://esphome.io/components/light/esp32_rmt_led_strip

Can you help me, how to map my config?

scottrknight, 2023-09-07:
Also watching this as I am getting the same "now it compiles but doesn't work" behavior with the onboard SK6812 on an M5 Stamp C3 (https://docs.m5stack.com/en/core/stamp_c3):
```
external_components:
  - source: github://pr#4257
    components: [fastled]
    refresh: always

light:
  - platform: fastled
    chipset: SK6812
    name: "SK6812 Light"
    id: LED
    pin: GPIO2
    num_leds: 1
    rgb_order: GRB
```
fastled_clockless works perfectly on a Stamp Pico (https://docs.m5stack.com/en/core/stamp_pico) with the same SK6812, but ESP32-PICO-D4 chip

lolorc, 2023-12-06:
hi there,
just to let you know, esp32_rmt_led_strip works ok with the one ws2812 led on my esp32-s3-devkitc-1 board. (GPIO48)

pr#4257 does compile with GPIO48 but the led never turns on.

stable esphome doesn't compile with GPIO48, afair it does with GPIO4 but I haven't been trying to attach a ws2812 strip on gpio 4

hitnrun30, 2024-01-17:
Two questions, why hasn't this been pushed to esphome. It seems silly that it's taking this long to merge it.  Second does this work if I set the chipset to ws2812b?