esphome/issues #4080: Docs: Please widen the page template

ChestersGarage, 2023-01-26:
### The problem

The page template used in the documentation is excessively narrow and the Table of Contents column does not display correctly on https://esphome.io/components/remote_transmitter.html  The Action links are truncated to the point of almost unusable.  Annotated screen cap attached.

[image]

Preferably, use a dynamic template that fits the content within whatever size browser window the user has.

Or maybe allow for the table of contents right side divider to be grabbed and dragged wider.

### Which version of ESPHome has the issue?

n/a

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

elproko, 2023-01-27:
I would opt-in for this change.

ChestersGarage, 2025-12-10:
The issue has been resolved without comments here.  Closing.