esphome/issues #4082: Script `build_codeowners.py` is failing.

Fabian-Schmidt, 2023-01-26:
### The problem

Hi,

As part of the submitting of a pull request I am requested to run `script/build_codeowners.py`.

But it fails on my machine with a non descriptive error and I am not a python person to troubleshoot it. This is on a fresh windows VM used just for developing EspHome.

Without fixing this I am blocked from submitting a PR.

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32

### Component causing the issue

build script

### Anything in the logs that might be useful for us?

```txt
C:\Repos\GitHub\Fabian-Schmidt\esphome\script>python build_codeowners.py
Unable to load component api.client:
Traceback (most recent call last):
  File "C:\Python312\Lib\site-packages\esphome\loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\importlib\__init__.py", line 124, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1175, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1148, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1119, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 682, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1007, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Python312\Lib\site-packages\esphome\components\api\client.py", line 6, in <module>
    from aioesphomeapi import APIClient, ReconnectLogic, APIConnectionError, LogLevel
  File "C:\Python312\Lib\site-packages\aioesphomeapi\__init__.py", line 3, in <module>
    from .client import APIClient
  File "C:\Python312\Lib\site-packages\aioesphomeapi\client.py", line 21, in <module>
    from .api_pb2 import (  # type: ignore
  File "C:\Python312\Lib\site-packages\aioesphomeapi\api_pb2.py", line 7, in <module>
    from google.protobuf import descriptor as _descriptor
  File "C:\Python312\Lib\site-packages\google\protobuf\descriptor.py", line 40, in <module>
    from google.protobuf.internal import api_implementation
  File "C:\Python312\Lib\site-packages\google\protobuf\internal\api_implementation.py", line 74, in <module>
    if _CanImport('google._upb._message'):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\google\protobuf\internal\api_implementation.py", line 64, in _CanImport
    mod = importlib.import_module(mod_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\importlib\__init__.py", line 124, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Metaclasses with custom tp_new are not supported.
Cannot find component bp1658cj. Make sure current path is pip installed ESPHome
```

mrzottel, 2023-02-05:
I am currently facing the same issue

ssieb, 2023-02-05:
That means you've configured something incorrectly and it can't find your component as the error message states.
If you used that pip command, you didn't follow the development install instructions.

mrzottel, 2023-02-05:
Yeah, shame on me. I manually executed the steps from the setup script and somehow skipped the line "pip3 install --no-use-pep517 -e .". After executing this, everything worked fine.

Fabian-Schmidt, 2023-02-15:
Thanks.

The install instruction is [Setting Up Development Environment](https://esphome.io/guides/contributing.html#setting-up-development-environment). It calls the linux bash script `script/setup`. I missed that this is actually a bash script.
So on Windows you must run the two steps from [the script](https://github.com/esphome/esphome/blob/dev/script/setup) manually.

The manual setup produces a bunch of issue as some components are not available on Windows. Afterwards I can run the `build_codeowners.py`. 
It still produces an error, but this time it works (See error below for reference). From my point of view I did not understand why for building code owners repointing the esphome instance was necessary at all. Just take the relative path... but again maybe some Python foo why this is not possible.

For reference the 'successful' output:
```
>python script/build_codeowners.py
Unable to load component api.client:
Traceback (most recent call last):
  File "c:\repos\github\fabian-schmidt\esphome\esphome\loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\importlib\__init__.py", line 124, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1175, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1148, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1119, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 682, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1007, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "c:\repos\github\fabian-schmidt\esphome\esphome\components\api\client.py", line 6, in <module>
    from aioesphomeapi import APIClient, ReconnectLogic, APIConnectionError, LogLevel
  File "C:\Python312\Lib\site-packages\aioesphomeapi\__init__.py", line 3, in <module>
    from .client import APIClient
  File "C:\Python312\Lib\site-packages\aioesphomeapi\client.py", line 21, in <module>
    from .api_pb2 import (  # type: ignore
  File "C:\Python312\Lib\site-packages\aioesphomeapi\api_pb2.py", line 7, in <module>
    from google.protobuf import descriptor as _descriptor
  File "C:\Python312\Lib\site-packages\google\protobuf\descriptor.py", line 40, in <module>
    from google.protobuf.internal import api_implementation
  File "C:\Python312\Lib\site-packages\google\protobuf\internal\api_implementation.py", line 74, in <module>
    if _CanImport('google._upb._message'):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\google\protobuf\internal\api_implementation.py", line 64, in _CanImport
    mod = importlib.import_module(mod_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\importlib\__init__.py", line 124, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Metaclasses with custom tp_new are not supported.
Wrote CODEOWNERS
```

closed 2023-02-15 by Fabian-Schmidt