esphome/issues #4085: Unable to get any device working with the media-player module

DJBenson, 2023-01-26:


alucryd, 2023-05-21:
Probably same issue here, I just got an ESP32-LyraTD-MSC, the media player is extremely finicky. Playing TTS has a good chance of crashing the board (like 50%), and playing local media crashes it every single time:
```
[19:33:18][D][media_player:059]: 'Media Player' - Setting
[19:33:18][D][media_player:066]:   Media URL: http://192.168.2.13:8123/api/tts_proxy/d9832b0a9d90b4dc814cce610ba0de96605958c2_en-us_a877e2b3bf_tts.piper.wav
[19:33:19][V][component:204]: Component i2s_audio.media_player took a long time for an operation (0.54 s).
[19:33:19][V][component:205]: Components should block for at most 20-30ms.
[19:33:24][V][component:204]: Component i2s_audio.media_player took a long time for an operation (0.51 s).
[19:33:24][V][component:205]: Components should block for at most 20-30ms.
[19:33:24][V][component:204]: Component i2s_audio.media_player took a long time for an operation (0.50 s).
[19:33:24][V][component:205]: Components should block for at most 20-30ms.
[19:33:27][D][esp32.preferences:114]: Saving 1 preferences to flash...
[19:33:27][V][esp32.preferences:137]: NVS data not changed skipping 2658603775  len=1
[19:33:27][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[19:33:32][D][media_player:059]: 'Media Player' - Setting
[19:33:32][D][media_player:066]:   Media URL: http://192.168.1.12:8123/media/local/1.01%20WORLD%20ORDER.mp3?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI0ODFkNjU4ZjU1YTY0NTdlYjc5N2UwMDc5YWVkNmFiNyIsInBhdGgiOiIvbWVkaWEvbG9jYWwvMS4wMSBXT1JMRCBPUkRFUi5tcDMiLCJwYXJhbXMiOltdLCJpYXQiOjE2ODQ2OTA0MTIsImV4cCI6MTY4NDc3NjgxMn0.Nm0ITM0vXYOLpRwS66tFbXlS-On2GSlByRl9duMbnlM
WARNING lyratd.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for lyratd.local
WARNING Disconnected from API
INFO Successfully connected to lyratd.local
```

Kosygor, 2023-06-17:
I have experienced similar issue. [Fixing HA URL helped me](https://github.com/esphome/issues/issues/3840#issuecomment-1595719087).

kizovinh, 2023-12-06:
I got the same issue, but for me sometime the speaker work normally, sometime it play only after pipeline end, and most of the time it wont play anything