esphome/issues #4088: I2S audio not working after update ESPhome firmware

Manhem, 2023-01-27:
### The problem

After I update ESPhome firmware I lost the opportunity to use my Mediaplayer I build my own.
It's a ESP32 with UDA1443 using I2S.
The description I found at your place.

I can see the WiFi level.
But if I try to play a file, it starts and directly closes.
I have installed Music Assistant and here I can't find the player.
Home Assistants Media can find it but I can't play anything.
If I take a closer look the player is idle all the time.

Maybe I have put in the wrong board, but I have tried some different ones with the same result.
The old one I used was not working so I also try others.

What am I doing wrong or maybe a bug..

You can find a picture over my board down here.

### Which version of ESPHome has the issue?

2022.12.xx

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.xx

### What platform are you using?

ESP32

### Board

ESP32

### Component causing the issue

I2S ??

### Example YAML snippet

```yaml
substitutions:
  devicename: "mediaplayer-1"
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
#  wifi_domain: !secret wifi_domain
#  api_password: !secret src/api_password
#  ota_password: !secret src/ota_password

esphome:
  name: ${devicename}

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
#  encryption:
#    key: "SP48vTwHlVxH4/gTVZd+YD4Q2euRBv7ND59+2bIRVfM="

ota:
  password: "31f579f71d90cf86945feb3b3aa678a5"
#  password: "08c3e7ca9a3440d528798b200e1e6081"

wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_password}

  fast_connect: true

  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.0.33
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.0.253
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Mediaplayer-1 Fallback Hotspot"
    password: "oGrIyPQJ692v"

captive_portal:

# Example configuration entry
media_player:
  - platform: i2s_audio
    name: Media Player
    dac_type: external
    i2s_lrclk_pin: GPIO33
    i2s_dout_pin: GPIO22
    i2s_bclk_pin: GPIO19
    mode: mono

sensor:
  - platform: wifi_signal
    name: "${devicename} Wi-Fi Signal"
    update_interval: 60s

text_sensor:
  - platform: version
    name: "${devicename} ESPhome version"
```

### Anything in the logs that might be useful for us?

```txt
Media Player changed to Idle triggered by service media_player.play_media
10:10:19 AM - 3 seconds ago - xxxxx
Media Player changed to Playing triggered by service media_player.play_media
```

### Additional information

[image]

duceduc, 2023-02-05:
Hello. I am having same issue. I was able to play mp3 files and the log shows that it is playing; however, there is no audio coming out of the speaker. Creating a tts message still works. Just unable to play mp3 files. I suspect it is one of the updates I but not sure which ones. I believe the last known working version was 2022.12.0.

Did you managed to get it working??

randybb, 2023-02-05:
@duceduc  are you sure that volume is not on 0? I have seen such problem... Anyway, my m5echo works just fine.

duceduc, 2023-02-05:
Yes, volume is at max. I can hear the tts message through the speaker, just not mp3 files. The url that outputs in the log I can play in the browser. So the url is valid.

Manhem, 2023-02-07:
It seems that the problem comes when I sending out the file to the device.
Then I get this:
[ssl_client.cpp:132] start_ssl_client(): socket error on fd 52, errno: 104, "Connection reset by peer"
[WiFiClientSecure.cpp:135] connect(): start_ssl_client: -1

So when it starts to play, there is nothing to play.

I have tried everything so I can't say exactly what version this happend, because I don't remember.

duceduc, 2023-02-07:
You are saying you never got it to work??
This is the log when I try to play an mp3 files. The url below does play from the pc browser.

    [17:36:47][D][media_player:059]: 'razer-leviathan-player' - Setting
    [17:36:47][D][media_player:066]:   Media URL: http://192.168.1.20:8123/media/local/music/Ray%20Charles%20-%20Ellie,%20My%20Love.mp3?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI3NjAyMWU5Njg3ZDQ0OWYzOTAwYjE2MmM1NGIyNTgwYyIsInBhdGgiOiIvbWVkaWEvbG9jYWwvbXVzaWMvUmF5IENoYXJsZXMgLSBFbGxpZSwgTXkgTG92ZS5tcDMiLCJwYXJhbXMiOnt9LCJpYXQiOjE2NzU3NTkwMDcsImV4cCI6MTY3NTg0NTQwN30.ODU3TOApCls2dXJftpp2HlnJQrMv5GkKwjOObo2x8Jg
    [17:36:47][D][main:317]: audio_info: Connect to new host: "http://192.168.1.20:8123/media/local/music/Ray%20Charles%20-%20Ellie,%20My%20Love.mp3?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI3NjAyMWU5Njg3ZDQ0OWYzOTAwYjE2MmM1NGIyNTgwYyIsInBhdGgiOiIvbWVkaWEvbG9jYWwvbXVzaWMvUmF5IENoYXJsZXMgLSBFbGxpZSwgTXkgTG92ZS5tcDMiLCJwYXJhbXMiOnt9LCJpYXQiOjE2NzU3NTkwMDcsImV4cCI6MTY3NTg0NTQwN30.ODU3TOApCls2dXJftpp2HlnJQrMv5GkKwjOObo2x8Jg"
    [17:36:47][D][main:317]: audio_info: buffers freed, free Heap: 174588 bytes
    [17:36:47][D][main:317]: audio_info: Connection has been established in 15 ms, free Heap: 173804 bytes
    [17:36:47][D][main:229]: Playback started!
    [17:36:47][D][main:317]: audio_info: MP3Decoder has been initialized, free Heap: 144604 bytes
    [17:36:47][D][main:317]: audio_info: stream ready
    buffer filled in 17816 ms
    [17:36:47][D][main:317]: audio_info: ID3 framesSize: 192461
    [17:36:47][D][main:317]: audio_info: ID3 version: 2.3
    [17:36:47][D][main:317]: audio_info: ID3 normal frames

Manhem, 2023-02-07:
It has been working in an older version of ESPhome, but I don't remember which one.
If I play the mp3 file in the browser it works, so the URL is ok.

Manhem, 2023-03-24:
How can I tell ESPhome to use this old version 2.1.0?
Is there a file anywhere that I shall change some stuff?

duceduc, 2023-03-24:
@Manhem From my understanding, there used to be a feature where we can specify what version to install. It was removed because it was too buggy.

Penty79, 2023-03-25:
In mi case I installed esphome  in my laptop with ubuntu  using python pip and  it´s  easy to change version:
pip3 install esphome==2022.11.5  this is  to downgrade to the last version that is working for me.

You cant tell which library is using looking the compilation log :

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- ESP32-audioI2S @ 2.0.6

Here are  they: 
The last version that is working for me 
2022.11.5  -> ESP32-audioI2S @ 2.1.0 
 
The first that is not working 
2022.12.0 -> ESP32-audioI2S @ 2.0.6

elik745i, 2023-05-13:
Same here used to play MP3s but after update can only play radio, nothing else not even text to speach

dwd1, 2024-03-20:
Esphome mediaplayer cant play anything from https.
https://github.com/esphome/feature-requests/issues/2179
Nobody cares about the solution.

Rokoel, 2024-10-10:
I'm experiencing a similar (or the same?) issue and it looks like the network setup is to blame. I'm using ZeroTier both to access my Home Assistant instance from elsewhere and to act as a proxy for Home Assistant (and, thus, for ESPHome). It looks like ESP tries to access the URL of the Home Assistant, but the ESP itself isn't connected to the Proxy, so it tries to access the remote server's IP address, but fails to do that. It seems like that's the case, though. I'm not sure about why exactly can we listen to radio, but I assume it has to do something with how accessing radio stations is done differently from accessing files on home assistant (?)

I haven't been able to fix the issue yet, but if I get there, I'll post an update. 

Another question is was anything related to networking (probably specifically DNS-related?) changed in the update? Because I think my setup has been working previously but after some updates it stopped.