esphome/issues #4089: Home Assistant is asking me to update to ESPhome 2022.12.4 but docker stable is only at 2022.12.3

jmlc11, 2023-01-27:
### The problem

Home assistant:
[image]
ESPhome:
[image]
In docker compose:
[image]

### Which version of ESPHome has the issue?

Docker stable (2022.12.3)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP8266

### Board

esp-wrover-kit

### Example YAML snippet

```yaml
substitutions:
  name: btproxy
  board: esp-wrover-kit

esphome:
  name: ${name}

esp32:
  board: ${board}
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Livingroomtrack Fallback Hotspot"
    password: !secret password

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret password

ota:
  password: !secret password

esp32_ble_tracker:

bluetooth_proxy:
  active: True
```

randybb, 2023-01-27:
Just changed it in my container, then `docker pull esphome/esphome:latest`, restarted the container and it is showing 2022.12.8. No issue here.

jmlc11, 2023-01-27:
I did that like 10 or more times without success, then i tried :dev and that worked until today's update which gave a bunch of compiling errors. 

Now i deleted the docker folder and pulled again and worked. Why it worked after this comment? no idea!

closed 2023-01-27 by jmlc11