esphome/issues #4090: WEMOS S2 mini suddenly no longer possible to upload over  USB

ferreto1978, 2023-01-27:
### The problem

Begin January 2023 I uploaded 5 new WEMOS S2 mini boards (Lolin v1) using ESPHome over USB via the "manual" mode. That alml worked fine, as long as I put the board in boot mode (pressing the boot button on startup).
But suddenly, one day that didn't wan't to work anymore. I tried this on several different boards, on 3 different PC's and with 2 differnt browsers (Chrome and Brave). The strange part is that the previously configured boards can still be updated over WiFi, but not over USB. Even when re-installing the drivers, no luck.
Using the WEMOS D1 mini, that still works, and even older ESP32 and ESP8266 modules using various USB 2 serial drivers all turn out te be successfull.
It seems to still talk to the WEMOS S2 mini in some way though, because if I don't start it in the correct mode, it automatically resets it to the right mode (noticed this when I suddenly released the COM port changed). So up to that part it still seems to do "something".
With the Arduino IDE, I can upload any code to any of those boards, so the boards themselves seem to be working fine.
Browsing through the forums I learned about the ESPHome-Flasher (v1.4.0) but even that generates an error.
Clearly something was broken in an update specifically for the WEMOS S2 mini boards.

### Which version of ESPHome has the issue?

Latest version (since 23nd half january 2023)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Several version tried, all same result

### What platform are you using?

ESP32

### Board

WEMOS S2 mini

### Component causing the issue

ESPHome web ; ESPHome Flasher

### Example YAML snippet

```yaml
esphome:
  name: test-wemoss2mini

esp32:
  board: lolin_s2_mini
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "***"

ota:
  password: "***"

wifi:
  ssid: ***
  password: ***

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "***"
    password: "***"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
Not sure how to get to those logs, but using the ESPHome-Flasher, this is the output:
Using 'COM3' as serial port.
Connecting...
Unexpected error: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
```

### Additional information

With the Arduino IDE, I can upload any code to any of those boards, so the boards themselves and the drivers seem to be working fine.

ferreto1978, 2023-02-18:
A few updates later, still issues with the Wemos S2 mini

ferreto1978, 2023-03-23:
March 23, again a few updates later, still issues with the Wemos S2 mini.
Doen anyone else have this issue?

nielsnl68, 2023-03-23:
I have the same board's and uploading local via USB.
What i see is that it tries first on high speed and false ad and then on a lower speed and that works.
I would suggest to set:
```yaml
# Example configuration entry
esphome:
  # ...
  platformio_options:
    upload_speed: 115200
```
and see of that will fix your issue.

ferreto1978, 2023-03-24:
No matter what speed I set, I keep getting this popup:
Failed to initialize. Try resetting your device or holding the BOOT button while selecting your serial port until it starts preparing the installation.
And that happens with every S2 mini board, all other boards work fine.
The ones I installed months ago, I can still update over wireless, but the ones I need to initialize (over USB), just keep failing.

PeerOlli, 2023-04-02:
Hello, I also recently got a Lolin s2 mini that I wanted to use with Esphome. I can only flash it with the Flashtool, but I have another problem....
I've tried different settings and just can't get the WiFi to work. Can someone tell me how to configure an S2 mini in Esphome for this to work?

HeathersZen, 2023-04-17:
I'm experiencing a similar issue. I've got a bunch of lolin-s2-minis that will now no longer program after programming successfully several times. I'm suspecting it is because of something that VSCode is doing. I've noted the following sequence of events:

- Using a new device, I can program it using Arduino IDE. It comes up fine on COM12 and is recognized as a LOLIN S2 MINI. 
- Then I tried to flash it using VSCode, and fails, and also it no longer comes up in either VSCode or Arduino as before. 
- It now is read as a "ESP32S2 Dev Module" on COM7 in Arduino and VSCode.
- When I try to program it, it fails, stating "A fatal error occurred: Could not open COM7, the port doesn't exist
Failed uploading: uploading error: exit status 2". I get this same behavior from both VSCode & Arduino IDE.

I'm staying away from VSCode for a minute. I'm running out of these things quick!

I'm guessing that VSCode is doing something to the bootloader? I honestly don't know. I'm pretty new to all this.

ferreto1978, 2023-04-19:
Does your boot button still work properly?
I too had 1 board where I could not upload anything in Arduino either because it didn't want to switch modes.
It "should" do it automatically to some point, but it doesn't always seem to happen.

lazyoldjack, 2023-04-23:
I have the same problem with an ESPS2 (it's an S2 Mini clone). I programmed it a few times OK, then I programmed with some code using the TinyUSB library and some code for MIDI and although it says it's programming OK, the program never seems to run. I'm going to try programming it with an external USB interface...

elik745i, 2023-05-12:
press both buttons and release programming button, s2 mini set for programming. ESPHome messes up its autoprogramming mode for some reason.
I got bunch of other issues: i2s distorted sound, on other board it cant connect to wifi.

ferreto1978, 2023-05-13:
> press both buttons and release programming button, s2 mini set for programming. ESPHome messes up its autoprogramming mode for some reason.

Tried that too, even with timing the pushbuttons in any combination I could think of. But even if I entirely skip this step, it does eventually end up in program mode (you can notice by the change of com port). However, the outcome is the same, it just won't upload.

> I got bunch of other issues: i2s distorted sound, on other board it cant connect to wifi.

I can't say I have experienced problems with WiFi. I only used I2C for BME280, BME680, EP1, and displays, and also here I can't say I experienced issues. Haven't tried sound though, so can't confirm I,m afraid.

thundo, 2023-12-15:
Same issue...

> board: lolin_s2_mini;
> framework: arduino; 
> platform: platformio/espressif32@6.4.0

Test #1
Flash via server port
> **LOLIN-S2-MINI - Tiny USB CDC**
> /dev/tty/ACM0

On uploading I get

```
INFO Successfully compiled program.
esptool.py v4.6.2
Serial port /dev/ttyACM0
Connecting..............
ERROR Running command failed: Could not configure port: (5, 'Input/output error')
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 115200 --port /dev/ttyACM0 --chip esp32s2 write_flash -z --flash_size detect 0x10000 /data/build/esp-test/.pioenvs/esp-test/firmware.bin 0x1000 /data/build/esp-test/.pioenvs/esp-test/bootloader.bin 0x8000 /data/build/esp-test/.pioenvs/esp-test/partitions.bin 0xe000 /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/partitions/boot_app0.bin locally.
INFO Upload with baud rate 115200 failed. Trying again with baud rate 115200.
esptool.py v4.6.2
Serial port /dev/ttyACM0
ERROR Running command failed: Could not open /dev/ttyACM0, the port doesn't exist
ERROR Please try running esptool.py --before default_reset --after hard_reset --baud 115200 --port /dev/ttyACM0 --chip esp32s2 write_flash -z --flash_size detect 0x10000 /data/build/esp-test/.pioenvs/esp-test/firmware.bin 0x1000 /data/build/esp-test/.pioenvs/esp-test/bootloader.bin 0x8000 /data/build/esp-test/.pioenvs/esp-test/partitions.bin 0xe000 /data/cache/platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/partitions/boot_app0.bin locally.
```

Test #2 (or simply hit "retry" previously)
Flash via server port
> **ESP32-S2**
> /dev/tty/ACM0

On uploading I get

```
INFO Successfully compiled program.
esptool.py v4.6.2
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 48:27:e2:5d:b9:5e
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00010000 to 0x00104fff...
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Compressed 999440 bytes to 666724...
Wrote 999440 bytes (666724 compressed) at 0x00010000 in 7.4 seconds (effective 1074.3 kbit/s)...
Hash of data verified.
Compressed 13552 bytes to 9525...
Wrote 13552 bytes (9525 compressed) at 0x00001000 in 0.2 seconds (effective 610.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 655.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 798.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200
ERROR Serial port closed!
```

dkebler, 2024-01-06:
see my comment here.  https://github.com/esphome/esp-web-tools/issues/338#issuecomment-1879862084

I think the webtool is doing a hard reset by default "before" and that takes the s2 out of program mode.  I see to no option to disable that and thus I am forced to use esptool.py at the cli.

thepuncake, 2024-06-09:
I seem to have the same problem. After uploading some wled custom build in vscode I'm no longer able to upload anything to the s2 mini, adafruit web tool manages to upload the esphome bin file but it doesn't connect to wifi afterwards.

dkebler, 2024-06-09:
@thepuncake  assuming you are using platformio with your wled build then try exactly what I do.  

https://github.com/esphome/issues/issues/4090#issuecomment-2156283666

Pay attention to what I say in point 2 about releasing the buttons at the right time.   Otherwise follow the advice about "fixing" the board which is not really necessary if you only intend to flash once (or rarely).

EmileSpecialProducts, 2025-04-11:
Yes , the ESP32S2 realy needs the updated [ESPTool-JS](https://github.com/espressif/esptool-js) to solve this.
Or use a older version 8.0.6 version 
This is related the [ESPtool-JS#135](https://github.com/espressif/esptool-js/issues/135)
I have also had some luke when first [erasing](https://espressif.github.io/esptool-js/) the ESP32-S2

thepuncake, 2025-04-11:
There is an easy workaround for this for anyone who's struggling. Upload tasmota on it via the web installer, and then upload custom bin file through the tasmota ui. Works for me every time.

mafrosis, 2025-04-22:
Agreed these boards are awful. I have a whole bunch of Lolin's across all SOCs, and it's just the S2 which misbehaves for flashing.

It's essential to press both buttons and release `RST` first, to get the board into flash mode. However, although this has certainly worked once in the past, I cannot get it to work today ¯\_(ツ)_/¯ 

```
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
INFO Upload with baud rate 115200 failed. Trying again with baud rate 115200.
```