esphome/issues #4091: bluetooth_proxy: Message Data for BluetoothGATTWriteRequest seems corrupted

tbe, 2023-01-28:
### The problem

While debugging some connection issues with a Home Assistant Bluetooth integration, i found some strange behavior.

The message send by `write_gatt_char` from the HA site is: `'\x0c\x97\xfa\x87\xdc\xe8\xe9\xb3\x81\xe6\xba%\xa7^\x1d\x1f\x11'`

But the debug log from the ESP device contains:

```
[03:00:34][VV][api.service:742]: on_bluetooth_gatt_write_request: BluetoothGATTWriteRequest {
  address: XXXXXX
  handle: 27
  response: NO
  data: '
         \x97\xfa\x87\xdc\xe8鳁\xe6\xba%\xa7^'
}
```

This message seems to lack some data after '\xa7^'.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

bluetooth_proxy

alistair23, 2024-01-14:
I'm seeing similar behaviour as well

bdraco, 2024-01-15:
Your terminal is trying to render the raw output as utf-8 when its not.  Try setting your terminal to ascii to see the contents of the message. Than verify if the data is missing or not

alistair23, 2024-01-16:
I'm using the "Logs" from the ESPHome view in HA. I don't see how to change it to ASCII

bdraco, 2024-01-16:
You'll need to use a terminal application with the cli

SasStu, 2024-01-16:
I don't know if it's the same problem, but I think it's related: After the update to 2023.12.6 my Everything Presence One devices weren't working anymore (HA entities unavailable). After reading this issue I removed the Bluetooth Proxy configuration, and they are now working again.