esphome/issues #4094: Dashboard: "UPDATE AVAILBLE" still shown after manual update

raoulbhatia, 2023-01-28:
### The problem

I am running 2 GL.iNet GL-S10 that cannot be updated via esphome running on RPi in a Docker environment, see #3076

Thus, I am updating these devices via my laptop.

After the update, the Dashboard on my RPi still shows "UPDATE AVAILBLE" even as the devices run the latest version.

ESPHome Dashboard version:
[image]

"UPDATE AVAILBLE" message for GL-S10
[image]

GL-S10 logs that indicate matching version

[image]

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1.7 (not relevnat)

### What platform are you using?

ESP32-IDF

DAVe3283, 2023-02-03:
I believe the update status is not pulled from the device, but written locally when the compile is complete. So if you want to update a device and choose to download the firmware instead of OTA flash, it will show as updated even when it isn't.

So the "update available" notice is misleading at best, IMO.

If you just want it to show as up to date, edit the file `/config/esphome/.esphome/gl-s10-bt-proxy.yaml.json` and put the latest version in the `esphome_version` field.

It would be nice if the actual running firmware version was queried and stored to this file, rather than the compiler optimistically writing it when a binary is produced. Would be much more accurate.

ssieb, 2023-03-18:
It's possible for the dashboard to get the version now: https://github.com/esphome/esphome/pull/4429
But up to this point, the dashboard has been very passive, so this is a larger change.