esphome/issues #4095: 'esp_timer_get_time' was not declared in this scope;

hunasdf, 2023-01-28:
### The problem

Error while complie code

### Which version of ESPHome has the issue?

Version: 2022.9.4

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3

### Component causing the issue

framework/platform/esp-idf

### Example YAML snippet

```yaml
esp32:
  board: esp32s3box
  variant: esp32s3
  framework:
    type: esp-idf
    version: 5.0.0
    platform_version: 6.0.0
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration ./config/test.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
********************************************************************************
Obsolete PIO Core v6.0.2 is used (previous was 6.1.6)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
********************************************************************************
Processing test (board: esp32s3box; framework: espidf; platform: platformio/espressif32 @ 6.0.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
 - framework-espidf @ 3.50000.0 (5.0.0) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-riscv32-esp @ 11.2.0+2022r1 
 - toolchain-xtensa-esp32s3 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- lvgl @ 8.3.4
Compiling .pioenvs/test/src/esphome/components/esp32/core.o
Compiling .pioenvs/test/src/esphome/core/color.o
Compiling .pioenvs/test/src/esphome/core/component.o
Compiling .pioenvs/test/src/esphome/core/component_iterator.o
src/esphome/components/esp32/core.cpp: In function 'uint32_t esphome::millis()':
src/esphome/components/esp32/core.cpp:26:53: error: 'esp_timer_get_time' was not declared in this scope; did you mean 'timer_gettime'?
   26 | uint32_t IRAM_ATTR HOT millis() { return (uint32_t)(esp_timer_get_time() / 1000ULL); }
      |                                                     ^~~~~~~~~~~~~~~~~~
      |                                                     timer_gettime
src/esphome/components/esp32/core.cpp: In function 'uint32_t esphome::micros()':
src/esphome/components/esp32/core.cpp:28:53: error: 'esp_timer_get_time' was not declared in this scope; did you mean 'timer_gettime'?
   28 | uint32_t IRAM_ATTR HOT micros() { return (uint32_t) esp_timer_get_time(); }
      |                                                     ^~~~~~~~~~~~~~~~~~
      |                                                     timer_gettime
In file included from src/esphome/components/esp32/core.cpp:13:
src/esphome/components/esp32/core.cpp: In function 'uint32_t esphome::arch_get_cpu_cycle_count()':
/home/waddle/.platformio/packages/framework-espidf/components/esp_hw_support/include/hal/cpu_hal.h:49:63: warning: 'uint32_t cpu_ll_get_cycle_count()' is deprecated [-Wdeprecated-declarations]
   49 | #define cpu_hal_get_cycle_count()       cpu_ll_get_cycle_count()
      |                                         ~~~~~~~~~~~~~~~~~~~~~~^~
src/esphome/components/esp32/core.cpp:59:10: note: in expansion of macro 'cpu_hal_get_cycle_count'
   59 |   return cpu_hal_get_cycle_count();
      |          ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/waddle/.platformio/packages/framework-espidf/components/esp_hw_support/include/hal/cpu_hal.h:16,
                 from src/esphome/components/esp32/core.cpp:13:
/home/waddle/.platformio/packages/framework-espidf/components/esp_hw_support/include/hal/cpu_ll.h:28:56: note: declared here
   28 | FORCE_INLINE_ATTR __attribute__((deprecated)) uint32_t cpu_ll_get_cycle_count(void)
      |                                                        ^~~~~~~~~~~~~~~~~~~~~~
In file included from src/esphome/core/color.cpp:1:
src/esphome/core/color.h: In member function 'esphome::Color esphome::Color::operator+(const esphome::Color&) const':
src/esphome/core/color.h:116:12: warning: implicitly-declared 'constexpr esphome::Color::Color(const esphome::Color&)' is deprecated [-Wdeprecated-copy]
  116 |     return ret;
      |            ^~~
src/esphome/core/color.h:61:17: note: because 'esphome::Color' has user-provided 'esphome::Color& esphome::Color::operator=(const esphome::Color&)'
   61 |   inline Color &operator=(const Color &rhs) ALWAYS_INLINE {  // NOLINT
      |                 ^~~~~~~~
src/esphome/core/color.h: In member function 'esphome::Color esphome::Color::operator-(const esphome::Color&) const':
src/esphome/core/color.h:139:12: warning: implicitly-declared 'constexpr esphome::Color::Color(const esphome::Color&)' is deprecated [-Wdeprecated-copy]
  139 |     return ret;
      |            ^~~
src/esphome/core/color.h:61:17: note: because 'esphome::Color' has user-provided 'esphome::Color& esphome::Color::operator=(const esphome::Color&)'
   61 |   inline Color &operator=(const Color &rhs) ALWAYS_INLINE {  // NOLINT
      |                 ^~~~~~~~
src/esphome/core/color.h: In member function 'esphome::Color esphome::Color::gradient(const esphome::Color&, uint8_t)':
src/esphome/core/color.h:164:12: warning: implicitly-declared 'constexpr esphome::Color::Color(const esphome::Color&)' is deprecated [-Wdeprecated-copy]
  164 |     return new_color;
      |            ^~~~~~~~~
src/esphome/core/color.h:61:17: note: because 'esphome::Color' has user-provided 'esphome::Color& esphome::Color::operator=(const esphome::Color&)'
   61 |   inline Color &operator=(const Color &rhs) ALWAYS_INLINE {  // NOLINT
      |                 ^~~~~~~~
*** [.pioenvs/test/src/esphome/components/esp32/core.o] Error 1
========================== [FAILED] Took 8.79 seconds ==========================
```

sekt1953, 2023-06-22:
I se same problem with an NSPANEL

clydebarrow, 2024-08-22:
That version of esp-idf is not usable with ESPHome. Use 
```
  framework:
    type: esp-idf
    version: 5.1.2
    platform_version: 6.5.0
```

closed 2024-08-22 by clydebarrow