esphome/issues #4097: Ultrasonic Sensor Stopped Working (2022.12.x)

broyuken, 2023-01-30:
### The problem

I have an HC-SR04 that I use to determine if there is a car in the garage or not. After upgrading ESPHome to 2022.12.x, it no longer reports correct values. No matter if there is something close or far from it, it always reports .4 inches away. I have tried this with 2 different ESP boards, and 2 different HC-SR04's. They all behave the same way. I have tried multiple usb cables, had it plugged into my desktop, laptop and the wall, and the same behavior exists. I had an old one that was working as well, updated it and that too stopped working. Is there a change I need to make to my config to get this working again? Perhaps something with the new filesystem doesn't like the ultrasonic sensors?

### Which version of ESPHome has the issue?

2022.12.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP32

### Board

hiletgo esp32

### Component causing the issue

ultrasonic

### Example YAML snippet

```yaml
esphome:
  name: garage-car-presence

esp32:
  board: esp32dev
  framework:
    type: arduino

# WiFi connection, correct these
# with values for your WiFi.
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pwd
  use_address: 10.0.86.82

# Enable Bluetooth Proxy
bluetooth_proxy:
  active: true

# Enable logging
logger:

# Enable Home Assistant API
api:

# Enable over-the-air updates.
ota:

# Enable Web server.
web_server:
  port: 80
  
# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: Car_position_sensor ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: Car_position_sensor IP
    ssid:
      name: Car_position_sensor SSID
    bssid:
      name: Car_position_sensor BSSID
  - platform: template
    name: "Left Garage Occupied"
    id: left_occupied
    lambda: |-
      if (id(rawdist).state > .5) {
        // Left Garage is Empty.
        id(left_occupied).set_icon("mdi:car-off");
        return {"Empty"};
      } else {
        // Left Garage is Occupied.
        id(left_occupied).set_icon("mdi:car");
        return {"Car Detected"};
      }
# Exposed switches.
# Switch to restart the Car_position_sensor.   
switch:
  - platform: restart
    name: "Car_position_sensor Restart"

sensor:
  # Uptime sensor.
  - platform: uptime
    name: Car_position_sensor Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: Car_position_sensor WiFi Signal
    update_interval: 60s

  - platform: ultrasonic
    id: rawdist
    trigger_pin: GPIO5
    echo_pin: GPIO18
    name: "Left Garage Car Distance Raw"
    update_interval: 1s
    timeout: 4.0m

  - platform: template
    name: "Left Garage Car Distance Inches"
    update_interval: 1s
    lambda: return id(rawdist).state * 39.3701;

  - platform: template
    name: "Left Garage Car Distance Feet"
    update_interval: 1s
    lambda: return id(rawdist).state * 3.28084;
```

DAVe3283, 2023-02-03:
Can you turn on verbose logging and post a log, see what the sensor is actually reporting.

```yaml
logger:
  level: VERBOSE
  logs:
    # Reduce verbosity of stuff we don't care about
    app: DEBUG
    sensor: DEBUG
```

Additionally, if you revert back to an older ESPHome version, does it work again? It is strange because the ultrasonic code hasn't been touched in almost a year.

Finally, have you double (or triple) checked the wiring matches your YAML file? Maybe you made a change a long time ago, but never flashed it, and now with the new version a previous change took effect and that was the actual problem.

broyuken, 2023-02-04:
How do you roll back to an older esphome version? I tried restoring a backup and it wouldn’t install. Something about not being able to delete an envs file.

DAVe3283, 2023-02-05:
Restoring a backup is supposed to work, and I think is the only officially supported way.

You can search the forums and see various hacky ways people have had success with, but fixing the issue restoring your backup might be the way to go.

broyuken, 2023-02-06:
I installed esphome 2022.11.3 and built a brand new bin with a brand new esp32 and a brand new ultrasonic sensor, with brand new wires. It's still not working. Is there something I did wrong with my code?

```
esphome:
  name: ultrasonic

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: "broyuken_IoT"
  password: "BRoy51cKn3554686"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ultrasonic Fallback Hotspot"
    password: "J0QdbLjnYiuo"

captive_portal:

web_server:
  port: 80

sensor:
  - platform: ultrasonic
    id: rawdist
    trigger_pin: GPIO5
    echo_pin: GPIO18
    name: "Test Distance"
    update_interval: 5s

text_sensor:
    # Expose ESPHome version as sensor.
  - platform: version
    name: Car_position_sensor ESPHome Version
    #         # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: Car_position_sensor IP
```

broyuken, 2023-02-06:
I'm dumb, I had the sensor plugged into 3.3, plugged it into 5v and it's good now.

closed 2023-02-06 by broyuken