esphome/issues #4098: ili9341 display is white after updating to version 2022.12.0

Yaro69PL, 2023-01-30:
### The problem

After updating to version 2012.12.0, the display based on ILI9341 stopped displaying the content and lights up completely white. The method of connecting the LED pin does not matter, i.e. through 3.3 kOhm to 3.3 V or to PWM.

Sorry for my English, I'm translating through a translator.

### Which version of ESPHome has the issue?

ESPHome 2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.01.1

### What platform are you using?

ESP32

### Board

ESP32 WROOM 32

### Component causing the issue

display

### Example YAML snippet

```yaml
display:
  - platform: ili9341
    model: TFT 2.4
    spi_id: spi1
    cs_pin: GPIO16
    dc_pin: GPIO05
    led_pin: GPIO04
    reset_pin: GPIO17
    rotation: 90
    lambda: |-
      it.fill(black);
      it.strftime(160, 10, id(font4), id(blue), TextAlign::TOP_CENTER, "%H:%M:%S", id(sntp_time).now());
      if (id(biuro_temperature).has_state()) {
```

nielsnl68, 2023-01-31:
Could you try your setup with my ili9xxx PR at:

```yanl
psram:

external_components:
  - source: github://pr#4275
    components: [ili9xxx]
    refresh: always
```

Yaro69PL, 2023-01-31:
Unfortunately, no changes.

Below is some log

[15:38:46][C][logger:293]: Logger:
[15:38:46][C][logger:294]:   Level: VERBOSE
[15:38:46][C][logger:295]:   Log Baud Rate: 115200
[15:38:46][C][logger:296]:   Hardware UART: UART0
[15:38:46][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:46][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:46][V][component:201]: Components should block for at most 20-30ms.
[15:38:46][C][spi:101]: SPI bus:
[15:38:46][C][spi:102]:   CLK Pin: GPIO19
[15:38:46][C][spi:104]:   MOSI Pin: GPIO18
[15:38:46][C][spi:106]:   Using HW SPI: YES
[15:38:46][C][ili9xxx:051]: ili9xxx
[15:38:46][C][ili9xxx:051]:   Rotations: 90 °
[15:38:46][C][ili9xxx:051]:   Dimensions: 320px x 240px
[15:38:46][C][ili9xxx:060]:   Color mode: 8bit 332 mode
[15:38:46][C][ili9xxx:064]:   Reset Pin: GPIO17
[15:38:46][C][ili9xxx:065]:   DC Pin: GPIO5
[15:38:46][C][ili9xxx:067]:   backlight Pin: GPIO4
[15:38:46][C][ili9xxx:071]:   Update Interval: 1.0s
[15:38:46][C][psram:020]: PSRAM:
[15:38:46][C][psram:021]:   Available: NO
[15:38:46][C][mdns:103]: mDNS:
[15:38:46][C][mdns:104]:   Hostname: biuro-kwadrat
[15:38:46][V][mdns:105]:   Services:
[15:38:46][V][mdns:107]:   - _esphomelib, _tcp, 6053
[15:38:46][V][mdns:109]:     TXT: version = 2022.12.8
[15:38:46][V][mdns:109]:     TXT: mac = c049efcd68d4
[15:38:46][V][mdns:109]:     TXT: platform = ESP32
[15:38:46][V][mdns:109]:     TXT: board = esp32dev
[15:38:46][V][mdns:109]:     TXT: network = wifi
[15:38:46][C][ota:093]: Over-The-Air Updates:
[15:38:46][C][ota:094]:   Address: biuro-kwadrat.local:3232
[15:38:46][C][ota:097]:   Using Password.
[15:38:46][C][api:138]: API Server:
[15:38:46][C][api:139]:   Address: biuro-kwadrat.local:6053
[15:38:46][C][api:143]:   Using noise encryption: NO
[15:38:47][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:47][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:47][V][component:201]: Components should block for at most 20-30ms.
[15:38:48][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:48][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:48][V][component:201]: Components should block for at most 20-30ms.
[15:38:49][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:49][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:49][V][component:201]: Components should block for at most 20-30ms.
[15:38:50][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:50][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:50][V][component:201]: Components should block for at most 20-30ms.
[15:38:51][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:51][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:51][V][component:201]: Components should block for at most 20-30ms.
[15:38:52][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:52][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:52][V][component:201]: Components should block for at most 20-30ms.
[15:38:53][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:53][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:53][V][component:201]: Components should block for at most 20-30ms.
[15:38:54][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:54][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:54][V][component:201]: Components should block for at most 20-30ms.
[15:38:55][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:55][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:55][V][component:201]: Components should block for at most 20-30ms.
[15:38:56][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)
[15:38:56][V][component:200]: Component ili9xxx.display took a long time for an operation (0.25 s).
[15:38:56][V][component:201]: Components should block for at most 20-30ms.
[15:38:57][V][ili9xxx:175]: Start display(xlow:0, ylow:0, xhigh:239, yhigh:319, width:240, heigth:320, start_pos:0)

Yaro69PL, 2023-02-01:
I solved.
Problem with incorrectly assigned PINs for SPI.
It didn't bother me until version 2022.12.0 after the changes did.
Thank you for your help.

closed 2023-02-01 by Yaro69PL