esphome/issues #4100: Editor closes unsaved file after a while

RubenKelevra, 2023-01-30:
### The problem

The yaml-editor discards the changes made and closes the editor window after a while:

#### How to reproduce:

- Open the editor for a yaml of a device
- Switch to a different tab in the browser
- Switch back after something like 10 minutes to the ESPHome tab

#### What happens:

The page will refresh and the editor is closed. All changes are discarded that way.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.6

### What platform are you using?

ESP8266

### Board

does not matter

### Component causing the issue

Web interface/yaml editor

### Additional information

I open the ESPHome page via a side-bar entry inside HA.

Browser I use is Firefox 109.0.

DAVe3283, 2023-02-03:
This is not an ESPHome issue, it is a Home Assistant issue. Luckily, there is a setting to disable this behavior:

1. Open Home Assistant.
2. Click your profile image (bottom left).
3. Turn off `Automatically close connection`.

If you don't see `Automatically close connection`, enable `Advanced Mode` and refresh the page.
If you don't see `Advanced Mode`, you need to be using an admin user.

RubenKelevra, 2023-02-20:
Man, that's a weird “feature” :/

Thanks for the help!

I wonder if add-ons could get a way to override this behavior for their views. Say you start an 'update all', log output, or editor and can disable this feature to avoid data-loss.