esphome/issues #4101: Issue with for condition

zoic21, 2023-01-31:
### The problem

Hello,

I have an issue with for condition seem not work correctly (in my case at least). Simply for condition do nothing, You can see in my code I do test without for and with for and with for condition is never valid.

Thank in advance for your help

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32

### Board

Olimex ESP32-GATEWAY

### Component causing the issue

pulse_meter

### Example YAML snippet

```yaml
substitutions:
  #water_max_duration : "10s"
  gas_max_duration : "20min"

esphome:
  name: esp-energy

esp32:
  board: esp32dev
  framework:
    type: arduino

globals:
   - id: water_alert
     type: int
     restore_value: no
     initial_value: '0'
   - id: gas_alert
     type: int
     restore_value: no
     initial_value: '0'

# Enable logging
logger:
  #level : INFO

web_server:
  port: 80
  auth:
   username: admin
   password: !secret web_server_password

ota:
  password: !secret ota_password

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  
mqtt:
  broker: !secret mqtt_ip
  username : !secret mqtt_username
  password : !secret mqtt_password
  
http_request:
  useragent: esphome/device
  timeout: 2s

text_sensor:
  - platform: version
    name: "Firmware control"
    
time:
  - platform: sntp
    servers: !secret ntp_ip
    on_time_sync:
      then:
        - logger.log: "Synchronized system clock"
    on_time:
      - minutes: 0
        hours: 0
        then:
          - logger.log: "Reset counter"
          - pulse_meter.set_total_pulses:
              id: "water"
              value: 0
          - pulse_meter.set_total_pulses:
              id: "gas"
              value: 0

sensor:
 - platform: pulse_meter
   pin:
      number: 13
      mode: INPUT_PULLUP
   id: water
   name: "Instant water consumption"
   unit_of_measurement: "L/min"
   timeout: 60s
   internal_filter: 50ms
   accuracy_decimals: 1
   filters:
    - delta: 0.1
   on_value:
       then:
        - logger.log: "New value"
        - if:
            condition:
              and:
                - lambda: return id(water).state == 0;
                - lambda: return id(water_alert) == 1;
            then:
              - logger.log: "Previously we are in alert release it"
              - globals.set:
                  id: water_alert
                  value: '0'
        - if:
            condition:
              and:
                - lambda: return id(water).state > 0;
                - lambda: return id(water_alert) == 0;
            then:
              - logger.log: "Water consumption above 0 (no for)"
        - if:
            condition:
              and:
                - lambda: return id(water).state > 0;
                - lambda: return id(water_alert) == 0;
                - for:
                    time: 10s
                    condition:
                      - lambda: return id(water).state > 0;
            then:
              - globals.set:
                  id: water_alert
                  value: '1'
              - logger.log: "Water consumption above 0 for more than 10s"
   total:
    name: "Total water consumption"
    unit_of_measurement: "m³"
    device_class: "water"
    accuracy_decimals: 2
    filters:
    - lambda: return (x * 0.001);

 - platform: pulse_meter
   pin:
      number: 14
      mode: INPUT_PULLUP
   id: gas
   name: "Instant gas consumption"
   unit_of_measurement: "L/min"
   timeout: 60s
   internal_filter: 200ms
   accuracy_decimals: 2
   filters:
    - lambda: return (x * 10);
   total:
    name: "Total gas consumption"
    unit_of_measurement: "m³"
    device_class: "gas"
    accuracy_decimals: 2
    filters:
      - lambda: return (x * 0.01);
```

### Anything in the logs that might be useful for us?

```txt
12:04:46	[D]	[sensor:127]	'Total water consumption': Sending state 0.00100 m³ with 2 decimals of accuracy
12:04:46	[D]	[sensor:127]	'Total water consumption': Sending state 0.00100 m³ with 2 decimals of accuracy
12:04:54	[D]	[sensor:127]	'Instant water consumption': Sending state 7.83801 L/min with 1 decimals of accuracy
12:04:54	[D]	[main:491]	New value
12:04:54	[D]	[main:110]	Water consumption above 0 (no for)
12:04:54	[D]	[sensor:127]	'Total water consumption': Sending state 0.00200 m³ with 2 decimals of accuracy
12:05:02	[D]	[sensor:127]	'Total water consumption': Sending state 0.00300 m³ with 2 decimals of accuracy
12:05:09	[D]	[sensor:127]	'Total water consumption': Sending state 0.00400 m³ with 2 decimals of accuracy
12:06:09	[D]	[pulse_meter:034]	No pulse detected for 60s, assuming 0 pulses/min
12:06:09	[D]	[sensor:127]	'Instant water consumption': Sending state 0.00000 L/min with 1 decimals of accuracy
12:06:09	[D]	[main:491]	New value
12:09:16	[I]	[ota:113]	Boot seems successful, resetting boot loop counter.
12:09:16	[D]	[esp32.preferences:113]	Saving 1 preferences to flash...
12:09:16	[D]	[esp32.preferences:142]	Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
12:10:40	[D]	[sensor:127]	'Total water consumption': Sending state 0.00500 m³ with 2 decimals of accuracy
12:10:47	[D]	[sensor:127]	'Instant water consumption': Sending state 7.88955 L/min with 1 decimals of accuracy
12:10:47	[D]	[main:491]	New value
12:10:47	[D]	[main:110]	Water consumption above 0 (no for)
12:10:47	[D]	[sensor:127]	'Total water consumption': Sending state 0.00600 m³ with 2 decimals of accuracy
12:10:55	[D]	[sensor:127]	'Total water consumption': Sending state 0.00700 m³ with 2 decimals of accuracy
12:11:03	[D]	[sensor:127]	'Total water consumption': Sending state 0.00800 m³ with 2 decimals of accuracy
12:12:03	[D]	[pulse_meter:034]	No pulse detected for 60s, assuming 0 pulses/min
12:12:03	[D]	[sensor:127]	'Instant water consumption': Sending state 0.00000 L/min with 1 decimals of accuracy
12:12:03	[D]	[main:491]	New value
12:13:59	[D]	[sensor:127]	'Total gas consumption': Sending state 0.01000 m³ with 2 decimals of accuracy
```

### Additional information

As you can see at 12:04:54 there is an instant water consumption (7.8l/min), so in log there is a `new value` then `Water consumption above 0 (no for)` but after 10s there is no log with `Water consumption above 0 for more than 10s`  and instant water consumption return to 0 at 12:06:09.

Normaly with my current configuration my for must be always true (it's just for test after that I will change duration of for at 15min).

ssieb, 2023-02-01:
The condition is only checked when the `on_value` trigger is triggered.  So yes, it was true for a while, but by the time it's checked, it's false again.  You need to do something different for this test.

ssieb, 2023-02-01:
Actually, if your final test is going to be 15mins, then that should be fine.  But you can't test with 10s.

closed 2023-02-01 by ssieb

zoic21, 2023-02-01:
Hello,
Ok I don't see in that way, so for work only if it's recurring check (at least one at t and one at t + time).

Thanks for the explanation