esphome/issues #4104: ESPHome Version text sensor doesn't compile

TheGroundZero, 2023-02-01:
### The problem

Adding the version text sensor to my ESPHome config causes the build to fail.

### Which version of ESPHome has the issue?

v2022.12.8

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP32

### Component causing the issue

version

### Example YAML snippet

```yaml
substitutions:
  name: esp32-bluetooth-proxy-ef4ddc
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .not.local
  use_address: 192.168.70.61

text_sensor:
  - platform: version
    name: "${name} ESPHome Version"
    hide_timestamp: true
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esp32-bluetooth-proxy-ef4ddc.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-bluetooth-proxy-ef4ddc (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- Improv @ 1.2.3
Linking .pioenvs/esp32-bluetooth-proxy-ef4ddc/firmware.elf
/config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/esp32-bluetooth-proxy-ef4ddc/src/esphome/components/api/api_connection.o:(.literal._ZN7esphome3api13APIConnection22send_text_sensor_stateEPNS_11text_sensor10TextSensorENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x4): undefined reference to `esphome::text_sensor::TextSensor::has_state()'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/esp32-bluetooth-proxy-ef4ddc/src/esphome/components/api/api_connection.o: in function `esphome::api::APIConnection::send_text_sensor_state(esphome::text_sensor::TextSensor*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/config/.esphome/build/esp32-bluetooth-proxy-ef4ddc/src/esphome/components/api/api_connection.cpp:481: undefined reference to `esphome::text_sensor::TextSensor::has_state()'
/config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/esp32-bluetooth-proxy-ef4ddc/src/main.o:(.literal._Z5setupv+0xac): undefined reference to `vtable for esphome::version::VersionTextSensor'
[... 79 lines cut ...]
Compiling .pioenvs/esp32-bluetooth-proxy-ef4ddc/src/esphome/core/controller.o
Compiling .pioenvs/esp32-bluetooth-proxy-ef4ddc/src/esphome/core/helpers.o
Compiling .pioenvs/esp32-bluetooth-proxy-ef4ddc/src/esphome/core/log.o
Compiling .pioenvs/esp32-bluetooth-proxy-ef4ddc/src/esphome/core/scheduler.o
Compiling .pioenvs/esp32-bluetooth-proxy-ef4ddc/src/esphome/core/util.o
Compiling .pioenvs/esp32-bluetooth-proxy-ef4ddc/src/main.o
Linking .pioenvs/esp32-bluetooth-proxy-ef4ddc/firmware.elf
RAM:   [==        ]  16.6% (used 54504 bytes from 327680 bytes)
Flash: [======    ]  64.9% (used 1191069 bytes from 1835008 bytes)
Building .pioenvs/esp32-bluetooth-proxy-ef4ddc/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/esp32-bluetooth-proxy-ef4ddc/firmware.bin"], [".pioenvs/esp32-bluetooth-proxy-ef4ddc/firmware.elf"])
Wrote 0x134330 bytes to file /config/.esphome/build/esp32-bluetooth-proxy-ef4ddc/.pioenvs/esp32-bluetooth-proxy-ef4ddc/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 40.27 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.70.61
INFO Uploading /config/.esphome/build/esp32-bluetooth-proxy-ef4ddc/.pioenvs/esp32-bluetooth-proxy-ef4ddc/firmware.bin (1196848 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.70.61 using esphome API
```

ssieb, 2023-02-01:
Given the bluetooth proxy, I assume you're using `esp-idf`.  Do a clean build.

TheGroundZero, 2023-02-02:
Cleaning build files and then installating did the trick

closed 2023-02-02 by TheGroundZero