esphome/issues #4106: ESP32-S2 - Media player - I2S - distorted audio

sgofferj, 2023-02-01:
### The problem

I have set up an ESPHome media player with 2 MAX98753 with an ESP32-S2. Playback of either MP3 streams or local files is distorted. The first half second plays OK, then the audio stream is interrupted and sounds somewhat slower. I'm no expert but it sounds like I would imagine dropped frames sound.
An identical configuration on an ESP32 DevKit (without any S's) works perfectly fine.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.7

### What platform are you using?

ESP32

### Board

esp32-s2-saola-1

### Component causing the issue

i2s

### Example YAML snippet

```yaml
esphome:
  name: player-1

esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret noise_key

ota:
  password: !secret ota-password

wifi:
  ssid: !secret ssid
  password: !secret wifi-password

status_led:
  pin:
    number: GPIO15
    inverted: true

media_player:
  - platform: i2s_audio
    name: Player 1
    id: player1
    dac_type: external
    i2s_lrclk_pin: GPIO33
    i2s_dout_pin: GPIO16
    i2s_bclk_pin: GPIO18
    mode: stereo

sensor:
  - platform: wifi_signal
    name: "Player 1 WiFi RSSI"
    id: player1_wifi_rssi
    update_interval: 60s

  - platform: rotary_encoder
    name: "Player 1 Encoder"
    id: player1_encoder
    min_value: 0
    max_value: 100
    resolution: 4
    publish_initial_value: True
    pin_a:
      number: GPIO9
      inverted: true
      mode:
        input: true
        pullup: true
    pin_b:
      number: GPIO11
      inverted: true
      mode:
        input: true
        pullup: true
    on_clockwise:
      - media_player.volume_set:
          id: player1
          volume: !lambda 'return id(player1_encoder).state / 100;'
    on_anticlockwise:
      - media_player.volume_set:
          id: player1
          volume: !lambda 'return id(player1_encoder).state / 100;'

binary_sensor:
  - platform: status
    name: "Player 1 Status"
    id: player1_status
```

### Anything in the logs that might be useful for us?

```txt
nope, all fine in the logs
```

magnets99, 2023-02-03:
when you say local, is it local to the esp board? like on an sd card or is it pulling over the network.
If over the network can you try local?
Also, just out of interest try commenting out the sensors (and binary sensor) as it might be a polling problem.

sgofferj, 2023-02-03:
Local as in local to Homeassistant. The board doesn't have an SD card slot. Not sure how I can get a file on the module.
I tried disabling the sensors already - no joy.

magnets99, 2023-02-05:
ok, can you try playing two different bitrate mp3's? one low bitrate (say 128k or less) and another high bitrate or lossless format.
I think it would be worth ruling out cpu headroom.
There are a fair few people have a similar issue with 'arduino code' sound players and i think esphome is likely using the same c++ code for streaming.

sgofferj, 2023-02-07:
What I had around was a 128k webradio stream, a 128k file and a 64k file. All show little difference. Might it be related to the fact that the S2 only has one XTensa core and the code is optimized for dual core? IIRC the Arduino framework pins all the Wifi handling to one core and the user code to the other.

Electronlibre2012, 2023-02-19:
Hello,

i can confirm that until the EspHome  2023.2.1 the level is very slow and bad, before i had a loud and clear sound with a MAX98357a for mono version.

I have a stereo version who is less affected by this problem because i have a 2x15W external amplifier. The soud is less loud than before, but its usable. The mono is no longuer usable because of poor quality sound and not loud at all. I mean i have a 3w speaker that worked great before with the MAX98357 who has a 3.2W internal amplifier for a 4 Ohm speaker.

Something has changed on the level side, sorry i cant help more.

sgofferj, 2023-05-10:
Problem persists on 2023.4.4

elik745i, 2023-05-12:
Same problem, help to fix it pls

sgofferj, 2023-05-13:
Some debugging:
MP3, stereo, 44.1kHz, 32Bit, 96kBit/s: bad
MP3, stereo, 44.1kHz, 32Bit, 128kBit/s: bad
MP3, mono, 44.1kHz, 32Bit, 128kBit/s: bad
MP3, mono, 32kHz, 32Bit, 128kBit/s: OK
MP3, mono, 24kHz, 32Bit, 32kBit/s: OK

Conclusion: The S2 doesn't like high samplerates.

gdschut, 2023-11-04:
Indeed changing my mp3s to 32kHz sample rate fixed the disturbed audio. Was easily done with audacity ;-)
Any idea if there is a bug to solve or any other development that will make playing 44.1kHz files working on the ESP32 S2?

sgofferj, 2023-11-07:
I'm not very deeply in the architecture of ESPHome but one possible reason is a lack of code optimization which has no impact on multicore micros because (at least in the Arduino devkit), Wifi is pinned to one core and the user code runs on the other one. With the S2 being single core, that obviously doesn't work, so Wifi might just eat up the resources..
I have built an ESP32-based lighting system for my recumbent trike and when I started pinning some stuff to the "Arduino"-core I saw flickering of the lights similar to the sound distortions I hear when playing 48kHz mp3s on an S2 when Wifi was active.

gdschut, 2023-11-09:
Thanks for your explanation!