esphome/issues #4107: On restart or reboot device triggers unwanted automation

magnets99, 2023-02-02:
### The problem

Hello there, 
I have a shelly 2.5pm connected to two physical switches. When the switches are used they each push an automation in Home assistant which toggles my garage door or gate.
This all works fine.

Randomly through the day the device will restart, or sometimes i'll update some firmware and have to reboot the device.
When the device reboots it triggers both automatons.

```

substitutions:
  name: "hallwaygaragebuttons"
packages:
  esphome.esp_web_tools_example: github://esphome/example-configs/esp-web-tools/esp8266.yaml@v1

esphome:
  name: ${name}
  name_add_mac_suffix: false
  board: esp01_1m

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-123456"
    password: "apassword"

binary_sensor:
#button in downstairs hallway by front door to open garage
#pulled low, goes hig to activate

  - platform: gpio
    pin:
      number: GPIO13
      inverted: TRUE
    name: "${name} garageswitch"

#button in downstairs hallway by front door to open gate
#pulled low, goes hig to activate
  - platform: gpio
    pin:
      number: GPIO5
      inverted: TRUE
    name: "${name} gateswitch"

# Prevent short circuit with "floating" pin!
  - platform: gpio
    pin: GPIO16
    name: "ade7953 IRQ pin"
    internal: true

switch:
  - platform: restart
    name: 'Garage Door REBOOT'

```
This problem occours when i use the inbuilt reboot function as well. The log does not show it triggering.

Whilst the device randomly rebooting is a problem that needs sorting out, it is not beyond the possiblility that the device will restart for other reasons (like a breaker being tripped or a power outage) and i need to ensure that it doesn't open my house.

Does anyone know why this is happening and what i can do to stop it?

### Which version of ESPHome has the issue?

2022.8.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.7.6

### What platform are you using?

ESP8266

### Board

Shelly 2.5pm

### Component causing the issue

binary switch

### Anything in the logs that might be useful for us?

```txt
This is the log during a reboot

[18:01:56][I][app:102]: ESPHome version 2022.8.1 compiled on Feb  2 2023, 17:14:24
[18:01:56][I][app:104]: Project esphome.esp_web_tools_example version 1.0
[18:01:56][C][wifi:491]: WiFi:
[18:01:56][C][wifi:353]:   Local MAC: redacted
[18:01:56][C][wifi:354]:   SSID: [redacted]
[18:01:56][C][wifi:355]:   IP Address: redacted
[18:01:56][C][wifi:356]:   BSSID: [redacted]
[18:01:56][C][wifi:358]:   Hostname: 'hallwaygaragebuttons'
[18:01:56][C][wifi:360]:   Signal strength: -69 dB ▂▄▆█
[18:01:56][C][wifi:364]:   Channel: 11
[18:01:56][C][wifi:365]:   Subnet: 255.255.255.0
[18:01:56][C][wifi:366]:   Gateway: 192.168.1.1
[18:01:56][C][wifi:367]:   DNS1: 192.168.1.1
[18:01:56][C][wifi:368]:   DNS2: 0.0.0.0
[18:01:56][C][logger:275]: Logger:
[18:01:56][C][logger:276]:   Level: DEBUG
[18:01:56][C][logger:277]:   Log Baud Rate: 115200
[18:01:56][C][logger:278]:   Hardware UART: UART0
[18:01:56][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'hallwaygaragebuttons garageswitch'
[18:01:56][C][gpio.binary_sensor:016]:   Pin: GPIO13
[18:01:56][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'hallwaygaragebuttons gateswitch'
[18:01:56][C][gpio.binary_sensor:016]:   Pin: GPIO5
[18:01:56][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'ade7953 IRQ pin'
[18:01:56][C][gpio.binary_sensor:016]:   Pin: GPIO16
[18:01:56][C][restart:022]: Restart Switch 'Garage Door REBOOT'
[18:01:56][C][restart:022]:   Icon: 'mdi:restart'
[18:01:56][C][captive_portal:088]: Captive Portal:
[18:01:56][C][web_server:125]: Web Server:
[18:01:56][C][web_server:126]:   Address: hallwaygaragebuttons.local:80
[18:01:56][C][mdns:094]: mDNS:
[18:01:56][C][mdns:095]:   Hostname: hallwaygaragebuttons
[18:01:56][C][ota:085]: Over-The-Air Updates:
[18:01:56][C][ota:086]:   Address: hallwaygaragebuttons.local:8266
[18:01:56][C][api:138]: API Server:
[18:01:56][C][api:139]:   Address: hallwaygaragebuttons.local:6053
[18:01:56][C][api:143]:   Using noise encryption: NO
[18:01:56][C][improv_serial:032]: Improv Serial:
[18:02:08][D][switch:013]: 'Garage Door REBOOT' Turning ON.
[18:02:08][D][switch:037]: 'Garage Door REBOOT': Sending state OFF
[18:02:08][I][restart:016]: Restarting device...
[18:02:08][I][app:134]: Rebooting safely...
INFO Disconnected from ESPHome API for hallwaygaragebuttons.local
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for hallwaygaragebuttons.local: Error resolving IP address: [Errno -2] Name or service not known
INFO Trying to reconnect to hallwaygaragebuttons.local in the background
INFO Successfully connected to hallwaygaragebuttons.local
```

DAVe3283, 2023-02-03:
Can you share the automations? Depending on how you set your automations to trigger, that could actually be the problem, not the ESPHome device itself.

magnets99, 2023-02-05:
ah ok, here are the actions associated

```
- id: '1663917510235'
  alias: Hallway switch to Gate
  description: uses the switch in the shelly 2 behind the switch in the hallway entry
    way to trigger the gate pushbutton
  trigger:
  - type: turned_on
    platform: device
    device_id: f7e994d6dadc477bd8b117fd91ad1c3e
    entity_id: binary_sensor.hallwaygaragebuttons_gateswitch
    domain: binary_sensor
  - type: turned_on
    platform: device
    device_id: eb7d2884f47e696c1144a30137673947
    entity_id: binary_sensor.gatetriggerfromgarage
    domain: binary_sensor
  condition: []
  action:
  - service: switch.toggle
    data: {}
    target:
      entity_id: switch.gatetoggle
  mode: queued
- id: '1663917590388'
  alias: Hallway Switch to Garage door
  description: uses the switch in the shelly 2 behind the switch in the hallway entry
    way to trigger the garage door pushbutton
  trigger:
  - type: turned_on
    platform: device
    device_id: f7e994d6dadc477bd8b117fd91ad1c3e
    entity_id: binary_sensor.hallwaygaragebuttons_garageswitch
    domain: binary_sensor
  condition: []
  action:
  - type: turn_on
    device_id: eb7d2884f47e696c1144a30137673947
    entity_id: switch.garage_door_activate
    domain: switch
  mode: queued
```

DAVe3283, 2023-02-05:
Hmm, those automations look fine to me, I wouldn't expect them to trigger when the ESP reboots.

You might have to hook a serial cable up and watch the serial log while it reboots (maybe increase log verbosity to at least DEBUG) and see if there are clues why it is reporting a button press. I would expect any noise/false signals on the GPIO pins to have long since stabilized before WiFi connects, let alone Home Assistant connecting.

Your version of Home Assistant is fairly old, so I don't know if it has the improvements to automations or not. If so, or if you are willing to upgrade, you could then use the Home Assistant automation trace records to see exactly what triggered them, and that might help narrow it down.

magnets99, 2023-02-14:
thanks for the advice. Hooking up the serial log to the shelly is tricky. not impossible but i'm keen not to start pulling the wiring out of the wall unless i have to.
I've just updated Home Assistant to 
Home Assistant 2023.2.4
Frontend 20220707.1 - latest
and tried again. same issue. on reboot (even a soft reboot using esphome) the gate close automation is triggered.

I went into show trace and restarted the esphome device. the gate closed.
this was what was in Hallway Swtich to Gate step details
```
Executed: 14 February 2023 at 18:02:11
Result:
params:
  domain: switch
  service: toggle
  service_data: {}
  target:
    entity_id:
      - switch.gatetoggle
running_script: false
limit: 10
0
Executed: 14 February 2023 at 18:02:11
Result:
params:
  domain: switch
  service: turn_on
  service_data: {}
  target:
    entity_id:
      - switch.gate_close_2
running_script: false
limit: 10
```
and this is the corresponding log in esphome (via the device webpage)
```

18:02:00 | [D] | [switch:013] | 'Garage Door REBOOT' Turning ON.
-- | -- | -- | --
18:02:00 | [D] | [switch:037] | 'Garage Door REBOOT': Sending state OFF
18:02:00 | [I] | [restart:016] | Restarting device...
18:02:00 | [I] | [app:134] | Rebooting safely...
```
This is beyond my skill level to diagnose. I might see if i can up the verbosity of the esphome device

magnets99, 2023-02-14:
hmmmm Verbose doesn't seem to help
```

18:12:53 | [I] | [restart:016] | Restarting device...
-- | -- | -- | --
18:14:09 | [V] | [json:031] | Attempting to allocate 512 bytes for JSON serialization
18:14:09 | [V] | [json:051] | Size after shrink 72 bytes
18:14:09 | [V] | [json:031] | Attempting to allocate 512 bytes for JSON serialization
18:14:09 | [V] | [json:051] | Size after shrink 148 bytes
18:14:09 | [V] | [json:031] | Attempting to allocate 512 bytes for JSON serialization
18:14:09 | [V] | [json:051] | Size after shrink 144 bytes
18:14:09 | [V] | [json:031] | Attempting to allocate 512 bytes for JSON serialization
18:14:09 | [V] | [json:051] | Size after shrink 140 bytes

```

magnets99, 2023-02-14:
ok, i've fixed it. i'm not sure why it did what it did but i've changed the triiger from

```
alias: Hallway switch to Gate
  description: uses the switch in the shelly 2 behind the switch in the hallway entry
    way to trigger the gate pushbutton
  trigger:
  - type: turned_on
    platform: device
    device_id: mydeviceidredacted
    entity_id: binary_sensor.hallwaygaragebuttons_gateswitch
    domain: binary_sensor
  - type: turned_on
    platform: device
    device_id: mydeviceidredactedbutdifferent
    entity_id: binary_sensor.gatetriggerfromgarage
    domain: binary_sensor
  condition: []
  action:
  - service: switch.toggle
    data: {}
    target:
      entity_id: switch.gatetoggle
  mode: queued
```

to

```
alias: Hallway switch to Gate
  description: uses the switch in the shelly 2 behind the switch in the hallway entry
    way to trigger the gate pushbutton
  trigger:
  - platform: state
    entity_id:
    - binary_sensor.hallwaygaragebuttons_gateswitch
    - binary_sensor.gatetriggerfromgarage
    from: 'off'
    to: 'on'
  condition: []
  action:
  - service: switch.toggle
    data: {}
    target:
      entity_id: switch.gatetoggle
  mode: queued
```

and this appears to have fixed the issue. or at least restarting the esp does not trigger the automation.
possibly because of this:
https://community.home-assistant.io/t/esphome-trigger-automation-after-restart/498296/4
which is hinting that the action gets triggered simply because the state changes from unavailable to ?something?
doesn't make sense to me. but at least my house doesn't open if you toggle the power.

closed 2023-02-14 by magnets99