esphome/issues #4109: Device reports being offline when using Wifi -> use_address

svrooij, 2023-02-02:
### The problem

I'm running esphome as a standalone container, it is set to use regular docker networking (no host mode).
I've set the static ip of all the devices with: `wifi` => `use_address: '192.168.x.x'`, this means logs just work, I can update them no problem. The issue I have is that al the devices report being offline in the main dashboard.

### Which version of ESPHome has the issue?

v2022.12.8

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.1

### What platform are you using?

ESP8266

### Component causing the issue

dashboard

### Example YAML snippet

```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: '192.168.x.x'
  ap:
    ssid: ${device_name}
```

### Additional information

[image]

And if I press the log button it just shows the correct log.
[image]

Updating also works. So it's just the **online** check in the main dashboard.

craigcabrey, 2023-02-02:
I can confirm seeing the same behavior. Running ESPHome in a k8s pod. Pod can access the device just fine & can fetch logs just like OP.

jknoflook, 2023-02-03:
I am also having this same issue. Tried the following in my docker-compose as a lot of previous posts have recommended but it made no difference. 
```
environment:
      - ESPHOME_DASHBOARD_USE_PING:"true"
```

svrooij, 2023-06-03:
It’s not fixed…. Any thoughts?

RichVarney, 2023-06-30:
Am having the same issue... device shows as offline in ESPHome but I can connect to it and view the logs wirelessly, so it clearly is online

svrooij, 2023-06-30:
> Am having the same issue... device shows as offline in ESPHome but I can connect to it and view the logs wirelessly, so it clearly is online

This did help for me. https://github.com/esphome/issues/issues/641#issuecomment-534156628

RichVarney, 2023-07-01:
Yep, that does it! Using ICMP PING shows as online for this device

svrooij, 2023-07-01:
With the above solution (use ping to check online status), I’m closing this issue.

closed 2023-07-01 by svrooij