esphome/issues #4112: Pico W I2C1 not working

z-eh, 2023-02-03:
### The problem

I try to use an ina219 with my Pico W and ESPHome.
It wired to GP6 and GP7 which is the I2C1 on the Pico W.
When I configure this in yaml, it stops booting correctly.
No usable logs or anything, just simply not start correctly.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2

### What platform are you using?

RP2040

### Example YAML snippet

```yaml
i2c:
  sda: 9
  scl: 10
  scan: true

sensor:
  - platform: dht
    pin: 2
    model: DHT22
    temperature:
      name: "Children's Room Temperature"
      filters:
        offset: -3.0
    humidity:
      name: "Children's Room Humidity"
    update_interval: 60s
  - platform: ina219
    address: 0x40
    shunt_resistance: 0.1 ohm
    current:
      name: "INA219 Current"
    power:
      name: "INA219 Power"
    bus_voltage:
      name: "INA219 Bus Voltage"
    shunt_voltage:
      name: "INA219 Shunt Voltage"
    max_voltage: 4.18V
    max_current: 0.6A
    update_interval: 60s
```

ruben-iteng, 2024-10-30:
Same with using I2C_1 on GPIO2/3 (SDA/SCL)

ruben-iteng, 2024-11-26:
Https://github.com/esphome/blob/2b9013699d841a7878a796f5621adfa3461f597d/esphome/components/i2c/i2c_bus_arduino.cpp#L28-L35

One problem might be that the wrong 'Wire' is selected. It is based on the order/amount of I2C busses defined, not what pins are being used.