esphome/issues #4116: ethernet_info ip_address error with ESP-IDF

TheOfficialMrBlah, 2023-02-03:
### The problem

If ip_address from ethernet_info component is used in conjunction with esp-idf, an error occurs during compilation.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.1

### What platform are you using?

ESP32-IDF

### Board

ESP32 DEV Board

### Component causing the issue

ethernet_info

### Example YAML snippet

```yaml
esphome:
  name: 'test'

esp32:
  board: esp32dev
  framework:
    type: esp-idf

ota:

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO5

#network:
#    enable_ipv6: true

text_sensor:
  - platform: ethernet_info
    ip_address:
      name: 'IP-Adresse'
```

### Anything in the logs that might be useful for us?

```txt
src/main.cpp:13:1: error: 'ethernet_info' does not name a type
 ethernet_info::IPAddressEthernetInfo *ethernet_info_ipaddressethernetinfo;
 ^~~~~~~~~~~~~
src/main.cpp: In function 'void setup()':
src/main.cpp:93:3: error: 'ethernet_info_ipaddressethernetinfo' was not declared in this scope
   ethernet_info_ipaddressethernetinfo = new ethernet_info::IPAddressEthernetInfo();
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:93:45: error: 'ethernet_info' does not name a type
   ethernet_info_ipaddressethernetinfo = new ethernet_info::IPAddressEthernetInfo();
                                             ^~~~~~~~~~~~~
Compiling /data/test/.pioenvs/test/asio/asio/asio/src/asio.o
*** [/data/test/.pioenvs/test/src/main.o] Error 1
========================= [FAILED] Took 31.31 seconds =========================
```

### Additional information

The Arduino framework works with the same configuration.

HeMan, 2023-06-19:
Is it still problem with ESPHome version 2023.5.5?

TheOfficialMrBlah, 2023-06-19:
> Is it still problem with ESPHome version 2023.5.5?

Thank you for your reply.

The error message seems to be a bit shorter, but it looks like the same error:
```
src/main.cpp:75:1: error: 'ethernet_info' does not name a type
 ethernet_info::IPAddressEthernetInfo *ethernet_info_ipaddressethernetinfo;
 ^~~~~~~~~~~~~
/config/esphome/testesp-01.yaml: In function 'void setup()':
/config/esphome/testesp-01.yaml:591:3: error: 'ethernet_info_ipaddressethernetinfo' was not declared in this scope
/config/esphome/testesp-01.yaml:591:45: error: 'ethernet_info' does not name a type
*** [/data/testesp-01/.pioenvs/testesp-01/src/main.o] Error 1
========================== [FAILED] Took 9.52 seconds ==========================
```

closed 2023-06-22 by jesserockz