esphome/issues #4118: Wifi power save mode ignored after restart

elpedriyo, 2023-02-04:
### The problem

Hello,

I am currently using  power_save_mode: none and I can see that the latency to the device reduces a lot. However after I restart the esp32, I notice how latency increases again. It continues like if it was as default, until I reflash the esp32 again.

You can see in here what was the latency after flashing the device with power_save_mode: none and after rebooting it
[image]

Kind regards

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

D1 mini

### Example YAML snippet

```yaml
esphome:
  name: livingroom  

# Example configuration entry
#dallas:
#  - pin: 16
#    update_interval: 2s
#  
#sensor:
#  - platform: dallas
#    address: 0xf30300a27931d028
#    name: "Temperature #1"
#  - platform: dallas
#    address: 0x7e0300a279319b28
#    name: "Temperature #2"
#  - platform: dallas
#    address: 0x080300a279c3c728
#    name: "Temperature #3"
switch:
#  - platform: gpio
#    name: "Luz led"
#    pin: 27
  - platform: gpio
    name: "Luz chip"
    pin: 2
esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino
  

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "trasto686"

ota:
  password: "trasto686"

wifi:
  ssid: "Amazonas"
  password: "******"
  power_save_mode: none
  output_power: 20dB
  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Livingroom Fallback Hotspot"
#    password: "XBt6aW0J9JGf"
#
#captive_portal:
#
```

elpedriyo, 2023-02-04:
I have tried setting power save off in arduino ide on my esp32, and the setting persists after reboot without any issue, with expected latency levels. So I guess it is an issue on how esphome sets up the power saving setting

ssieb, 2023-02-13:
Add this to your config:
```yaml
sensor:
  - platform: template
    name: wifi ps mode
    lambda: |-
      wifi_ps_type_t ps_type;
      esp_err_t err = esp_wifi_get_ps(&ps_type);
      ESP_LOGD("wifi_test", "get_ps returned %d", err);
      return ps_type;
```
See what is in the log and sensor value.

elpedriyo, 2023-02-14:
This is what the log shows:
`[C][template.sensor:023]: Template Sensor 'wifi ps mode'
[C][template.sensor:023]:   State Class: ''
[C][template.sensor:023]:   Unit of Measurement: ''
[C][template.sensor:023]:   Accuracy Decimals: 1
[C][template.sensor:024]:   Update Interval: 60.0s`

elpedriyo, 2023-02-14:
I have reflashed esp with your code and this is the report after flashing:
[V][sensor:076]: 'wifi ps mode': Received new state 0.000000
[D][sensor:127]: 'wifi ps mode': Sending state 0.00000  with 1 decimals of accuracy

And this other one after rebooting:
[V][sensor:076]: 'wifi ps mode': Received new state 1.000000
[D][sensor:127]: 'wifi ps mode': Sending state 1.00000 with 1 decimals of accuracy

jesserockz, 2023-02-14:
Can you set the logger to `VERBOSE`
Check for this message during startup
```c++
    if (!this->wifi_apply_power_save_()) {
      ESP_LOGV(TAG, "Setting Power Save Option failed!");
    }
```

elpedriyo, 2023-02-14:
I already have VERBOSE logger AFAIK: https://pastebin.com/xLt3THU9
But I can not find that message 
This is my currently yaml file: https://pastebin.com/wnY9jti1

ssieb, 2023-02-14:
You will only see that from serial logs if you watch from boot.

elpedriyo, 2023-02-14:
The logs I have pasted in there were taken from serial

elpedriyo, 2023-02-14:
Here you have full log after OTA flashing, and reboot via reset button: https://pastebin.com/aFYiaUuA

elpedriyo, 2023-02-14:
Just tried very verbose, and not showing either https://pastebin.com/eei1CFcV

ssieb, 2023-02-14:
> The logs I have pasted in there were taken from serial

Sorry, I should have checked first.  Try this:
```yaml
external_components:
  - source: github://ssieb/esphome@wifipower
    components: [ wifi ]
```
Use verbose logging.

elpedriyo, 2023-02-14:
New logs after flashing and after rebooting: https://pastebin.com/uu7nE52a
However issue not solved

ssieb, 2023-02-14:
I was not expecting it to be solved.  I'm just trying to get more information on what is happening.  And apparently, it's not even trying to set the power mode at all.

elpedriyo, 2023-02-14:
Weird thing is that it seems it sets it up after flashing, but not after rebooting :/

ssieb, 2023-02-14:
I forgot to actually commit my changes... :man_facepalming: 
Try again.

elpedriyo, 2023-02-14:
When I run the flashing it pop up this message:
C:\Users\Pedro>esphome run prueba.yaml
INFO Reading configuration prueba.yaml...
INFO Updating https://github.com/ssieb/esphome.git@wifipower
INFO Generating C++ source...
INFO Compiling app...

But when I get into https://github.com/ssieb/esphome.git@wifipower it shows nothing. I guess that is not expected, right?

elpedriyo, 2023-02-14:
However I just flashed it again, and reboot, and here are the new logs: https://pastebin.com/iesq8pjQ

ssieb, 2023-02-14:
> But when I get into https://github.com/ssieb/esphome.git@wifipower it shows nothing. I guess that is not expected, right?

That's not a real url, it's a short form for esphome to use.

> However I just flashed it again, and reboot, and here are the new logs: https://pastebin.com/iesq8pjQ

I don't see any mention of it trying to set the mode, so I'll have to go through the code again to see how that's possible.

ssieb, 2023-02-14:
I really don't understand how that's happening.  I added some more logging, try again.

ssieb, 2023-02-14:
I'm still confused about what's going on with the code, but I also can't reproduce your issue.  My ESP32 always uses the configured power save mode.

elpedriyo, 2023-02-14:
Here the new logs: https://pastebin.com/rQSZED8K

ssieb, 2023-02-14:
I wonder if your problem is related to the auth fail and reconnect because the mode starts correctly and changes after all that.

ssieb, 2023-02-14:
Can you try with debug level logging instead of verbose?

elpedriyo, 2023-02-14:
> I wonder if your problem is related to the auth fail and reconnect because the mode starts correctly and changes after all that.

I see at the beginning as you said is reporting the correct PS mode and after it finishes booting it's reporting the wrong one.
Any idea on why it is giving me all time that auth fail?

> Can you try with debug level logging instead of verbose?

Let me try

elpedriyo, 2023-02-14:
It seems you discovered the issue:
https://pastebin.com/ZvyJEiss

I have just created a mobile network with my smartphone with no password, and it is booting on the correct mode. 
So I guess some sort of bug when it tries to reconnect to the wifi.
Do we know if that can be fixed in the code?

elpedriyo, 2023-02-14:
Now, I would also like to know why it is erroring when trying to connect to my wifi xD

ssieb, 2023-02-14:
I don't know why your AP is failing the auth request, but maybe we can solve the power mode issue.  Can you try again with debug logging?

elpedriyo, 2023-02-14:
I think it has something to do with 5ghz or wpa3 encryption. Will solve that on my own. But yes, lets focus on the issue with the power mode as it seems that even if it goes ok at first time, if wifi fails at any stage it will go wrong the power mode again.
Let me change to debug and will send you logs

elpedriyo, 2023-02-14:
The logs with debug mode: https://pastebin.com/Lp8dx44v

ssieb, 2023-02-15:
I made a change, try updating and see what happens.

elpedriyo, 2023-02-15:
Going now

elpedriyo, 2023-02-15:
It seems the issue is now fixed: https://pastebin.com/H385TWqC

I have rebooted the device a few times, and the power mode seems to stay after all of them :)

elpedriyo, 2023-02-15:
Are you able to update esphome code for the project to be fully fixed?

ssieb, 2023-02-15:
No, it's not a good fix and only applies to one specific wifi type.  That was just to verify the problem.

elpedriyo, 2023-02-15:
Anything else I can help with for this fix to get implemented?

ssieb, 2023-02-15:
No, someone just needs to do it.

elpedriyo, 2023-02-15:
ok, sooo, I guess that you are part of the team and this will get under track for someone to pick it when possible?

ssieb, 2023-02-15:
No idea about an ETA.  I'll leave my branch available until there is a fix though, so you can use that for now.

elpedriyo, 2023-02-15:
hehe thanks :)

shehzman, 2023-06-23:
Also experiencing this issue with my esp32 wroom devkit