esphome/issues #4120: ESPHOME/HA not connecting back to device after flash and disconnect

bert269, 2023-02-04:
### The problem

After flashing the ESP8266 device and verified network access in LOGs and visible in HA (flashing from Windows device via ESPHOME Web) - it works fine.
Disconnect ESP from PC and connect to power supply. Device address appears in IP router list, but is no longer visible to HA (or ESPHOME). Cannot access device.
Even accessing it directly from browser via port 6053, is not working.

Attached is the .yaml code and the connection log.
Even connecting it back to PC (from where it was flashed) does not see device until I re-install code, it is not seen by HA/ESPHOME by default
Also, while still connected to Windows PC and active on dashboard, ESPHOME still displays device as 'offline'.

What is the problem with this and why does device disappear from HA /ESPHOME when connected to power other that PC and why does it have to be re-flashed each time?

### Which version of ESPHome has the issue?

ESPHome 2022.12.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant Versions 2023.2.1

### What platform are you using?

ESP8266

### Board

ACEIRMC ESP8266 ESP-12 ESP-12F NodeMcu Mini D1 Module Lua 4M Bytes

### Component causing the issue

TBD

### Example YAML snippet

```yaml
esphome:
  name: gamepc-on

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "*********"

ota:
  password: "f4********"

wifi:
  # ...
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.188
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
    dns1: 192.168.1.247

  ssid: "ASUS5300_24"
  password: "**1t0*******"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "gamepc-on Fallback Hotspot"
    password: "I2V9O2zPKPcZ"

# Example configuration entry
switch:
  - platform: gpio
    pin: 
       number: 2
       inverted: true
 #     pullup: false
 #     pulldown: false
    id: relay
    name: "gamepc-on"
    icon: "mdi:gate"
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: relay

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
_[logger:258]: Log initialized
[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[C][switch.gpio:011]: Setting up GPIO Switch 'gamepc-on'...
[D][switch:017]: 'gamepc-on' Turning OFF.
[D][switch:056]: 'gamepc-on': Sending state OFF
[D][switch:017]: 'gamepc-on' Turning OFF.
[C][wifi:037]: Setting up WiFi...
[C][wifi:038]:   Local MAC: C8:C9:A3:5A:A5:CE
[D][wifi:386]: Starting scan...
[D][wifi:401]: Found networks:
[I][wifi:444]: - 'ASUS5300_24' [redacted]▂▄▆█
[D][wifi:446]:     Channel: 4
[D][wifi:447]:     RSSI: -28 dB
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[D][wifi:449]: - [redacted] [redacted]▂▄▆█
[I][wifi:257]: WiFi Connecting to 'ASUS5300_24'...
[I][wifi:518]: WiFi Connected!
[C][wifi:362]:   Local MAC: C8:C9:A3:5A:A5:CE
[C][wifi:363]:   SSID: [redacted]
[C][wifi:364]:   IP Address: 192.168.1.188
[C][wifi:365]:   BSSID: [redacted]
[C][wifi:367]:   Hostname: 'gamepc-on'
[C][wifi:369]:   Signal strength: -31 dB ▂▄▆█
[C][wifi:373]:   Channel: 4
[C][wifi:374]:   Subnet: 255.255.255.0
[C][wifi:375]:   Gateway: 192.168.1.1
[C][wifi:376]:   DNS1: 192.168.1.247
[C][wifi:377]:   DNS2: 0.0.0.0
[D][wifi:527]: Disabling AP...
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: 192.168.1.188:8266
[C][ota:097]:   Using Password.
[C][api:025]: Setting up Home Assistant API server...
[I][app:062]: setup() finished successfully!
[I][app:102]: ESPHome version 2022.12.8 compiled on Feb  3 2023, 15:59:19
[C][wifi:504]: WiFi:
[C][wifi:362]:   Local MAC: C8:C9:A3:5A:A5:CE
[C][wifi:363]:   SSID: [redacted]
[C][wifi:364]:   IP Address: 192.168.1.188
[C][wifi:365]:   BSSID: [redacted]
[C][wifi:367]:   Hostname: 'gamepc-on'
[C][wifi:369]:   Signal strength: -31 dB ▂▄▆█
[C][wifi:373]:   Channel: 4
[C][wifi:374]:   Subnet: 255.255.255.0
[C][wifi:375]:   Gateway: 192.168.1.1
[C][wifi:376]:   DNS1: 192.168.1.247
[C][wifi:377]:   DNS2: 0.0.0.0
[C][logger:293]: Logger:
[C][logger:294]:   Level: DEBUG
[C][logger:295]:   Log Baud Rate: 115200
[C][logger:296]:   Hardware UART: UART0
[C][switch.gpio:076]: GPIO Switch 'gamepc-on'
[C][switch.gpio:078]:   Icon: 'mdi:gate'
[C][switch.gpio:084]:   Inverted: YES
[C][switch.gpio:098]:   Restore Mode: restore defaults to OFF
[C][switch.gpio:031]:   Pin: GPIO2
[C][captive_portal:088]: Captive Portal:
[C][mdns:103]: mDNS:
[C][mdns:104]:   Hostname: gamepc-on
[C][ota:093]: Over-The-Air Updates:
[C][ota:094]:   Address: 192.168.1.188:8266
[C][ota:097]:   Using Password.
[C][api:138]: API Server:
[C][api:139]:   Address: 192.168.1.188:6053
[C][api:141]:   Using noise encryption: YES
[D][api:102]: Accepted 192.168.1.233
[D][api.connection:918]: Home Assistant 2023.2.0 (192.168.1.233): Connected successfully
[D][api:102]: Accepted 192.168.1.233_
```

nagsterFVZ, 2023-02-04:
Having basically the same issue since updating to Home Assistant Versions 2023.2.1 yesterday. I can flash the esp fine, use a static ip. Device shows on my network with correct IP, I can ping it from my pc and from the terminal in HA but espHome showing it as either offline or online but cannot get logs as it fails to connect. Also the entities are just showing unavailable for data.

MichielioZ, 2023-02-05:
After updating several of my ESPs won't come back online, but initially they did operate normally. Most of my ESPHome devices seem to not like the last update when it comes to network stability. Then again, it's always been hit and miss for me with ESPHome updates ...

poundy, 2023-02-07:
Did you move from using the deprecated API password to use the encryption key ?  I am having the same issue, ESPHome sees my device in the dashboard, but the ESPHome integration in HA can't connect.

bert269, 2023-02-07:
I am new to ESPHOMEm, only started about a week or so ago. (Therefore have
not used any old deprecated keys.

All ESPHOME setups are using the new encrypted keys.

I have three devices with WLED running on the same hardware and they work
just fine - from day one, no problems.

ESPHOME really has a problem and I hope someone that knows what is going
on, will step up - it's very frustrating and not a good sign/image for Home
Assistant.
With each new release of HA, everyone RAVES about the wonderful integration
with ESPHOME - really?

Kimbaras, 2023-03-15:
I'm having basically the same issue with a different board (XIAO esp32c3). Also in my case, I've started playing with it a couple of days ago (and I keep HA updated to the latest).

Basically I'm able to flash the board while connected to the PC and with the esphome web option (HA is currently a VM, so connecting directly to it might be tricky) and I can see the logs, HA reports it working and I can properly use it (as a simple light switch for now).

One I disconnect the board and reboot it, HA still sees it online and I can perform OTA updates, but seems it's unable to connect to the API port / key and so the functionality does not work.
Of course after the reboot I can see the board in the devices lists and also ping it but the APIs is no bueno.

Did anyone find a solution or know if a fix was released?

THX2112, 2023-05-09:
Exact same issue.

vcjansen, 2023-06-25:
exact same issue as well over here, up until ESPHome 2023.6.2 still persisting. Wemos C3 mini

davidfierro, 2023-07-14:
> exact same issue as well over here, up until ESPHome 2023.6.2 still persisting. Wemos C3 mini

Same thing happening to me. Using a Wemos C3 pico

vcjansen, 2023-07-14:
Update from my side. Haven't solved the problem, but found an alternative option that works. I installed the same setup on the esp-idf platform instead of the Arduino. That worked for me on a C3 mini, first time connection in 2 seconds after reboot, full API.

bdraco, 2023-11-13:
Can you try 2023.11.0b4 ?

bdraco, 2023-12-21:
Closing this due to lack of response

closed 2023-12-21 by bdraco