esphome/issues #4121: Can't compile due to C++ #define conflict

ajacques, 2023-02-04:
### The problem

I'm encountering a compilation error when trying to compile because the C++ term `BUTTON` is being defined as a number in `pins_arduino.h`, but the term `BUTTON` appears to be generic and used in other contexts in the esphome source code.

```
Compiling .pioenvs/minimal-bug/src/main.cpp.o
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.h:29,
                 from /config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:79,
                 from /config/.esphome/platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:29,
                 from src/esphome/components/i2c/i2c_bus_arduino.h:7,
                 from src/esphome.h:10,
                 from src/main.cpp:3:
/config/.esphome/platformio/packages/framework-arduinoespressif32/variants/adafruit_qtpy_esp32c3/pins_arduino.h:14:16: error: expected identifier before numeric constant
 #define BUTTON 9
                ^
src/esphome/core/component_iterator.h:94:5: note: in expansion of macro 'BUTTON'
     BUTTON,
     ^~~~~~
[...]
```

My device is an [Adafruit Qt Py ESP32-C3](https://www.adafruit.com/product/5405) and I defined my config like:

```yaml
esp32:
  board: adafruit_qtpy_esp32c3
  variant: ESP32C3
  framework:
    type: arduino
    version: 2.0.6
```

From what I can tell, the pins_arduino.h file for this board has a `#define BUTTON 9` [here](https://github.com/espressif/arduino-esp32/blame/master/variants/adafruit_qtpy_esp32c3/pins_arduino.h#L14)

But then ESPHome uses BUTTON in a variety of places like [here](https://github.com/esphome/esphome/blob/dev/esphome/core/component_iterator.cpp#L122) and the C++ preprocessor comes along and breaks that code.

I've narrowed it down to me introducing a template button. If I compile the example YAML snippet without the button, it'll succeed, but it'll fail when I add this button.

```yaml
button:
  - platform: template
    name: "Reset SI1145"
    on_press: [
      # Intentionally left blank
    ]
```

Checking some other board variants ([See here](https://github.com/search?q=%23define+BUTTON+repo%3Aespressif%2Farduino-esp32+filename%3Apins_arduino.h&type=Code&ref=advsearch&l=&l=)), this doesn't seem particularly common problem, but this appears to be an issue that could be accidentally caused by somebody contributing a breaking change to the espressif/arduino-esp32 repository and it's hard to know for esphome users if they can/should use a specific board or not.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.2.0

### What platform are you using?

ESP32

### Board

adafruit_qtpy_esp32c3

### Component causing the issue
[... 113 lines cut ...]
   } state_{IteratorState::NONE};
                               ^
src/esphome/core/component_iterator.h:19:7: note: candidate: 'constexpr esphome::ComponentIterator::ComponentIterator()'
 class ComponentIterator {
       ^~~~~~~~~~~~~~~~~
src/esphome/core/component_iterator.h:19:7: note:   candidate expects 0 arguments, 1 provided
src/esphome/core/component_iterator.h:19:7: note: candidate: 'constexpr esphome::ComponentIterator::ComponentIterator(const esphome::ComponentIterator&)'
src/esphome/core/component_iterator.h:19:7: note:   conversion of argument 1 would be ill-formed:
src/esphome/core/component_iterator.h:19:7: note: candidate: 'constexpr esphome::ComponentIterator::ComponentIterator(esphome::ComponentIterator&&)'
src/esphome/core/component_iterator.h:19:7: note:   conversion of argument 1 would be ill-formed:
src/esphome/core/component_iterator.h:121:5: error: cannot declare variable 'esphome::state_' to be of abstract type 'esphome::ComponentIterator'
   } state_{IteratorState::NONE};
     ^~~~~~
src/esphome/core/component_iterator.h:19:7: note:   because the following virtual functions are pure within 'esphome::ComponentIterator':
 class ComponentIterator {
       ^~~~~~~~~~~~~~~~~
src/esphome/core/component_iterator.h:43:16: note: 	'virtual bool esphome::ComponentIterator::on_button(esphome::button::Button*)'
   virtual bool on_button(button::Button *button) = 0;
                ^~~~~~~~~
src/esphome/core/component_iterator.h:126:1: error: expected declaration before '}' token
 }  // namespace esphome
 ^
*** [.pioenvs/minimal-bug/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.15 seconds ==========================
```

ajacques, 2023-02-04:
One workaround for this issue is to switch `esphome.board:adafruit_qtpy_esp32c3` to `esphome.board: esp32-c3-devkitm-1`, but then I lose the friendly pin names.

oxan, 2023-02-12:
Ugh, Arduino's pollution of the global namespace is quite annoying. We probably should get rid of the Arduino-includes from all core headers, and use forward-declarations instead.

ssieb, 2023-02-12:
@oxan I was just going to comment about how terrible it is that Arduino does things like that.  I agree.

lcnittl, 2024-11-18:
Also an issue with `adafruit_feather_esp32_v2` (ESPHome 2024.10.3)