esphome/issues #4124: Deep sleep component not working on RP2040

jonfpersson, 2023-02-04:
### The problem

It seems like the deep sleep component doesn't work on the RP2040, my board consumes as much power as when not in deep sleep. This makes it very hard to run on batteries.

Both in "deep sleep" and normally the board consumes around 40 mA.

### Which version of ESPHome has the issue?

2022.12.8-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.1

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Example YAML snippet

```yaml
deep_sleep:
  run_duration: 1min
  sleep_duration: 1min
```

### Anything in the logs that might be useful for us?

```txt
[17:10:10][C][deep_sleep:049]: Setting up Deep Sleep...
[17:10:10][C][deep_sleep:052]:   Sleep Duration: 60000 ms
[17:10:10][C][deep_sleep:055]:   Run Duration: 60000 ms

[17:11:14][I][deep_sleep:116]: Beginning Deep Sleep
[17:11:14][I][deep_sleep:118]: Sleeping for 60000000us
[17:11:14][D][rp2040.preferences:117]: Saving preferences to flash...
```

studioj, 2023-02-07:
It doesn't seem to be supported for now if I check the docs. It would be a nice feature though.
Pitty I'm not into c++🤔. It seems to be supported on the rp2040 recently.

vposloncec, 2023-02-14:
+1 would love this feature

jgriesfeller, 2023-02-16:
Will not help much for battery life since the buck boost regulator on the Pi Pico is not very energy efficient compared to the regulators used e.g. on the um_tinys2 boards

tesanders, 2023-04-10:
Unfortunately, without deep sleep support, there is no way to effectively compare the efficiency of the Pi Pico W regulator to those used on ESP-S2/S3 boards.  Deep sleep support needs to be added for the Pi Pico W in order for it to be considered as a viable alternative to the ESP32 variants

ringerc, 2023-10-08:
Based on a look at https://esphome.io/api/deep__sleep__component_8cpp_source.html it appears the the actual sleep functionality is not compiled at all unless macros `USE_ESP32`, `USE_ESP32_VARIANT_ESP32C3` and/or `USE_ESP8266` are defined. See in particular [DeepSleepComponent::begin_sleep](https://esphome.io/api/classesphome_1_1deep__sleep_1_1_deep_sleep_component#aa58ff2f9621d6858c85aef10e87f3a16)

So based on a 5 min skim read it looks like for the pi pico w it's basically a no-op.

The pico SDK does have control over sleep modes. I'll see if I can cook up a patch, as this is something crucial to my intended use, and pico W is what I have so it's what I'm stuck with for now.

ringerc, 2024-06-08:
I didn't manage to get to this, as calculations showed that even for best-case pico W sleep it was unlikely I could use them for the intended battery-supported application. I'd still like to return to it.

bdraco, 2026-03-12:
Closing — deep sleep isn't implemented for RP2040, so this is a feature request rather than a bug. The best place for it is https://github.com/esphome/feature-requests. As @ringerc noted, the Pico W's power consumption even in sleep mode may not be competitive with ESP32 variants for battery applications due to the onboard regulator.