esphome/issues #4125: OTA upgrade stops with connections reset by peer error

mbrudka, 2023-02-04:
### The problem

Affected device: Sonoff Daul R3 (v2 probably).

esphome was flashed in two steps:
- tasmota using esptool.py and serial flashing
- minimal esphome (attached) upload using tasmota web config

esphome runs sonoff as expected, the relay joins WiFi and more advanced version of config with web server logs correctly and enables to manipulate switches. However, there is no way to upgrade esphome using OTA. The upload of firware stalls for a while then an error is reported:

<pre>
INFO Uploading .esphome/build/gate/.pioenvs/gate/firmware.bin (955072 bytes)
Uploading: [=                                                           ] 1% 
ERROR Error sending data: [Errno 104] Connection reset by peer
</pre>

esphome logs:
<pre>
[D][sensor:127]: 'WiFi Signal': Sending state -53.00000 dBm with 0 decimals of accuracy
[19:10:54][D][ota:143]: Starting OTA Update from ::FFFF:C0A8:7C0A...
[19:10:54][D][ota:312]: OTA in progress: 0.1%
INFO yard-gate: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO yard-gate: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for yard-gate
WARNING Disconnected from API
INFO Successfully connected to yard-gate
</pre>

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

ota

### Example YAML snippet

```yaml
# Basic Config
substitutions:
  device_name: gate

captive_portal:

esphome:
  name: ${device_name}

esp32:
  board: esp32dev

wifi:
  ssid: ""
  password: ""
  reboot_timeout: 0s
  # Enable fallback hotspot (captive portal) in case Wi-Fi connection fails
  ap:
    ssid: ""
    password: ""

ota:
  password: ""
logger:
  level: DEBUG
  logs:
    script: ERROR
```

mbrudka, 2023-02-11:
Serial logs after `esphome -v upload --device yard-gate.brudka.waw.pl gate.yaml` included.
[serial.txt](https://github.com/esphome/issues/files/10712942/serial.txt)

mbrudka, 2023-02-11:
Problem resolved by flashing esphome crafted for this device by invoking serial upload 
`esphome -v upload --device /dev/ttyUSB0 gate.yaml`
Upload logs uses different addresses when compared to original tasmota upload by esptool.py
`Uploading 
INFO Reading configuration gate.yaml...
DEBUG Running:  esptool.py --before default_reset --after hard_reset --baud 460800 --port /dev/ttyUSB0 --chip esp32 write_flash -z --flash_size detect 0x10000 /home/marek/workspace/ha/esphome-devs/src/yard/gate/.esphome/build/gate/.pioenvs/gate/firmware.bin 0x1000 /home/marek/.platformio/packages/framework-arduinoespressif32/tools/sdk/bin/bootloader_dio_40m.bin 0x8000 /home/marek/workspace/ha/esphome-devs/src/yard/gate/.esphome/build/gate/.pioenvs/gate/partitions.bin 0xe000 /home/marek/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin`

Possible  reason was an the address mismatch when flash via tasmota.

closed 2023-02-11 by mbrudka

NightcoreIsla, 2023-02-11:
I have the same problem with it
tasmota works fine, and migration to esphome successful
but cant wireless update esphome after migration.

mbrudka, 2023-02-11:
@NightcoreIsla Please try to flash esphome via serial port directly (connect TX, RX, GND, 3.3V if necessary, button pressed during restart, then upload as above). It resolved my issue.

mbrudka, 2023-02-11:
Please try to flash esphome via serial port directly (connect TX, RX, GND, 3.3V if necessary, button pressed during restart, then upload as above). It resolved my issue.