esphome/issues #4126: Fingerprint sensor security issue

eddimangit, 2023-02-04:
### The problem

Hi all, 
today I made an observation about an R503 Fingerprintsensor. I have a sensor installed and thought I had the communication secured by setting the password on the sensor. For communication with this sensor, I verified that the password is necessary, so far so good. Since short, the finger touch signal does not work anymore and thus I replaced the sensor with another. This new sensor did not have a password set and I was really surprised, that putting an already learned finger to the sensor, it opened my garage door. 
As the sensor is reachable from outside, this is an obvious security fraud. Anyone could replace my sensor with an own and get access.
... in the code it looks like the passwordcheck is only done once on setup...
Looks like command 13h of the R503 could be done at any time. So to call to bool FingerprintGrowComponent::checkpassword() in  void FingerprintGrowComponent::scan_andmatch() as condition to send the match could help here. Do you think this would be a possible way?
It is clear, that this would not prevent a sniffing (one layer of complexity higher) attack but would prevent a simple replacement attack...

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

fingerprint_grow

### Example YAML snippet

```yaml
any configuration of fingerprint_grow
```

Antti, 2023-02-09:
Perhaps we can change the module's default address (default `0xFFFFFFFF`) using `SetAdder` command (`15H`), then, if the new module address doesn't match, it won't respond to any commands. Again, this will not prevent MITM/emulation attack but should help with a simple module replacement attack, as the attacker would need to know the address.

blackwiz4rd, 2025-01-22:
Hello all,
I think this issue needs to be taken in consideration for development as it's a security concern for the users

Antti, 2025-01-23:
I've seen how others do it:
They store a secret in a "Notepad" using "WriteNotepad" command (`18H`) and then they check it each time when they communicate with the sensor via "ReadNotepad" command (`19H`), this way if the sensor is swapped, the notepad won't contain the stored secret.

Of course, you could still sniff traffic between the sensor and the controller and find the notepad content, but it's more complicated than just swapping the sensor.