esphome/issues #4127: PCF8574 initilization toggles output at boot

swifty99, 2023-02-04:
### The problem

A similar issue was present a long time ago, and has been fixed. Now it is the opposite way. Any output which is not inverted in the beginning of boot will go on for short time.
For me this triggers relais at boot an may damage devices. Interlocking has probably no effect at this stage.

see: [PCF8574 initilization toggles output at boot (if inversed logic)](https://github.com/esphome/issues/issues/755)

### Which version of ESPHome has the issue?

2022.12.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.1.4

### What platform are you using?

ESP8266

### Board

d1 mini, not important

### Component causing the issue

PCF8574 

### Example YAML snippet

```yaml
- platform: gpio
    name: "Office Displays Monitor"
    id: "sw_display"
    pin:
      pcf8574: pcf8574_hub
      # Use pin number A4
      number: 3
      # One of INPUT, INPUT_PULLUP or OUTPUT
      mode:
        output: true
      inverted: false
```

swifty99, 2023-02-05:
Maybe I should explain a little further,

The PCF8574Component provides additional GPIO pin.
The Esphome treats these pins like internal pins.
Means if a mode or state is changed it will be written to I2C.
At initialization stage of the first pin, the state of the other pins are not known.
However, there is no write one pin to the PCF8574Component. Always a complete byte is written,
Means at initialization of first pins the 7 other pins are written to a default value. Highside relays will turn off for a short time, lowside one go to on.
The construction of the PCF8574 has a strange specialty. The pins can only drive current to low state. High state is constructed by a pull up only which is always active (while not pulled down actively) . This at the same time is the read IO configuration, which can be pulled low externally and read from the registers.
Due to this strange construction, a direct write of first init pin should be avoided. Until init is complete no command should be written to the device.

I assume this is the only way to survive a glitch free reset. Does ESPHome have a "init complete" stage to prevent communication till then?

baacktothefuture, 2023-09-08:
same problem as @swifty99 - replicated also with PCF8575

Jabol86, 2023-09-27:
Find any solution?

swifty99, 2023-10-28:
Nope.

Using a full blown port expander helps.

Another solution would be to read at boot before writing.

Maybe I will find some time to look into the code. Seems not to many people are troubled with this. Although it is really nasty.

rkandilarov, 2024-02-17:
Just to confirm the same problem for me with pcf8574.

radioelf, 2024-04-27:
Hi. Just to confirm, I'm experiencing the same problem with pcf8574. Thank you

moibueno, 2024-09-04:
I'm facing the same issue...

duckfullstop, 2024-12-18:
I'm getting this issue too, definitely not stale. Strangely, for me I'm seeing all outputs turn on upon a reboot, then they all turn off (bar one of my choice) when I toggle one on through software?

I'm suspicious that this may be something to do with having too many devices on the same i2c line - I'm using a KC868-A16 board which has 4 of the PCF8574's on the same bus.

Smandurlo, 2025-03-18:
Same here. I first tried with tasmota, setting the ports to output not inverted: high on boot. Turning on and then off, the output works eventually correctly.

Then I tried with esphome: same result.

Into the datasheet i found out this: "At power on, all the ports are HIGH with a weak 100 uA
internal pull-up to VDD, but can be driven LOW by an internal transistor, or an external signal".

So, I am not really sure how to deal with it. Any idea?