esphome/issues #4129: UART bus is totally bugged

majuss, 2023-02-05:
### The problem

I have a ESP8266 connected via an UART to RS232 Adapter to my LG TV.

I can control the TV via serial with commands like: `ka 0 0`

So now to the wierd things. I send the commands via an MQTT rule like this:
```yaml
mqtt:
  broker: 192.168.133.213
  username: username
  password: pw
  discovery: true
  on_message:
    - topic: uartbridge/on
      then:
        - uart.write: "mc 0 95\r\n"
```

This works totally fine via **MQTT**, but when I try to do it via an **UART switch** it doesn't. **Only** turning the TV on and off works but nothing else. I'am at a complete loss here

So this doesn't work for some reason:
```yaml
  - platform: uart
    name: "TV Toggle Brightness"
    data: "mc 0 95\r\n"
    uart_id: tv
```

I tried to mess around with the control characters etc., no luck at all.

In the past I had a shit ton of problems with the UART bus and ESPhome, this is definetly not the first problem I face with it.

My guess is, it's related to this timing issue: https://github.com/esphome/issues/issues/3912 

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### Component causing the issue

uart

### Example YAML snippet

```yaml
substitutions:
  devicename: "uartbridge"

esphome:
  name: ${devicename}
  platform: ESP8266
  board: d1_mini

# Enable logging
#logger:
#  baud_rate: 0

web_server:
  port: 80

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Uartbridge Fallback Hotspot"
    password: ""

captive_portal:

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600
  id: tv

sensor:
  - platform: wifi_signal
    name: "${devicename} WiFi Signal"
    device_class: signal_strength
    update_interval: 60s

mqtt:
  broker: 192.168.133.213
  username:
  password:
  discovery: true
  on_message:
    - topic: uartbridge/on
      then:
        - uart.write: "ka 0 1\r\n"
    - topic: uartbridge/off
      then:
        - uart.write: "ka 0 0\r\n"
    - topic: uartbridge/toggle
      then:
        - uart.write: "mc 0 95\r\n"
    - topic: uartbridge/dark
      then:
        - uart.write: "jq 0 3\r\n"
    - topic: uartbridge/medium
      then:
        - uart.write: "jq 0 2\r\n"
    - topic: uartbridge/bright
      then:
        - uart.write: "jq 0 1\r\n"
    - topic: uartbridge/max
      then:
        - uart.write: "jq 0 0\r\n"

switch:
  - platform: gpio
    name: "BT_out"
    pin: 5
    inverted: True
  - platform: gpio
    name: "TV_out"
    pin: 0
    restore_mode: ALWAYS_ON

  - platform: uart
    name: "TV Toggle Brightness"
    # data: "mc 0 95\r\n"
    data: 'mc 0 95'
    uart_id: tv
  - platform: uart
    name: "TV Turn On"
    data: "ka 0 1"
    uart_id: tv
  - platform: uart
    name: "TV Turn Off"
    data: "ka 0 0"
    uart_id: tv
  - platform: uart
    name: "TV Brightness Dark"
    data: "jq 0 3"
    uart_id: tv
  - platform: uart
    name: "TV Brightness Medium"
    data: "jq 0 2"
    uart_id: tv
  - platform: uart
    name: "TV Brightness Bright"
    data: "jq 0 1"
    uart_id: tv
  - platform: uart
    name: "TV Brightness Super Bright"
    data: "jq 0 0"
    uart_id: tv
  - platform: uart
    name: "TV Mute"
    data: "ke 0 1"
    uart_id: tv
```

ssieb, 2023-02-05:
Why didn't you include logs?  Try enabling `debug:` on the uart and see what's happening.

majuss, 2023-02-10:
It started today when I compiled with the logger and debug. I can not reproduce the isse.

closed 2023-02-10 by majuss