esphome/issues #4130: name_add_mac_suffix breaks OTA

geiseri, 2023-02-05:
### The problem

When I use `name_add_mac_suffix: true` in my configuration OTA fails because it will try to upload to the base name without the appended mac suffix.   Am I misunderstanding the purpose of this feature?

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.2

### What platform are you using?

ESP32

### Board

tinypico,m5stack-atom,esp32dev,wio_link,d1_mini

### Component causing the issue

wifi

### Example YAML snippet

```yaml
Main yaml file:

substitutions:
  friendly_devicename: Basement AQI Sensor
  log_level: DEBUG
  project: basement-aqi-sensor
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  noise_key: !secret noise_key

# Will only work when uncommented
# wifi:
#  use_address: esphome-863488.local

esp32:
  board: tinypico
  framework:
    type: arduino
  
web_server:
  local: true

packages:
  device_base: !include common/base-newnames.yaml
```

The `base-newnames.yaml` is below:

```
esphome:
  name: esphome
  name_add_mac_suffix: true
  project:
    name: geiseri.${project}
    version: "1.0"
wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_password}
  ap:
    ssid: "${friendly_devicename}"
    password: "13BwH730pGki"
    ap_timeout: 10min 
logger:
  level: ${log_level}

api:
  encryption:
    key: ${noise_key}
ota:

```
```

### Anything in the logs that might be useful for us?

```txt
Logs from install:

INFO Reading configuration /config/esphome/esphome-863488.yaml...
INFO Detected timezone 'America/New_York'
INFO Generating C++ source...
INFO Compiling app...
Processing esphome (board: tinypico; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling /data/esphome/.pioenvs/esphome/src/main.cpp.o
Linking /data/esphome/.pioenvs/esphome/firmware.elf
RAM:   [=         ]  12.4% (used 40776 bytes from 327680 bytes)
Flash: [=====     ]  53.7% (used 984713 bytes from 1835008 bytes)
Building /data/esphome/.pioenvs/esphome/firmware.bin
Creating esp32 image...
Successfully created esp32 image.
esp32_create_combined_bin(["/data/esphome/.pioenvs/esphome/firmware.bin"], ["/data/esphome/.pioenvs/esphome/firmware.elf"])
Wrote 0x101d10 bytes to file /data/esphome/.pioenvs/esphome/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 16.75 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of esphome.local
ERROR Error resolving IP address of esphome.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname
```

If I put the `use_address` to the full name I see the following:

```
[16:04:19][C][web_server:125]: Web Server:
[16:04:19][C][web_server:126]:   Address: esphome-863488.local:80
[16:04:19][C][mdns:103]: mDNS:
[16:04:19][C][mdns:104]:   Hostname: esphome-863488
[16:04:19][C][ota:093]: Over-The-Air Updates:
[16:04:19][C][ota:094]:   Address: esphome-863488.local:3232
[16:04:19][C][api:138]: API Server:
[16:04:19][C][api:139]:   Address: esphome-863488.local:6053
[16:04:19][C][api:141]:   Using noise encryption: YES
[16:04:19][C][wifi_signal.sensor:009]: WiFi Signal 'Basement AQI Sensor WiFi Signal'
[16:04:19][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[16:04:19][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[16:04:19][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[16:04:19][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[16:04:19][C][wifi_info:009]: WifiInfo IPAddress 'Basement AQI Sensor IP Address'
[16:04:22][D][api:102]: Accepted ::FFFF:10.0.12.183
[16:04:22][D][api.connection:918]: Home Assistant 2023.2.2 (::FFFF:10.0.12.183): Connected successfully
```

I can also ping it by that name.
```

### Additional information

I tried to delete the `.esphome` directory in case there was stale information.  Am I misunderstanding this feature?

ssieb, 2023-02-05:
As already discussed on discord, you were misunderstanding the feature and this issue is explicitly called out in the [docs](https://esphome.io/components/esphome.html#adding-the-mac-address-as-a-suffix-to-the-device-name).  This feature is primarily useful for OEMs so they can flash a single binary image to all their devices.

geiseri, 2023-02-06:
Yes, the correction to close this bug was to clarify the docs since according to google this is a popular "issue" I was going to do a PR for the documentation.  Please keep this open until that happens so that it is resolved once the PR is merged.

closed 2023-03-06 by jesserockz