esphome/issues #4131: AC Dimmer Component doesn't dim correctly using RobotDyn AC Dimmer

Jetro97, 2023-02-06:
### The problem

As already seen on [this issue](https://github.com/esphome/issues/issues/1632), when using the ac_dimmer component in combination wit the RobotDyn AC dimmer the lamp has an irregular and exagerated flicker. Tried with both LED and traditional light bulb. I live in Italy, we have 230v at 50hz.
An user found out the problem but after some time the issue was closed even if it's still not working.

I know there is a warning on the wiki page stating this is still in development, but RobotDyn was the suggested dimmer so I wasn't expecting this behaviour.

Also I read all [this discussion](https://github.com/esphome/esphome/pull/880), but found nothing.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.2

### What platform are you using?

ESP32

### Board

ESP32 Development Board

### Component causing the issue

RobotDyn AC Dimmer (4A Single Channel)

### Example YAML snippet

```yaml
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO21
    method: trailing
    zero_cross_pin:
      number: GPIO19
      mode: INPUT_PULLUP
      inverted: yes
```

### Anything in the logs that might be useful for us?

```txt
Nothing useful
```

### Additional information

 Tried all the possible setting for the ac_dimmer component and found nothing that can be used.

cvjensen, 2023-02-22:
Any news ? I'm having the same problem ...

Jetro97, 2023-02-24:
Unfortunately nothing. I suggest to place a warning in the wiki stating this is not working with RoboDyn AC Dimmer.

There are various thread about this  since at least 2 years and nothing has changed.

I don't know if there's something to do but i hope someone will find out

randybb, 2023-02-24:
This is a triac dimmer, so the method needs to be _leading pulse_ (default). I have been using the same dimmer for a fan in my pain cave, which is a good resistive load and it works just fine. Not sure about incandescent bulbs - don't have any.

cvjensen, 2023-02-24:
> This is a triac dimmer, so the method needs to be _leading pulse_ (default). I have been using the same dimmer for a fan in my pain cave, which is a good resistive load and it works just fine. Not sure about incandescent bulbs - don't have any.

I've just tested a LED bulb with leading pulse and it didnt work . leading did however .
Another of my bulbs is buzzing alot however . It says dimmable but it just buzzes way too much for it to be useful . 
So I actually are having a little bit of succes with some ikea bulbs .
But I have to use method: LEADING

Jetro97, 2023-02-24:
I've tried with every method with similar results, while with shelly or old-style dimmer every lamp work perfectly.

Tried with incandescent lamp and LED constant current lights (various). I can understand problems with particular LEDs, but neither the incandescent 40W bulb is working...

cesarpuig, 2024-01-18:
Hello

Has anyone been able to solve it? I am in the same situation. I'm trying dimmable Philips lamps.

Using the leading method, it works but it does not regulate well, it regulates in a stepwise, non-linear way. Using leading pulse mode directly does not regulate.

I have read all the information I have found both in the Esphome docs and on other sites and I have tried countless configurations but I can't get it to work.

Thank you

Jetro97, 2024-01-18:
Same here, best combo was shelly dimmer with constant current psu, but it still flicker too much to be used.

cesarpuig, 2024-01-25:
I have tried incandescent lamps and it doesn't work for me either. I look forward to future updates. If anyone gets any progress, please publish it.

Thank you

medassi, 2024-03-18:
Hello, I also have a problem with the esphome ac dimmer library. I want to route my surplus electricity from solar panels to my electric heating slab from 0 to 3700w. I have a 40 amp robotdyn. Unfortunately, the opening of the triac is not linear at all. Up to 40%, the aperture is very small, and then much larger. What's wrong with this bookstore. For your information, I am in France with a voltage of 230v at 50 Hz. Thank you in advance for the solutions you can provide me.
[image]

benoit-cty, 2024-05-19:
Hello @medassi, I'm trying to make the same things.
Did you manage to make it work ? What is your esphome config ?

I understand that the non-liinear behaviour is due to brightness/gamma correction for lights.

medassi, 2024-05-20:
Hi. Yes indeed I managed to obtain something satisfactory with +1 on the opening if the surplus is greater than 200w and -2 if I consume more than 1W from the network. It seems that the opening percentage of the Triac does not correspond to the power percentage but ultimately it is not that important. For the moment, it is HomeAssistant which is responsible for managing this opening of Triac (via EspHome) but in the long term, perhaps I will have EspHome do the work directly.

benoit-cty, 2024-05-20:
Thanks !

Can you share your esphome config ?

My Robodyn dimmer was almost not working with my esphome config : no just flickering but also blinking.

I build a custom Arduino code using https://github.com/fabianoriccardi/dimmable-light and it works better.

Here is my template in configuration.yaml:
```yml
template:
  - sensor:
        name: "commande_dimmer"
        # Le dimmer va de 0 à 255
        state: >
            {{
              ((max(
                min(
                  (states('sensor.puissance_compteur') | int(0) ) * -255 / 2000 +
                  (states('sensor.commande_dimmer') | int(0))
                ,255)
              ,0)
              + (states('sensor.commande_dimmer') | int(0))
              ) / 2) | round (0)
            }}
```

And my automation:
```yml
- id: routage_surplus_solaire
  alias: ECS - Pilotage intensite du chauffe-eau
  description: ''
  trigger:
  - platform: state
    entity_id: sensor.commande_dimmer
  action:
  - service: mqtt.publish
    data_template:
      topic: home/dimmer/set
      payload: '{{states(''sensor.commande_dimmer'')}}'
```

medassi, 2024-05-21:
My configurations:

### Esphome for dimmer :
```
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: 12
    zero_cross_pin:
      number: 14
      mode: INPUT_PULLDOWN
```
And for my red,green,yellow Leds : (not important here)

```
light:
  - platform: monochromatic
    output: dimmer1
    name: Dimmer Triac
    default_transition_length: 50ms
    id: dimmer_percent
    on_state: 
      then:
        - lambda: 
            if( id(dimmer_percent).remote_values.is_on() ) {
              if( id(dimmer_percent).remote_values.get_brightness() < .33f ){
                id(gpio_4).turn_on() ;
                id(gpio_2).turn_off() ;
                id(gpio_15).turn_off() ;
              }else if( id(dimmer_percent).remote_values.get_brightness() < .66f ){
                id(gpio_4).turn_off() ;
                id(gpio_2).turn_on() ;
                id(gpio_15).turn_off() ;
              }else{
                id(gpio_4).turn_off() ;
                id(gpio_2).turn_off() ;
                id(gpio_15).turn_on() ;
              }
            }else{
              id(gpio_4).turn_off() ;
              id(gpio_2).turn_off() ;
              id(gpio_15).turn_off() ;
            }
```
### In HomeAssistant after conditions tests:
   ```
 default:
      - variables:
          nouvelleOuverture: >-
            {% set consoReseau=states('sensor.envoy_production') | int %}  {% if
            is_state('light.esphome_chauffage_local_dimmer_triac','off') %}
              {% set ouvertureActuelle = 0 %}
            {% else %}
              {% set ouvertureActuelle = (100*states.light.esphome_chauffage_local_dimmer_triac.attributes.brightness / 255) | round %} 
            {% endif %}  {% if ( consoReseau < -300 ) %}
              {% set nouvelleOuverture = ouvertureActuelle+1 %}
            {% elif ( consoReseau > -10 ) %}
              {% set nouvelleOuverture = ouvertureActuelle-5 %}
            {% else %}
              {% set nouvelleOuverture = ouvertureActuelle %}
            {% endif %}  {% if nouvelleOuverture < 0 %} {% set
            nouvelleOuverture=0 %} {% endif %} {% if nouvelleOuverture > 95 %}
            {% set nouvelleOuverture=95 %} {% endif %} {{ nouvelleOuverture }}
      - service: light.turn_on
        metadata: {}
        data:
          brightness_pct: "{{ nouvelleOuverture }} "
        target:
          entity_id: light.esphome_chauffage_local_dimmer_triac
```

shibida, 2025-05-28:
> Hello, I also have a problem with the esphome ac dimmer library. I want to route my surplus electricity from solar panels to my electric heating slab from 0 to 3700w. I have a 40 amp robotdyn. [...]

it is because of the default gamma of monochromatic :
you can use it like that :
`light:`
`  - platform: monochromatic`
`    id: gradateur`
`    output: dimmer`
`    gamma_correct: 1.0`