esphome/issues #4134: Climate Thermostat preset_change triggered before defining `this->custom_preset`

0x3333, 2023-02-06:
### The problem

When the preset_change trigger is issued, the `this->custom_preset` variable is not set yet. This means that from the lambda we can't know which preset has been just set(default or custom).

Trigger lambda should receive the new preset or the `this->custom_preset` variable should be updated before calling the trigger.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

nodemcu

### Component causing the issue

Thermostat

### Example YAML snippet

```yaml
climate:
  - platform: thermostat
    id: heater
    name: $name
    sensor: sensor_temperature

    heat_deadband: "0.5 °C"
    heat_overrun: "0.5 °C"
    min_heating_run_time: 1s
    min_heating_off_time: 1s
    min_idle_time: 1s

    visual:
      min_temperature: ${min_temp} °C
      max_temperature: ${max_temp} °C
      temperature_step: ${inc_temp} °C

    heat_action:
      - switch.turn_on: heater_switch
    idle_action:
      - switch.turn_off: heater_switch

    heat_mode:
      - switch.turn_on: fan_switch
    off_mode:
      - switch.turn_off: fan_switch

    preset_change:
      - lambda: |-
          std::string preset_name = id(heater).preset.value();
          ESP_LOGD("preset_change", "Name: %s", preset_name);
```

### Anything in the logs that might be useful for us?

```txt
[10:07:19][D][preset_change:398]: Name: \xac#\xfb?
```

0x3333, 2023-02-06:
This happens with preset and custom_preset.

[image]

Jorre05, 2023-03-20:
Same problem here.

gyengus, 2023-05-18:
I have the same issue, but with version 2023.5.1

closed 2023-09-20 by jesserockz