esphome/issues #4137: FastLED is out of date and does not compile for  um_feathers2 board.

mzakharo, 2023-02-06:
### The problem

Problem compiling FastLED library for `um_feathers2` board with clock/data pins of 40/45.  Compiler complains that the pins are not valid. Problem is resolved by changing fastLED library to the latest version (3.5.0)
file:
```esphome\components\fastled_base\__init__.py```
```
 cg.add_library("fastled/FastLED", "3.5.0")
 ```

P.S. override of fastLED version through yaml fails as esphome forces a specific 3.3.2 version.
P.S. fastLED library v3.5.0 still needs modification to function correctly on um_feathers2. Issue tracked [here](https://github.com/FastLED/FastLED/issues/1475)

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.2

### What platform are you using?

ESP32

### Board

um_feathers2

### Component causing the issue

FastLED 

### Example YAML snippet

```yaml
light:
  - platform: fastled_spi
    chipset: APA102
    data_pin: 40
    clock_pin: 45
    num_leds: 1
    rgb_order: BGR
    color_correct: [33%, 33%, 100%]
```

nielsnl68, 2023-02-07:
checkout my PR https://github.com/esphome/esphome/pull/4257)

ssieb, 2023-06-07:
Use https://esphome.io/components/light/esp32_rmt_led_strip.html instead.

closed 2023-06-07 by ssieb