esphome/issues #4138: Cant connect esp8266 to wifi on asus router

totopesce, 2023-02-07:
### The problem

Recently installed home assistant and installed the esphome addon.
When trying to add a new device it hangs and says it cant make out esp.local or a dns problem. Sometimes it wont even connect to my wifi and just throw an error.
I tried using esphome web to flash the firmware that HA made, also tried installing the first time installation and using HA to complete the confige and tried to connect to wifi from the esphome web as well, and I either cant get it to connect to wifi or it gives errors trying to find the device in the network.
Tried finding the esp ip in my asus router, but it doesnt appear in the devices list.

Ive read that esp8266, d1 mini in my case, have problems connecting to asus routers, i have an 86u, and either changing the arduino core version to 2.5.2 works or changing the network standart to 802.11 g. Havent found any way to do either of those.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.2

### What platform are you using?

ESP8266

### Board

D1 mini

### Example YAML snippet

```yaml
Its the ones created by HA and web esphome setups, havent tried adding anything cause I cant get it to connect yet.
```

### Anything in the logs that might be useful for us?

```txt
Error message
ERROR Error resolving IP address
```

ssieb, 2023-02-07:
You need to watch the serial logs from the device for more information.

totopesce, 2023-02-07:
It scans all networks, but when I try to get it to connect I get that error
`
[0;32m[I][wifi:257]: WiFi Connecting to 'SSID'...[0m
[0;33m[W][wifi_esp8266:212]: wifi_apply_hostname_(esphome-web-d0503a): lwIP error -16 on interface st (index 0)[0m
[0;33m[W][wifi:549]: Error while connecting to network.[0m
`

totopesce, 2023-02-07:
Managed to get it working here is what I did.

On my router:
Added the esp with a manual IP
the 2.4ghz band is in legacy
IGMP snooping is disabled

On the yaml file:
changed the board to d1_mini
added manual_ip where the dns1 is the same as the gateway

closed 2023-02-07 by totopesce