esphome/issues #4139: select sensor state read from HA always returns nan

oziee, 2023-02-07:
### The problem

I have an esphome fan device that creates a select sensor and this is added to HA. In HA the sensor is there, I can select items and the fan changes as it should

I have another esphome spaniel device that reads the HA sensor, but the state is always nan during on_value
In the on_value I log out some data for testing, and when in HA the select sensor value is changed the esphome device logs out
```
[11:04:13][D][sensor:127]: 'fantimer': Sending state nan  with 1 decimals of accuracy
[11:04:13][D][main:273]: **** fantimer Value check
[11:04:13][I][main:215]: *****Raw Value of my sensor: nan
[11:04:13][I][main:216]: *****Value of my sensor: nan
```
[image]

[image]

Not able to get either the state name or index etc..

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2202

### What platform are you using?

ESP32

### Board

esp32dev

### Example YAML snippet

```yaml
in the fan esphome device that creates the select sensor, all working good

select:
  - platform: template
    name: $device_name Timer
    optimistic: true
    options:
      - "Off"
      - "1hour"
      - "4hour"
    initial_option: 'Off'
    on_value:
      then:
        - lambda: |-
            int value = 0x0;
            if( x == "1hour" ) {
              value = 0x1;
            } else if( x == "2hour") {
              value = 0x3;
            } 
            id($name).set_enum_datapoint_value(103,value);
```

in the NSPanel esphome device I read the sensor back from HA, 
```
 - platform: homeassistant
    id: fantimer
    internal: true
    entity_id: select.lounge_room_fan_timer 
    on_value:
      then:
        - logger.log: "**** fantimer Value check"
        - lambda: |-
            ESP_LOGI("main", "*****Raw Value of my sensor: %f", id(fantimer).raw_state);
            ESP_LOGI("main", "*****Value of my sensor: %f", id(fantimer).state);
```
```

ssieb, 2023-02-07:
You're trying to read a text value as a number, which isn't going to work.  I would expect you have some "can't convert" errors in either the esphome or HA logs.  Github issues are not for support requests.  Ask on the discord server or one of the forums.

closed 2023-02-07 by ssieb

oziee, 2023-02-07:
sorry yes.. the code has been changing as I have tried everything like %s  but it still fails

oziee, 2023-02-07:
just so if anyone else in future has the same problem, its because you use a text_sensor for select entities