esphome/issues #4140: Socket connection fails after api encyption key cahnge

waterler, 2023-02-07:
### The problem

After I change api encryption key to new one, entity becomes unavailable in the UI.
Strange enough these logs are show connected wirelessly, so it can connect.

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.2

### What platform are you using?

ESP8266

### Board

SONOFF MINI

### Component causing the issue

api

### Example YAML snippet

```yaml
...
api:
  encryption: <redacted>
...
```

### Anything in the logs that might be useful for us?

```txt
[19:41:49][D][api:102]: Accepted 192.168.0.145
[19:41:49][W][api.connection:071]: 192.168.0.145: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
```

geeneo, 2023-02-17:
I have the same error on esp32 

I have disabled the encryption and still get this error.

[13:07:10][W][api.connection:071]: ::FFFF:172.xx..xx.xx: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[13:07:10][W][api.connection:071]: ::FFFF:172.xx..xx.xx: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11

Steps followed :

1) I have remove device.
2) reflash with web flash
3) add device to ESP Home

[13:07:10][W][api.connection:071]: ::FFFF:172.xx..xx.xx: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[13:07:10][W][api.connection:071]: ::FFFF:172.xx..xx.xx: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11

Code----
substitutions:
  name: esp32-bluetooth-proxy-off
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
---

MeAngry, 2023-02-20:
I am seeing the same issue.  I was changing over from api password to key and the first time I set the key it worked fine.  Went back and changed to a new key and I am seeing the same error message and entities are no longer updating in HA:

[api.connection:071]: ::FFFF:10.x.x.x: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11

mattfox27, 2023-02-25:
Me too same issue

Rantaki, 2023-03-29:
Any update on this?

descipher, 2023-04-17:
This seems to be device specific, I see this issue on a generic ESP23-S module but not a WROOM32 module running exactly the same esphome code.
 
The esptool reports a ESP32-D0WD-V3 chip is dected when flashed via serial.
Need to check the whats report on the ESP-WROOM-32, should be the same though.

```
(board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5

[13:21:52][D][api:102]: Accepted x.x.x.x
[13:21:52][VV][api.socket:358]: x.x.x.x noise_handshakestate_read_message failed: MAC_FAILURE
[13:21:52][W][api.connection:071]: x.x.x.x: Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11

```

rlust, 2023-06-23:
Having same issue on a M5 Atom Echo.

rlust, 2023-06-23:
Has anyone found a solution for this? What is causing the socket error?