esphome/issues #4143: esphome crashes after several switch on/off - Guru Meditation Error: Core  1 panic'ed (LoadProhibited)

vazir, 2023-02-08:
### The problem

After a few dosens of on/off clicks on the switch component over MQTT (via HA interface) - Any version of the ESP32 - esphome will crash when configured with switch component on GPIO. I just  click on the on/off switch as fast as i can. Sometimes I need to click several hundred times, sometimes much quicker. It's unpredictable. Look at the full log file since ESP start.  If everything except SWITCH component deleted from the config - it needs more clicks to crash, but it will crash anyways. Tested on different boards.
[logs_water-controller-test-switch_upload.txt](https://github.com/esphome/issues/files/10689357/logs_water-controller-test-switch_upload.txt) - the .yaml configuration is also attached.

### Which version of ESPHome has the issue?

latest stable and latest dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32-Devkit-C

### Component causing the issue

AsyncTCP

### Example YAML snippet

```yaml
substitutions:
  devicename: "water-controller-pt1"
  idpfx: "wctrlpt1"

esphome:
  name: $devicename
  platform: ESP32
  board: esp-wrover-kit

globals:
  - id: adc_zero_bar_volts
    type: float
    restore_value: yes
    initial_value: '0.167'
  - id: adc_bar_volts
    type: float
    restore_value: yes
    initial_value: '0.52'
  - id: adc_bar
    type: float
    restore_value: yes
    initial_value: '3.2'
# Formula: (CURRENT_ADC_READINGS - adc_zero_bar_volts) * (adc_bar/(adc_bar_volts-adc_zero_bar_volts));

wifi:
  ssid: "iot-ng"
  password: "NOPASSWORDHERE"
  reboot_timeout: 30s
  # power_save_mode: none
  power_save_mode: LIGHT
  fast_connect: false
  manual_ip:
    static_ip: 192.168.200.232
    gateway: 192.168.200.1
    subnet: 255.255.255.0
    dns1: 192.168.200.1

captive_portal:

# Enable logging
logger:

[... 246 lines cut ...]
 (inlined by) AsyncMqttClientInternals::PublishPacket::parsePayload(char*, unsigned int, unsigned int*) at /config/.esphome/build/water-controller-test-switch/.piolibdeps/water-controller-test-switch/AsyncMqttClient-esphome/src/AsyncMqttClient/Packets/PublishPacket.cpp:83
WARNING Decoded 0x400ea89e: AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int) at /config/.esphome/build/water-controller-test-switch/.piolibdeps/water-controller-test-switch/AsyncMqttClient-esphome/src/AsyncMqttClient.cpp:453
WARNING Decoded 0x400ea8b6: std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncMqttClient::AsyncMqttClient()::{lambda(void*, AsyncClient*, void*, unsigned int)#6}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at /config/.esphome/build/water-controller-test-switch/.piolibdeps/water-controller-test-switch/AsyncMqttClient-esphome/src/AsyncMqttClient.cpp:36
 (inlined by) _M_invoke at /config/.esphome/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
WARNING Decoded 0x40159dcd: std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at /config/.esphome/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
 (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at /config/.esphome/build/water-controller-test-switch/.piolibdeps/water-controller-test-switch/AsyncTCP-esphome/src/AsyncTCP.cpp:951
WARNING Decoded 0x40159e45: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at /config/.esphome/build/water-controller-test-switch/.piolibdeps/water-controller-test-switch/AsyncTCP-esphome/src/AsyncTCP.cpp:1230
WARNING Decoded 0x4015a5e2: _async_service_task(void*) at /config/.esphome/build/water-controller-test-switch/.piolibdeps/water-controller-test-switch/AsyncTCP-esphome/src/AsyncTCP.cpp:162
 (inlined by) _async_service_task at /config/.esphome/build/water-controller-test-switch/.piolibdeps/water-controller-test-switch/AsyncTCP-esphome/src/AsyncTCP.cpp:197
[10:30:01]
[10:30:01]
[10:30:01]
[10:30:01]
[10:30:01]ELF file SHA256: 0000000000000000
[10:30:01]
[10:30:01]Rebooting...
[10:30:01]ets Jun  8 2016 00:22:57
[10:30:01]
[10:30:01]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:30:01]configsip: 0, SPIWP:0xee
```

### Additional information

Tried in different configurations. Also tried AsyncTCP version from https://github.com/me-no-dev/ESPAsyncWebServer/issues/932 as issue looks alike. Crashes anyway.

vazir, 2023-02-09:
A little python code to test - will crash ESPHOME in a few seconds

import asyncio

import asyncio_mqtt as aiomqtt

async def push_cmd(client, topic, state):
    print(f'Sending: {topic} -> {state}')
    await client.publish(topic=topic, payload=state)

async def main():
    async with aiomqtt.Client(hostname='192.168.35.96', username='client', password='client') as client:
        while True:
            await push_cmd(client, 'water-controller-test-switch/switch/water-controller-test-switch_sw_1/command',
                           'ON')
            await push_cmd(client, 'water-controller-test-switch/switch/water-controller-test-switch_sw_2/command',
                           'ON')
            await push_cmd(client, 'water-controller-test-switch/switch/water-controller-test-switch_sw_3/command',
                           'ON')
            await push_cmd(client, 'water-controller-test-switch/switch/water-controller-test-switch_sw_4/command',
                           'ON')
            await asyncio.sleep(0.2)
            await push_cmd(client, 'water-controller-test-switch/switch/water-controller-test-switch_sw_1/command',
                           'OFF')
            await push_cmd(client, 'water-controller-test-switch/switch/water-controller-test-switch_sw_2/command',
                           'OFF')
            await push_cmd(client, 'water-controller-test-switch/switch/water-controller-test-switch_sw_3/command',
                           'OFF')
            await push_cmd(client, 'water-controller-test-switch/switch/water-controller-test-switch_sw_4/command',
                           'OFF')
            await asyncio.sleep(0.2)

asyncio.run(main())

vazir, 2023-02-09:
[logs_water-controller-test-switch_upload.txt](https://github.com/esphome/issues/files/10700606/logs_water-controller-test-switch_upload.txt)

vazir, 2023-02-09:
> 1: try without MQTT (use API) 2: try with MQTT boker in the same layer2 network as ESPHome

1. Native API is good, cannot replicate. 

```python
#!/usr/bin/env python3

import aioesphomeapi
import asyncio

async def main():
    api = aioesphomeapi.APIClient("192.168.200.231", 6053, "client")
    await api.connect(login=True)

    # Get API version of the device's firmware
    print(api.api_version)

    # Show device details
    device_info = await api.device_info()
    print(device_info)

    # List all entities of the device
    entities = await api.list_entities_services()
    print(entities)

    keys = [1252111057, 1252111056, 1252111058, 1252111063]
    it = 0
    while True:
        it += 1
        print(f'iter: {it}')
        for k in keys:
            await api.switch_command(k, True)
        await asyncio.sleep(0.1)
        for k in keys:
            await api.switch_command(k, False)
        await asyncio.sleep(0.1)

asyncio.run(main())
```

ssieb, 2023-02-09:
Probably a similar issue to https://github.com/esphome/esphome/pull/3951

vazir, 2023-02-12:
@oxan made a fix, not yet available in mainsteam or dev, applicable to -dev revision 045489e6d7c9b772509feda0f962be35d7b766b2 or see here:  https://github.com/oxan/esphome/tree/wip/scheduler-locking - fixes the issue fine for me.

```diff
diff --git a/esphome/components/mqtt/mqtt_client.cpp b/esphome/components/mqtt/mqtt_client.cpp
index aa0cf56c..acb86324 100644
--- a/esphome/components/mqtt/mqtt_client.cpp
+++ b/esphome/components/mqtt/mqtt_client.cpp
@@ -485,16 +485,16 @@ static bool topic_match(const char *message, const char *subscription) {
 }
 
 void MQTTClientComponent::on_message(const std::string &topic, const std::string &payload) {
-#ifdef USE_ESP8266
-  // on ESP8266, this is called in LWiP thread; some components do not like running
-  // in an ISR.
+#ifdef USE_ARDUINO
+  // on Arduino, this is called in lwIP/AsyncTCP task; some components do not like running
+  // from a different task.
   this->defer([this, topic, payload]() {
 #endif
     for (auto &subscription : this->subscriptions_) {
       if (topic_match(topic.c_str(), subscription.topic.c_str()))
         subscription.callback(topic, payload);
     }
-#ifdef USE_ESP8266
+#ifdef USE_ARDUINO
   });
 #endif
 }
diff --git a/esphome/core/scheduler.cpp b/esphome/core/scheduler.cpp
index d880f0fd..8c1ef32f 100644
--- a/esphome/core/scheduler.cpp
+++ b/esphome/core/scheduler.cpp
@@ -13,6 +13,11 @@ static const uint32_t MAX_LOGICALLY_DELETED_ITEMS = 10;
 // Uncomment to debug scheduler
 // #define ESPHOME_DEBUG_SCHEDULER
 
+// A note on locking: the `lock_` lock protects the `items_` and `to_add_` containers. It must be taken when writing to
+// them (i.e. when adding/removing items, but not when changing items). As items are only deleted from the loop task,
+// iterating over them from the loop task is fine; but iterating from any other context requires the lock to be held to
+// avoid the main thread modifying the list while it is being accessed.
+
 void HOT Scheduler::set_timeout(Component *component, const std::string &name, uint32_t timeout,
                                 std::function<void()> func) {
   const uint32_t now = this->millis_();
@@ -121,7 +126,7 @@ void HOT Scheduler::set_retry(Component *component, const std::string &name, uin
   args->backoff_increase_factor = backoff_increase_factor;
   args->scheduler = this;
 
-  // First exectuion of `func` immediately
+  // First execution of `func` immediately
   this->set_timeout(component, args->name, 0, [args]() { retry_handler(args); });
 }
 bool HOT Scheduler::cancel_retry(Component *component, const std::string &name) {
@@ -150,30 +155,42 @@ void HOT Scheduler::call() {
     std::vector<std::unique_ptr<SchedulerItem>> old_items;
     ESP_LOGVV(TAG, "Items: count=%u, now=%u", this->items_.size(), now);
     while (!this->empty_()) {
+      this->lock_.lock();
       auto item = std::move(this->items_[0]);
+      this->pop_raw_();
+      this->lock_.unlock();
+
       ESP_LOGVV(TAG, "  %s '%s' interval=%u last_execution=%u (%u) next=%u (%u)", item->get_type_str(),
                 item->name.c_str(), item->interval, item->last_execution, item->last_execution_major,
                 item->next_execution(), item->next_execution_major());
 
-      this->pop_raw_();
       old_items.push_back(std::move(item));
     }
     ESP_LOGVV(TAG, "\n");
[... 87 lines cut ...]
+  std::lock_guard<std::mutex> guard(this->lock_);
   bool ret = false;
   for (auto &it : this->items_) {
     if (it->component == component && it->name == name && it->type == type && !it->remove) {
diff --git a/esphome/core/scheduler.h b/esphome/core/scheduler.h
index a758198b..7d39e435 100644
--- a/esphome/core/scheduler.h
+++ b/esphome/core/scheduler.h
@@ -3,6 +3,7 @@
 #include "esphome/core/component.h"
 #include <vector>
 #include <memory>
+#include <mutex>
 
 namespace esphome {
 
@@ -71,6 +72,7 @@ class Scheduler {
     return this->items_.empty();
   }
 
+  std::mutex lock_;
   std::vector<std::unique_ptr<SchedulerItem>> items_;
   std::vector<std::unique_ptr<SchedulerItem>> to_add_;
   uint32_t last_millis_{0};
```

nagyrobi, 2023-02-12:
@vazir please don't post walls of quotimg codes.

closed 2023-02-26 by jesserockz