esphome/issues #4144: How to exit safe-mode?

kripper, 2023-02-08:
### The problem

In the current documentation it is not clear:

- if the "unexpected resets counter" is automatically reset after some time. Otherwise, each time I plug off and on the ESP32, the counter will increment and at some point it will unnecesarily boot into safe mode, where I have no control of the device any more (until I upload a new firmware via OTA?).
- if the device will exit safe mode automatically after some time. The OTA Component has a `reboot_timeout` parameter defined as "The amount of time to wait before rebooting when in safe mode". Does this mean that after `reboot_timeout` seconds, the ESP32 will exit from safe mode and boot normally again?

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

WROOM

ssieb, 2023-02-08:
Github issues are not for support requests, ask on discord or one of the forums.

> * Does this mean that after `reboot_timeout` seconds, the ESP32 will exit from safe mode and boot normally again?

Yes, or you can just reboot the device yourself.

closed 2023-02-08 by ssieb

kripper, 2023-02-08:
@ssieb thanks.

Is there any way to reset the "unexpected resets counter"?
I don't want to disable safe-mode, but I also don't want to enter safe-mode after every 8 reboots.

ssieb, 2023-02-09:
It's 10 reboots and the counter is reset after a few minutes of running.

kripper, 2023-02-09:
Excellent. As expected. Hopefully this could be added to the documentation.