esphome/issues #4145: Mopeka Pro Check Discarding Data

dejab, 2023-02-09:
### The problem

The sensor code discards data from the Mopeka Sensor that is marked as QUALITY_LOW (1 Star)
I realize the data reliabilty issue associated with the star rating, however, some LPG tanks just wont give higher quality readings due to debris inside tank, corrosion, paint job on outside of tank, etc... Also, the Mopeka App accepts and displays 1-star data quality.

Please consider an option to select minimum quality accepted overall or per tank sensor.

I myself have two 500 gallon tanks. One I can get a 3 star rating, the other 1 star only (2 stars if I'm lucky)

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.3

### What platform are you using?

ESP32-IDF

### Component causing the issue

esphome/components/mopeka_pro_check/mopeka_pro_check.cpp

### Additional information

Lines 73 - 80 of mopeka_pro_check.cpp

    if (quality_value < QUALITY_HIGH) {
      ESP_LOGW(TAG, "Poor read quality.");
    }
    if (quality_value < QUALITY_MED) {
      // if really bad reading set to 0
      ESP_LOGW(TAG, "Setting distance to 0");
      distance_value = 0;
    }

dejab, 2023-03-21:
> What's your custom configuration if you don't mind me asking for the 500 gallon tanks? As there's no 500 gallon tank type. I am about to set one up today.
>
> https://esphome.io/components/sensor/mopeka_pro_check.html#tank-types

Here are the setting I’m using:

 

    tank_type: CUSTOM

    custom_distance_full: 94cm

    custom_distance_empty: 3.8cm

phurth, 2023-04-18:
I'll second this. The current code appears to discard any reading with less than 2 out of 3 quality "stars". I don't know if there are 3 or 4 possible quality readings (where zero stars would be the 4th). Assuming there are, it might be better if the code were something like "if (quality_value < QUALITY_**LOW**)...". Even better would be to not update the sensor to 0 if you get a bad reading - just don't update it at all. That way you'd just retain whatever the last value was and usually these quality drops are transient. That appears to be what the native Mopeka app does.

thenotoriousdrr, 2023-07-10:
I too have this issue. Nothing much to contribute, just trying to bump the thread.

spbrogan, 2023-10-28:
I will try to find some time to review this PR again and get it ready to merge.  https://github.com/esphome/esphome/pull/4928

deon-dup, 2024-08-23:
Read about this issue, and thought it has been resolved. I agree with the data quality being available, so one can perhaps override the reset to zero, if data quality is low. I don get readings, have applied the sensor correctly as the App indicates as such, but through my ESP32Dev, I cannot get readings, because it is reset to zero. Any advice?

closed 2024-10-29 by jesserockz