esphome/issues #4146: Bedjet not seen by ESPHome

merak69, 2023-02-09:
### The problem

I've flashed ESPHome 2022.12.8 via Home Assistant (and directly with esphomeflasher) on two different ESP32 devices and neither one sees my BedJet v3 device.  I've turned up logging and can see a bunch of devices but the BedJet is never listed.  I've tried removing the bedjet components from the yaml as well to no avail.

Thinking maybe my devices were bad, I flashed BLE-Scanner ([Link](https://github.com/gromeck/BLE-Scanner)) and it shows the BedJet right away.

Config (with obvious keys removed):
```
esphome:
  name: bedjet-esp

esp32:
  board: esp32dev

# Enable logging
logger:
  level: VERY_VERBOSE
  
# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bedjet-Esp Fallback Hotspot"
    password: ""

captive_portal:

esp32_ble_tracker:

ble_client:
  - mac_address: e0:e2:e6:d4:d1:d2
    id: bedjet_ble_id1

bedjet:
  - id: bedjet_1
    ble_client_id: bedjet_ble_id1

climate:
  - platform: bedjet
    id: my_bedjet_climate_entity
    name: "My BedJet"
    bedjet_id: bedjet_1

fan:
  - platform: bedjet
    id: my_bedjet_fan_entity
    name: "My BedJet Fan"
    bedjet_id: bedjet_1

```

Hopefully it's just something stupid I'm doing but I've been banging my head on the desk for awhile now.  

Attached is a screenshot of the logs from BLE-Scanner showing the mac address listed above in the config.
[image]

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32D

merak69, 2023-02-09:
With Verbose turned off, this is basically what I see forever:

```
[18:50:28][C][logger:293]: Logger:
[18:50:28][C][logger:294]:   Level: DEBUG
[18:50:28][C][logger:295]:   Log Baud Rate: 115200
[18:50:28][C][logger:296]:   Hardware UART: UART0
[18:50:28][C][esp32_ble_tracker:870]: BLE Tracker:
[18:50:28][C][esp32_ble_tracker:871]:   Scan Duration: 300 s
[18:50:28][C][esp32_ble_tracker:872]:   Scan Interval: 320.0 ms
[18:50:28][C][esp32_ble_tracker:873]:   Scan Window: 30.0 ms
[18:50:28][C][esp32_ble_tracker:874]:   Scan Type: ACTIVE
[18:50:28][C][esp32_ble_tracker:875]:   Continuous Scanning: True
[18:50:28][C][ble_client:027]: BLE Client:
[18:50:28][C][ble_client:028]:   Address: E0:E2:E6:D4:D1:D2
[18:50:28][C][captive_portal:088]: Captive Portal:
[18:50:28][C][mdns:103]: mDNS:
[18:50:28][C][mdns:104]:   Hostname: bedjet-esp
[18:50:28][C][ota:093]: Over-The-Air Updates:
[18:50:28][C][ota:094]:   Address: bedjet-esp.local:3232
[18:50:28][C][ota:097]:   Using Password.
[18:50:28][C][api:138]: API Server:
[18:50:28][C][api:139]:   Address: bedjet-esp.local:6053
[18:50:28][C][api:141]:   Using noise encryption: YES
[18:50:28][C][bedjet:485]: BedJet Hub 'E0:E2:E6:D4:D1:D2'
[18:50:28][C][bedjet:486]:   ble_client.app_id: 1
[18:50:28][C][bedjet:487]:   ble_client.conn_id: 65535
[18:50:28][C][bedjet:488]:   Update Interval: 15.0s
[18:50:28][C][bedjet:489]:   Child components (2):
[18:50:28][C][bedjet:491]:     - BedJet Climate
[18:50:28][C][bedjet:491]:     - BedJet Fan
[18:50:28][C][bedjet:039]: BedJet Climate 'My BedJet'
[18:50:28][C][bedjet:042]:   Supported modes:
[18:50:28][C][bedjet:044]:    - OFF
[18:50:28][C][bedjet:044]:    - HEAT
[18:50:28][C][bedjet:044]:    - FAN_ONLY
[18:50:28][C][bedjet:044]:    - DRY
[18:50:28][C][bedjet:049]:    - BedJet heating mode: HEAT
[18:50:28][C][bedjet:052]:   Supported fan modes:
[18:50:28][C][bedjet:057]:    - 10% (c)
[18:50:28][C][bedjet:057]:    - 100% (c)
[18:50:28][C][bedjet:057]:    - 15% (c)
[18:50:28][C][bedjet:057]:    - 20% (c)
[18:50:28][C][bedjet:057]:    - 25% (c)
[18:50:28][C][bedjet:057]:    - 30% (c)
[18:50:28][C][bedjet:057]:    - 35% (c)
[18:50:28][C][bedjet:057]:    - 40% (c)
[18:50:28][C][bedjet:057]:    - 45% (c)
[18:50:28][C][bedjet:057]:    - 5% (c)
[18:50:28][C][bedjet:057]:    - 50% (c)
[18:50:28][C][bedjet:057]:    - 55% (c)
[18:50:28][C][bedjet:057]:    - 60% (c)
[18:50:28][C][bedjet:057]:    - 65% (c)
[18:50:28][C][bedjet:057]:    - 70% (c)
[18:50:28][C][bedjet:057]:    - 75% (c)
[18:50:28][C][bedjet:057]:    - 80% (c)
[18:50:28][C][bedjet:057]:    - 85% (c)
[18:50:28][C][bedjet:057]:    - 90% (c)
[18:50:28][C][bedjet:057]:    - 95% (c)
[18:50:28][C][bedjet:060]:   Supported presets:
[18:50:28][C][bedjet:062]:    - BOOST
[18:50:28][C][bedjet:065]:    - EXT HT (c)
[18:50:28][C][bedjet:065]:    - M1 (c)
[18:50:28][C][bedjet:065]:    - M2 (c)
[18:50:28][C][bedjet:065]:    - M3 (c)
[18:50:28][C][bedjet:011]: BedJet Fan 'My BedJet Fan'
[18:50:28][C][bedjet:154]:   Speed: YES
[18:50:28][C][bedjet:155]:   Speed count: 20
[18:50:40][D][bedjet:505]: [E0:E2:E6:D4:D1:D2] Not connected, will not send status.
[18:50:55][D][bedjet:505]: [E0:E2:E6:D4:D1:D2] Not connected, will not send status.
[18:51:10][D][bedjet:505]: [E0:E2:E6:D4:D1:D2] Not connected, will not send status.
[18:51:17][D][esp32.preferences:113]: Saving 1 preferences to flash...
[18:51:17][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[18:51:18][D][bedjet:344]: [My BedJet] update()
[18:51:18][D][bedjet:348]: [My BedJet] update_status result=false
[18:51:22][D][bedjet:093]: [My BedJet Fan] update()
[18:51:22][D][bedjet:097]: [My BedJet Fan] update_status result=false
[18:51:25][D][bedjet:505]: [E0:E2:E6:D4:D1:D2] Not connected, will not send status.

```

InfiniteFeld, 2023-02-09:
Same thing for me... Working great for months and then nothing, can't connect at all

confuserated, 2023-02-15:
same thing for me too, with Xiaomi mijia bluetooth temp/humidity sensors, they initially connect but then start failing after several days and nothing I do can make them reconnect (I see no bluetooth devices in the ESPHome logs. where I would normally see even the broadcasts from devices not configured for that controller), including reflashing the ESP32 with the same code again.

If I revert back to ESPHome 2022.11.5 everything works properly again. I've tried virtually every release since then and they all have the same problem and restoring 2022.11.5 and updating the ESP32 controllers (I have 5 microcontrollers of two different varieties and they all display the same problems) enables bluetooth comms to work again.

confuserated, 2023-02-15:
Aha, well, I have a backup :)

I'm running Home Assistant Operating System (the one with the Supervisor) so updating an add-on gives you the option to backup before updating, which you can then restore by going into settings:system:backups and selecting the backup you want to restore over your updated ESPHome.

If you are running in a docker container then things will be a little more complicated but you could copy all the config yaml files for ESPHome, then delete your existing ESPHome container and re-installing with the earlier version.

merak69, 2023-02-18:
Is there a way to manually build ESPHome to an old version?  I'm not finding good documentation on how to do that.

merak69, 2023-02-18:
FWIW, I figured out how to install an older version of ESPHome locally in Windows.

1. Install WSL 2.0 in Windows [https://learn.microsoft.com/en-us/windows/wsl/install#update-to-wsl-2](https://learn.microsoft.com/en-us/windows/wsl/install#update-to-wsl-2)  - Make sure you're on v2 not v1
2. Install Ubuntu in WSL
3. Start WSL and follow the instructions here: [https://esphome.io/guides/installing_esphome.html](https://esphome.io/guides/installing_esphome.html) except don't do the install command.
4. Use this command to install the 2022.11.5 mentioned above: `pip3 install esphome==2022.11.5`
5. Go to the directory where your yaml is and do a `esphome compile youryaml.yaml`

confuserated, 2023-02-23:
> FWIW, I figured out how to install an older version of ESPHome locally in Windows.
> 
> 1. Install WSL 2.0 in Windows https://learn.microsoft.com/en-us/windows/wsl/install#update-to-wsl-2  - Make sure you're on v2 not v1
> 2. Install Ubuntu in WSL
> 3. Start WSL and follow the Linux instructions here: https://esphome.io/guides/installing_esphome.html except don't do the install command.
> 4. Use this command to install the 2022.11.5 mentioned above: `pip3 install esphome==2022.11.5`
> 5. Go to the directory where your yaml is and do a `esphome compile youryaml.yaml`
> 
> Edit: Not that this mattered - I got the device to connect once to my bedjet thus far but after a reboot, it doesn't ever find the bed again (lots of other devices in the logs).

just to be clear, for me, once I installed 2022.11.5 firmware on my esp32 I upgraded ESPHome to latest version and all works ok now, I've kept a copy of the firmware in case I need to replace the esp32 and still have 2022.11.5 firmware on it.

Hoping that they fix their issue so I can update.

jrames, 2023-03-13:
I had such high hopes, but dropping down to `2022.11.5` and flashing again didn't fix it for me.

Strangely, I had this working intermittently with 2023.2.4 but at the moment nothing gets it connecting.

merak69, 2023-03-13:
2023.2.4 also has it working intermittently for me as well but I've noticed when it is connected, eventually it'll start reporting wrong values.  Last night the bedjet was on for a few hours but the ESP said it was off.  The logs even showed it was "connected" and that the state was off, which was incorrect.

merak69, 2023-04-06:
As an update, I've noticed that after about 24-48 hours, the ESP device drops the connection to the bedjet and never reestablishes until a reboot is done.  I'm not sure what to do with this at this point.

confuserated, 2023-04-06:
me too, bluetooth devices drop after a couple of days on latest firmware, I keep one device on latest to test for fix to no avail, all the others are still happy on 2022.11.5, wish I knew why that doesn't work for you.

confuserated, 2023-06-13:
I've been trying 

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf

for a couple of days (instead of Arduino) with no errors on the latest version on same devices, will update in a week if I still have no errors.

MrMSch, 2023-06-20:
I tried with esp-idf and it worked for about 18 hours and then stopped being reliable (shows my bedjet being on for over 30 hours now even though it shows as off on the remote).

I've tried multiple ESP32 devices too, all with the same result.  I'm not sure where to go from here.