esphome/issues #4149: ESP32-S3-WROOM-2 reboot loop after flash

royboyroy28, 2023-02-09:
### The problem

After flashing the ESP32-S3-WROOM-2 the device goes into reboot loop.

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377474
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x4037786a:0x3fceb1800x4037dd15:0x3fceb1a0 0x40383c11:0x3fceb1c0 0x42027722:0x3fceb2f0 0x403771d5:0x3fceb320 0x403cd6ef:0x3fceb350 0x403cd9ae:0x3fceb380 0x403c992d:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Espressif ESP32-S3-Devkitc-1 v1.1

### Anything in the logs that might be useful for us?

```txt
ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377474
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x4037786a:0x3fceb1800x4037dd15:0x3fceb1a0 0x40383c11:0x3fceb1c0 0x42027722:0x3fceb2f0 0x403771d5:0x3fceb320 0x403cd6ef:0x3fceb350 0x403cd9ae:0x3fceb380 0x403c992d:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED
```

evlo, 2023-10-10:
does this only happen when using esp-idf?

Lateralleap, 2024-02-26:
No, arduino too. It's still an ongoing issue

Lateralleap, 2024-03-08:
These settings work:
```
esphome:
  name: n32r8v-test
  friendly_name: N32R8V Test
  platformio_options:
    board_build.arduino.memory_type: opi_opi

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
  flash_size: 32MB
```
Many thanks to @ssieb for his assistance in solving this issue

closed 2024-03-09 by ssieb