esphome/issues #4151: Denky D4 board pins are wrong

Djiit, 2023-02-10:
### The problem

Hey team,

When trying to save and install my configuration on a Denky-D4 board (see: https://github.com/hallard/Denky-D4) wit h the following configuration, I got an error about a pin I'm not supposed to use. But it is used on my board as a Tx pin (see github repo).

Looking at the code, it seems the Denky D4 is "just" using esp32 base pin configuration -- maybe it would need a specific one? see https://github.com/esphome/esphome/blob/f500f448b12d10c62df0ca31d82729e9612aa342/esphome/components/esp32/boards.py#L1211

Thanks

### Which version of ESPHome has the issue?

2022.12.8

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.12.1

### What platform are you using?

ESP32

### Board

denky_d4

### Example YAML snippet

```yaml
esphome:
  name: denky-d4

esp32:
  board: denky_d4

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Denky-D4 Fallback Hotspot"
    password: "redacted"

captive_portal:

uart:
  rx_pin: 8 # error appears here
  baud_rate: 9600

teleinfo:

sensor:
  - platform: teleinfo
    tag_name: "HCHC"
    name: "hchc"
    unit_of_measurement: "Wh"
    icon: mdi:flash
  - platform: teleinfo
    tag_name: "HCHP"
    name: "hchp"
    unit_of_measurement: "Wh"
    icon: mdi:flash
  - platform: teleinfo
    tag_name: "PAPP"
    name: "papp"
    unit_of_measurement: "VA"
    icon: mdi:flash

text_sensor:
  - platform: teleinfo
    tag_name: "OPTARIF"
    name: "optarif"
```
```

### Additional information

Board on platformIO: https://docs.platformio.org/en/latest/boards/espressif32/denky_d4.html

alucryd, 2023-09-15:
Have you tried using the `adafruit_qtpy_esp32` instead? I have yet to receive my D4, but it's based off that Adafruit board, it allows me to choose `GPIO8` in the ESPHOME editor.

hallard, 2023-09-15:
Denky D4 boards use a specific version ESP32-PICO-V3-02 chip with 8Mb Flash and 2Mb PSRAM integrated.

I had the same issue in Tasmota until they added this version.

It's the same chip than in the https://www.adafruit.com/product/5395 so if ESPHome support this one all should be OK

Not sure if `adafruit_qtpy_esp32` (without PICO) works, looks like @alucryd validated this point :-)

alucryd, 2023-09-15:
I can't find the pico variant in PlatformIO, the regular one does have 8MB of RAM, but there's no mention of the integrated PSRAM. Will try flashing as soon as I receive the D4, will look into submitting a PR to PlatformIO of it does not work.

closed 2023-10-02 by jesserockz

hallard, 2023-10-02:
Standard mode is 9600 baud instead of 1200 with more data and subject to checksum error depending on some magic, like Linky model, wires, conectors, ... not really ideal value I always got some checksum errors but not on each frame for sure.

Thanks for the PR

Toniob, 2023-10-16:
I’ve tried to configure and upload an esphome firmware with the new "denky_d4" board, but the resulting firmware makes the board go into a bootloop.

If I change the board to adafruit_qtpy_esp32, the board boots fine and the device is available in my esphome dashboard.

I’m using esphome version 2023.9.3, so it should work, right ?

alucryd, 2023-10-16:
@Toniob It should indeed, does it go far enough to get some logs when flashing over usb?

Toniob, 2023-10-17:
I’ve got those messages in the logs : 
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0030,len:6608
load:0xffffffff,len:-1
ets Jul 29 2019 12:21:46
```

Here is my configuration : 
```
esphome:
  name: denky-d4
  friendly_name: Linky
  name_add_mac_suffix: false

esp32:
#  board: adafruit_qtpy_esp32
  board: denky_d4
  framework:
    type: esp-idf

# Enable logging
logger:
  baud_rate: 0

api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

uart:
  rx_pin: GPIO8
  baud_rate: 9600
  parity: EVEN
  data_bits: 7
  stop_bits: 1

light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    chipset: SK6812
    pin: GPIO14
    num_leds: 1
    rmt_channel: 0
    name: "Status LED"
    restore_mode: ALWAYS_OFF

teleinfo:
  id: linky
  update_interval: 60s
  historical_mode: false

sensor:
  ... (list of tags from teleinfo)
```